opendatacam/opendatacam #38: video not visible while counting/tracking

nekromoff, 2018-04-14:
Only black background is shown when counting starts. Until then, the video is visible and "lines" can be placed. Once the counting starts, video is not shown anymore.

Also, FPS is low (6 FPS) with Logitech C922 and counting from above (view point) is hardly precise. I know, multiples issues...

florianporada, 2018-04-17:
Hi @nekromoff, 
thanks for testing the open-data-cam! 👍

For now, the video-stream is only visible during the setup process. 

Concerning the FPS rate; at the moment we have to run on this rate because of the Jetson Board's computing power. (I assume you run the Jetson on overclocking mode?)

Keep in mind: this is still alpha. 🛠
Feel free to contribute. If you see room for improvements, just create a PR. 🙃

nekromoff, 2018-04-17:
Thanks for the explanation and your project!

Yes, I am running Jetson TX2 in overclocking. Is there a stronger board
that could handle higher FPS?

florianporada, 2018-04-17:
Sure, more computing power leads probably to an higher FPS rate 😀
But we only tested it on the Jetson TX2 so far. 

If you have an test rig (can be any pc) with:
- ubuntu 14.04 (or above)
- cuda
- opencv
- and the dependencies mentioned in the install guide

then you should be able to test it with a higher frame-rate.
Keep us updated! 📡

nekromoff, 2018-04-17:
Thanks, will do (and also report regarding success of traffic counting).

florianporada, 2018-04-19:
I'll look into the exporting issue.

Concerning the cyclist tracking: Sometimes the trained model we use has issues with distinguishing cyclists and scooters or motorcycles. (a potential fix would be to use another/better trained model)

tdurand, 2018-08-22:
Closing this one:

- as stated by @florianporada video-stream is only visible during the setup process because of the computing power of the jetson board for now...

- the scooter / bicycle distinction is poor, this is due to the YOLO model beeing poor at it.. when we will have more computing power we may be able to have a stronger model for this.

closed 2018-08-22 by tdurand