opendatacam/opendatacam #39: websocket server: client can't reconnect after a certain time.

florianporada, 2018-04-17:
After running some longtime tests I encountered a problem with the websocket connection.

```
Tue Apr 17 2018 11:45:12 GMT+0000 (UTC) Peer ::ffff:127.0.0.1 disconnected.
```

The client loses the connection to the server process. Reloading the Browser results in the following error 

```
(node:13929) UnhandledPromiseRejectionWarning: Error: connect ECONNREFUSED 127.0.0.1:80
    at Object._errnoException (util.js:1022:11)
    at _exceptionWithHostPort (util.js:1044:20)
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1198:14)
(node:13929) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 2)
(node:13929) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

My first guess is, that the Browser tries to connect on port __80__ instead of 8080 (or is the websocket process running on 80?)

cc @tdurand any ideas on that?

florianporada, 2018-04-17:
After some testing I found out that the error above (`(node:13929)....`) is not related with the issue.
Still, after some time (2h - 3h) the open-data-cam stops counting.
The terminal log however does not provide any information. (maybe darknet or some other process is stopping..?)

tdurand, 2018-04-17:
Hum... Yeah it sounds very likely that the YOLO process (./darknet) is crashing, disconnecting from the websocket.

As it was a rough POC, if YOLO crashes, we aren't restarting it... We aren't monitoring whether it crashes also... 

You could try to add more logs here: https://github.com/moovel/lab-opendatacam/blob/master/server/processes/YOLO.js#L48

for example:
```javascript
 YOLO.process.on("stdout", (data) => {
      console.log('Log from yolo process');
      console.log(data)
    });

    YOLO.process.on("stderr", (data) => {
      console.log('Error from yolo process');
      console.log(data)
    });
```

Also the way I start the YOLO process could restart it automaticaly in case of crash, I had set it to just start it once for the POC but we might try to change the value to something else so the YOLO process restarts itself in case of crash (max param of forever.Monitor)

```
YOLO.process = new (forever.Monitor)(['./darknet','detector','demo','cfg/voc.data','cfg/yolo-voc.cfg','yolo-voc.weights','-c','1', '-address','ws://localhost','-port','8080'],{
      max: 10000, 
      cwd: config.PATH_TO_YOLO_DARKNET,
      killTree: true
    });
```

But think first we should identify why it crashes

florianporada, 2018-04-19:
Ok, I guess the best thing to do is to see if the YOLO process is still running (or not running) when this issue occurs. 
And from there see where it goes.

tdurand, 2018-04-19:
Yep, you can try to see the process running on the machine also with something like `ps -ef | grep darknet`

florianporada, 2018-04-25:
I had the camera running for 2 days now and the process did not stop. 
Maybe it is a problem with PM2 because I had it running directly via `node start ...` (as in the `package.json`)

tdurand, 2018-04-25:
Hum... who knows ! But pm2 does not orchestrate the yolo process, it just start the node server that then orchestrate the yolo and webcam process. I think in your case the node process wasn't crashing so pm2 might not be the problem...

florianporada, 2018-05-07:
After another long time test I could not reproduce the issue. I think we can close this for now.

closed 2018-05-07 by florianporada