opendatacam/opendatacam #41: Tracked item canvas visualization

tdurand, 2018-05-25:
Hi @b-g , 

I spent half-day to do a really quick and dirty POC of this, I think what would be usefull next would be to discuss it on skype and see what we do from there.  The code is on the path-visualization branch (https://github.com/moovel/lab-opendatacam/tree/path-visualization) but I think it is not really usefull for you to review this.

The current POC look like this:
[image]

You can try it here: https://open-traffic-cam-fqhztoewzf.now.sh/ , it simulates YOLO detections at 8 FPS.

Best, 

Thibault

b-g, 2018-05-27:
@tdurand Great! I won't be available the next days/week ... Will be traveling in Russia the entire week. Sorry! Will ping you once I'm back. Should have time on Jun 4 or Jun 8.

tdurand, 2018-06-08:
Todo:

- [ ] Json export of history
- [ ] Store history in database , maybe nedb https://github.com/louischatriot/nedb instead of in node.js memory (RAM) to avoid node.js crashing after a few days
- [ ] Properly integrate the canvas rendering in its slide

tdurand, 2018-06-11:
Ping @florianporada  , I've implemented everything we discussed on friday.

Please note I didn't get the time to try it on my jetson, I think this should work as I didn't touch any core things like YOLO or webcam stream... But yeah, you never now, so if this bugs, just ping me and I'll run it on my jetson first.

I've saved the previous state of the software on the v1 branch in case we want to revert, the latest code beeing master, you just need to pull that branch and re-run the thing on the jetson

To discuss then:

- For the path visualization, need to see if we put the slide first ? when do we clear the canvas? (for now I do it when you move from the slide) .. .... 
- The json of the history of all tracked item can become quite huge, normally we shouldn't run into bugs because I'm just appending to the file and never reading all the file in memory _(learned from experience, I've hit a memory error after running it 2h and trying to export)_ , its just that people that want to use it should read it as a stream.

if you can try this tomorrow so I can maybe fix thing tomorrow / wednesday it would be great as my end of week is a bit crazy ! thanks !

florianporada, 2018-06-11:
see https://github.com/moovel/lab-opendatacam/issues/42

b-g, 2018-06-12:
Great! Looking forward to see it. Only comment ...

> For the path visualization, need to see if we put the slide first ? when do we clear the canvas? (for now I do it when you move from the slide)

I would vote not to clear the canvas on a navigation event, to keep it consistently with an erase via time mechanism e.g. to paint every nth frame a very transparent rect with the background color across the entire screen. That way it will always show a visualisation of the "past" of the duration x.

tdurand, 2018-06-12:
👍 , I did a first shot at your idea, drawing a 0.3 opacity rect every minute, it's kind of hard to know how much opacity / timing of the clearing because it will much depend of how much we are drawing. 

I'll let you tweak the params here: https://github.com/moovel/lab-opendatacam/blob/master/components/shared/PathVisualization.js#L27 and timing here: https://github.com/moovel/lab-opendatacam/blob/master/components/shared/PathVisualization.js#L83

b-g, 2018-06-12:
@tdurand Thanks!
@florianporada Can you tweak the parameters? I would rather draw every sec something tiny like 0.03 opacity ...

florianporada, 2018-06-13:
I'll tweak the params a bit and see how it goes. 👍

florianporada, 2018-06-14:
So I tweaked the config a bit and the path-viz now has a quite nice "fade-out" animation style. 
The only thing we need to do is, figuring out a good ratio between opacity/timing (as @tdurand mentioned) 
I guess it will also depend on how many objects are tracked at a certain time.

b-g, 2018-06-14:
@florianporada Great! Happy to tweak this with you next Monday ...

tdurand, 2018-06-14:
nice, yeah I figured maybe also later on it could be interesting to have a "live" view on the canvas of only the thing that are currently tracked

b-g, 2018-08-14:
Hi @tdurand, could you re-deploy the POC with latest changes (of https://github.com/moovel/lab-opendatacam/issues/41#issue-326684525) again? Would be great to facilitate discussion today 14°°. THX!

b-g, 2018-08-14:
Closed. Added to new todo list in https://github.com/moovel/lab-opendatacam/issues/46.

closed 2018-08-14 by b-g