opendatacam/opendatacam #42: Screenshot of Frame + Counting-lines.

florianporada, 2018-06-08:
Hi @tdurand, 

one quick question. Is it possible to take a screenshot of the lines + frame when the counting starts and display that as background image during the counting with opacity 0.3 or something? :)

tdurand, 2018-06-08:
Yes, you mean having the canvas drawing of the lines in background of the counters ? What do you mean by frame ?

florianporada, 2018-06-08:
By Frame I mean the webcam.jpg at the point where the user is pressing "start counting" + the canvas with the drawn lines. (like this https://imgur.com/a/BrOLtqo) on top of that maybe a slightly transparent black layer so that the counter has a good visibility. 
The background should just be a reminder for the user where the lines where drawn :)

tdurand, 2018-06-08:
Ah yeah, good idea 👍

florianporada, 2018-06-11:
Hi @tdurand, I just checked the version and found some issues. 

When I'm creating the counting-lines, I have to press the "start counting" button 5(ish) times in order to switch to the counting mode. In the DOM inspector I can see, that everytime I click the button, a new element with a screenshot gets added.

The screenshot however is not being displayed in the counting screen :( ... I added some styling in the inspector in order to display the screenshot. But created screenshot is not showing the lines. 

Did you encounter something similar? 
Here a reference:

https://imgur.com/a/pLTetY2

The path viz works 👍 ... i'll test it in the next few days 😃

tdurand, 2018-06-11:
Will fix tomorrow !

tdurand, 2018-06-12:
@florianporada it should work now, it was a cross origin mess when doing things from client, I'm doing it from server-side now it's much cleaner.

tdurand, 2018-06-12:
Don't forget to run npm install I've added a new dependency

florianporada, 2018-06-12:
Thanks 👍 I'll test it ASAP 🙂

florianporada, 2018-06-12:
@tdurand, I just tested the new version (removed node_modules, reinstalled and built it). 

In the counting view, the webcam-frame is not visible. 🤔 In the path-viz-view however the frame is visible . 🙂.. I'm not sure, if the frame should be stretched or not. I think when it is displayed in the counting view it can be stretched to fit the screen-dimensions but the on the path-viz the frame has to align with the lines which get drawn. Any idea on how to tackle that? 

And is it possible to include the "counting lines", which you draw before you start the counting, in the webcamframe?

Reference.
https://imgur.com/a/xdg3eaF

tdurand, 2018-06-12:
1. On it, I will stretch but having scrollbars like the view when you define the counting lines. It's normal I didn't include the background picture on the "counting" slides... will add it

2. Hum... It is, not super straithforward to do I would need to rethink a bit how it works.. but my question is more, do we want to do that ? Because it would be a big mess with the other lines.... We could maybe think of some animation of stuff to do when we count something..... not sure 😄

tdurand, 2018-06-12:
ah, you may mean only on the counting view... ok trying something now 😉

tdurand, 2018-06-12:
Ok done ! you can check it out on master, no need to run npm install, also the canvas resolution is never higher than 1280x720 so if you open on a larger screen, it will look a bit pixelated... we could do something that scales the canvas content size to avoid this, but this would need maybe half day as everything is though to have a fixed canvas size.

[image]

florianporada, 2018-06-13:
Wow that looks quite nice! 🤩
I'll check it out as soon as I can

florianporada, 2018-06-14:
I just checked the changes and now the background image looks good. 👌There are now the scrollbars showing on the screen, though. But maybe I can fix that with `overflow: hidden` or something on the background-image element.

tdurand, 2018-06-14:
Yes the scroll bars show up when the screen ratio is different than the 1280x720 ratio.

We need to keep that in order not to change the proportions, especially to keep the app usable in mobile

tdurand, 2018-06-14:
We could stretch or add overflow hidden but this would mean to maybe miss some important data if the screen ratio isn’t right.

florianporada, 2018-06-14:
I would suggest that we leave it like that for now, then.  😁

b-g, 2018-08-14:
Closed. Added to new todo list in https://github.com/moovel/lab-opendatacam/issues/46.

closed 2018-08-14 by b-g