opendatacam/opendatacam #43: Darknet-net fails during 'make' on the Jetson TX2

florianporada, 2018-06-14:
The installation during the make build process of the `darknet-net` dependency it fails.

Quick Fix: 
Change: 
```
ARCH= -gencode arch=compute_20,code=[sm_20,sm_21] \
      -gencode arch=compute_30,code=sm_30 \
      -gencode arch=compute_35,code=sm_35 \
      -gencode arch=compute_50,code=[sm_50,compute_50] \
      -gencode arch=compute_52,code=[sm_52,compute_52]
```
to:

```
ARCH= #-gencode arch=compute_20,code=[sm_20,sm_21] \
      -gencode arch=compute_30,code=sm_30 \
      -gencode arch=compute_35,code=sm_35 \
      -gencode arch=compute_50,code=[sm_50,compute_50] \
      -gencode arch=compute_52,code=[sm_52,compute_52]
```

RoaldWikene, 2018-06-29:
hi  i dont know if it makes any difference but i added also 
-gencode=arch=compute_62,code=sm_62 \
as the Jetson TX2 is compute_62 – Drive-PX2, Tegra (Jetson) TX2, Denver-based GPU
needs CUDA 8 or higher

RoaldWikene, 2018-07-02:
this bug should now be fixed in source ref: https://github.com/meso-unimpressed/darknet-net/commit/2ce405e8ff1ed41c5c774eb5b9227b9918e5450e