opendatacam/opendatacam #44: Changing time in export.csv

smacinnes, 2018-08-07:
Everything is working wonderfully so far, so congratulations on that. I was wondering if there was something I could do so that the export.csv file displays local time or if it must stay at UTC. Can you help me out?

florianporada, 2018-08-08:
Hi @smacinnes,

thanks for the feedback 🙏.
If you want to use plain javascript you could edit this part of the code: 
https://github.com/moovel/lab-opendatacam/blob/4fe02e51b1adf728320bb7393ebbdd1b158f8bdd/server/counter/Counter.js#L108

```
const d = new Date();
const offset = 4;  /* set the offset to your desired time */ 
d.setTime(
  d.getTime() +
  d.getTimezoneOffset() * 60 * 1000 /* convert to UTC */ +
    (offset) * 60 * 60 * 1000
);

// Add it to the history (for export feature)
Counter.countedItemsHistory.push({
  date: d.toISOString(),
  area: countingAreaKey,
  name: trackedItem.name
})
```

You also could use a library like moment.js (https://momentjs.com/docs/#/manipulating/add/) and manipulate the code: 

```
// Add it to the history (for export feature)
Counter.countedItemsHistory.push({
  date: moment().add(2, 'hours').toISOString(),
  area: countingAreaKey,
  name: trackedItem.name
})
```

I hope this helps 🙂✌️

smacinnes, 2018-08-08:
Hey @florianporada!

That worked! Thanks so much :-)

closed 2018-08-08 by smacinnes