opendatacam/opendatacam #45: Evaluate upcoming Jetson Xavier for OpenDataCam

b-g, 2018-08-14:
> NVIDIA Jetson Xavier is the latest addition to the Jetson platform. It’s an AI computer for autonomous machines, delivering the performance of a GPU workstation in an embedded module under 30W. With multiple operating modes at 10W, 15W, and 30W, Jetson Xavier has greater than 10x the energy efficiency and more than 20x the performance of its predecessor, the Jetson TX2.

> The platform developer kit will be available in August for $1,299

https://developer.nvidia.com/jetson-xavier-devkit
https://developer.nvidia.com/embedded/jetson-xavier-faq

b-g, 2018-08-17:
@florianporada Thanks for the assessment! Agreed! :) We should buy one once it becomes available. I signed up for a notification at nvidia ... let's wait.

b-g, 2018-08-31:
Order! Shipping Sep 20 2018, should arrive beginning of Oct.

tdurand, 2019-01-09:
First benchmark of Jetson Xavier 👀

## Running YOLO

Our YOLO implem "mesos" fork doesn't work out of the box on Xavier, running it leads to a **Seg fault** 😐 , might be because of opencv, might be because of CUDA or any other dependency, not idea on why is failing, error doesn't give any info

After a bit of research, found out that the current "state of the art" implem of YOLO (v3) seems to be this repo: https://github.com/AlexeyAB/darknet , which has improvement other the "official" _pjreddie/darknet_ one regarding specific architecture like Jetson Xavier or other GPUs modules

So I went ahead, and after compiling and running it,  turns out performances are really good 🎉, significantly faster than the TX2, seeing a **4-10x perf boost** :

_YOLOv2 & VOC (same as current open data cam)_

| Power mode  | FPS |
| ------------- | ------------- |
| Default (15W)  | 25 FPS  |
| Max out (30W+) | 30+ FPS (webcam steam can't go higher) |

We were getting max 7 FPS with TX2

_YOLOv3 & COCO_

| Power mode  | FPS |
| ------------- | ------------- |
| Default (15W)  | 11.5 FPS  |
| Max out (30W+) | 22.5 FPS |

We were getting max 2 FPS with TX2 on YOLOv2 + COCO

## FFServer + YOLO v3

From a really quick benchmark, seems to be possible without much perf hit ( 1 FPS penalty..), see [full notes](https://docs.google.com/document/d/1oDk7Nf95Pm2eJqHsC2TL24v87CsHW8wtENbJlkBSmNE/edit?usp=sharing) for more on this

## Full notes

I wrote more notes on all the setup + perf exploration: https://docs.google.com/document/d/1oDk7Nf95Pm2eJqHsC2TL24v87CsHW8wtENbJlkBSmNE/edit#heading=h.6px293e6d645

## Further potential exploration:

- Try to run this new darknet implementation on Jetson TX2 to see if perfs improves
- Test a lighter implementation of yolov3: https://github.com/AlexeyAB/yolo2_light , claim +30% speedup for only -1% "accuracy"

## Up next 

I think this is promising as perfs are significantly better this can enable:

- a live preview of counting / detection to improve UX 
- potential better detection and tracking, more FPS + Yolov3 trained with COCO dataset should be better at detecting things and more reliable

But we won't avoid to update our darknet implementation / port the websocket feature if we want to go further with Jetson Xavier... I'm not sure of the complexity of this but I remember looking in the C code done on the mesos fork and it didn't seem much.. but things are always more complex than they seem, and I don't have much expertise on both darknet source code and C code..

closed 2019-01-29 by b-g