opendatacam/opendatacam #46: Review Aug 14

b-g, 2018-08-14:
General
- [ ] Add transition (move x) between count and path view
- [ ] Just have a single arrow (left or right) as there are just two views currently
- [ ] Change label of rec button to "Start Tracking" or "Stop Tracking"
- [ ] Style download data buttons
- [ ] Show download data buttons always (also while the tacking is running)
- [ ] Style current "Count again" button like the rec button

Path view
- [ ] Show canvas with background image full screen
- [ ] Add button "export screenshot"

rraapphhaaeell, 2018-08-15:
**Export // Data Format Tings**
I've rechecked the export after our skype yesterday and saw that it basically already has all the things needed for the counter function except for 1 thing, which is the Unique Object ID (added in example below):
- [ ]  Expert Data Output: Counter - Add Unique Vehicle ID to export: 

`csv`
```
"Timestamp", "Counter", "ObjectClass", "objectID"
"2018-06-18T16:45:30.174Z", "yellow", "person", "281"
```

The other idea of adding time-stamp to X/Y Coordinates of each tracker line wont really work I think. If their idea is to calculate speed of vehicles it wont be possible with this information. I belive the unique ID of an object is already stored in the JSON - if not this would be cool to add here (added in example below - and corrected data->timestamp). 
- [ ]  Expert Data Output Tracker: 

`JSON`
```
[
   {
      "0":{
         "id":0,
         "x":650,
         "y":413,
         "w":1110,
         "h":612,
         "bearing":290,
         "name":"person"
      },
      "objectID":"281"
      "timestamp":"2018-06-18T16:45:31.225Z"
   },
]
```
- [ ]  (Optional) Expert Data Export: If this is chosen, users will get CSV also as JSON (to have a unified data output)

tdurand, 2018-08-20:
Hey @b-g @rraapphhaaeell , just a quick update, I have been sick the past few days , slowly recovering, I will make the update this week but not as soon as I expected 😉

b-g, 2018-08-21:
Hi @tdurand, oh no! Get well soon! 🤒NP. Ping us once you are back in office.

b-g, 2018-08-24:
Hi @tdurand, many thanks for all this! Great! Feedback:

**UI**
* Find it slightly odd with the two modes in place, to force the user at the beginning to define the counter lines ... can we find a way around this without spending a lot effort on this?
* The carousel animation is nice and help to differentiate between the modes. But then I would expect the standard behaviour for the positioning of the arrow buttons. Find it odd that there is always just the right one.
* Use Title Case for the text labels: Tracking Data, Counting Data, Save Frame, Stop Tracking
* Can we make the way the button bar appear in modes more consistent? It should IMO be: 
```
Counter Mode: Counting Data, Stop Tracking
```
```
Path Mode: Tracking Data, Save Frame, Stop Tracking
```
* Will ask @mmmmmmmmmmmmmmmm to fine tune the icons once he has (re-)joined the team in September

**Data**
* Raw data structure. Would it not be nicer to have objects in every frame grouped together in an array?

```js
  // 1 Frame
  {
    "object": [...], // Detected objects of frame grouped in array
    "timestamp": "2018-08-23T08:46:59.677Z" // Time of the frame
  },
  // Other frames ...
```

**Documentation**
* Documentation of running on video file is not yet fully clear. You uncomment the stated line in YOLO.js (add a link to the file) ... but then I do what with the video?

Happy to discuss everything on Skype if helpful ... either today 1:30 - 2:45pm or have good availability next week.

tdurand, 2018-08-24:
@b-g many thanks for this quick feedback, I will be able to do the fixes on monday... After those fixes maybe it could be usefull to have a quick chat to discuss further changes (like having a direct access to the "path view mode" but I may try something first)

tdurand, 2018-08-27:
@b-g Did the little fixes you mentioned, please review:

- UI: 
- Doc on running from video: https://github.com/moovel/lab-opendatacam/tree/0.2#-run-open-data-cam-on-a-video-file-instead-of-the-webcam-feed , I tried to make it more clear, but if it is still not good enough let me know.. I think also this is more clear when you have gone threw the steps of installing open data cam on a jetson

Regarding of finding a way to run the thing in "path visualization" only without doing any counting... This is tricky because all the flow was though for the counting "mode"... The tracking view was first just thought to be a visualization and not a mode itself...

That said, we could do a quick fixe which would be to add below the "Draw to define active area" a link that says "Or run only in tracking mode without counting" which would lead the the carousel view but with only the path visualization view... 

Or maybe better and more clear, have two buttons at the beggining that says "Counting + Tracking", "Tracking Only"  .

What do you think ? Open to have a quick chat about this in the afternoon, after 3 p.m I will be around on skype

tdurand, 2018-08-27:
PS: link to updated UI: https://open-data-cam-aonsmxozth.now.sh/

b-g, 2018-08-27:
@tdurand Thanks + true!

Feedback:
* UI ✅. Markus just have to polish the icons ... then done.
* Would leave the flow as it is ... seems too much effort and not really a lot better.
* Docs:
> 2. Then open YOLO.js, and uncomment those lines ...

Could you link up YOLO.js in the markdown file? (Or where is this file located)

> 3. Copy the video file you want to run in the darknet-net folder on the Jetson (if you did auto-install, it is this path: ~/darknet-net) and replace YOUR_FILE_PATH_RELATIVE_TO_DARK_NET_FOLDER.mp4 with the name of your file.

Where/what is "YOUR_FILE_PATH_RELATIVE_TO_DARK_NET_FOLDER.mp4"? Is it a setting in a config/code file? Or a dummy video file that I overwrite? Or do I have to simply put any video file in the "~/darknet-net" folder?

PS. Can't do a skype this afternoon ... but tomorrow 8 - before 15:30 would work. But it seems all clear.

tdurand, 2018-08-27:
Ok ! Merged + improved again the README for the doc to run on a file, got the confusion, it should be clear now, let me know if you want further explanations: https://github.com/moovel/lab-opendatacam#-run-open-data-cam-on-a-video-file-instead-of-the-webcam-feed , or close this issue if good for you !

Yep skype is unnecessary for this, we'll be in touch via email with an update on non-public stuff 😉

b-g, 2018-08-27:
@tdurand Great + thanks! All good now! Closed.

closed 2018-08-27 by b-g