opendatacam/opendatacam #48: Line placement bug

smacinnes, 2018-08-17:
When viewing the tracker on different screen sizes, the lines appear in different places. It seems that this is purely visual and does not affect the counting itself. Just thought I'd report it in case you guys weren't aware. 

From my phone (where I set it up):
[image]

From my desktop:
[image]

tdurand, 2018-08-22:
Hey, thanks for you comment, we are currently working on an update that will improve the responsiveness of the UI.

tdurand, 2018-08-23:
fixed on branch 0.2 : https://github.com/moovel/lab-opendatacam/commit/ba2768b792600e66005d43d9fe3d3576cc3a5c97

closed 2018-08-23 by tdurand