opendatacam/opendatacam #50: v0.2

b-g, 2018-08-27:
- [ ] polish icons @mmmmmmmmmmmmmmmm 
- [ ] introduce changelog @tdurand 
- [ ] bump version in package.json shortly before release
- [ ] create release on github

b-g, 2018-08-27:
@tdurand It seems we forgot to do v0.1 release ... could you tag the latest commit before the path mode was introduced with v0.1?

tdurand, 2018-08-28:
Done @b-g , I've wrote the changelog directly on the release page: https://github.com/moovel/lab-opendatacam/releases

b-g, 2018-08-28:
@tdurand Thanks!!

tdurand, 2019-01-09:
Ok, saw that @mmmmmmmmmmmmmmmm had pushed the new icons, the path thing was already working properly. Can close, the latest release is v0.2.2, published here: https://github.com/moovel/lab-opendatacam/releases/tag/v0.2.2

closed 2019-01-09 by tdurand