opendatacam/opendatacam #55: Show Video Screenshot/Thumbnail when running on a video file

yannikw93, 2018-11-02:
When running the Open Data Cam on a file would it be possible to show a video screenshot. It would really help with drawing the lines.

Currently I can only see either the webcam feed or screenshots from previous sessions with the webcam.

tdurand, 2019-01-29:
Next version will have a live preview

closed 2019-01-29 by tdurand

gabbyhan, 2019-04-29:
Hi,

I am having the same issue, except that when I try to run Open Data Cam on a file I only see a black screen so I have no way of knowing where to draw the lines. Also when I do draw a line, it seems to hang on "initializing neural network" forever. Any suggested fixes?

Thanks!

tdurand, 2019-04-29:
Hi @gabbyhan , thanks for you message.

We will release the v2 version of opendatacam , if you are using a Jetson TX2 or Nano you should already be able to install the 2.0.0-beta.1 , look at the https://github.com/moovel/lab-opendatacam/tree/v2#-get-started-quick-setup For docker setup for v2

Regarding your issue on v1 it seems that opendatacam has no access to the webcam, is is plugged in to your jetson via usb ?

gabbyhan, 2019-04-29:
Hi,

Sorry for not specifying. I meant I have the same issue when I am trying to run the program on an existing video file.

tdurand, 2019-04-29:
It's likely it does not succeed to read the file, could you post the stout log of the node process so I can have more info from the error ?

gabbyhan, 2019-04-29:
Sorry for my ignorance, but how can I access that?

tdurand, 2019-04-29:
If you used pm2 to start the node app, you should be able to access the logs with this command: `pm2 logs`

Or Instead of starting the app with pm2 you should start it with `npm run start` so you will see the logs in the terminal

gabbyhan, 2019-04-29:
When I start with `npm run start` this is the output:

```
> open-data-cam@0.2.3 start /home/nvidia/libwsclient/liblo-0.29/json-c/darknet-net/lab-open-data-cam
> PORT=8080 NODE_ENV=production node server.js

Please specify the path to the raw detections file
Process YOLO initialized
Process WebcamStream initialized
events.js:183
      throw er; // Unhandled 'error' event
      ^

Error: listen EADDRINUSE :::8080
    at Server.setupListenHandle [as _listen2] (net.js:1360:14)
    at listenInCluster (net.js:1401:12)
    at Server.listen (net.js:1485:7)
    at app.prepare.then (/home/nvidia/libwsclient/liblo-0.29/json-c/darknet-net/lab-open-data-cam/server.js:121:10)
    at <anonymous>
    at process._tickCallback (internal/process/next_tick.js:189:7)
    at Function.Module.runMain (module.js:696:11)
    at startup (bootstrap_node.js:204:16)
    at bootstrap_node.js:625:3
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! open-data-cam@0.2.3 start: `PORT=8080 NODE_ENV=production node server.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the open-data-cam@0.2.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

```

tdurand, 2019-04-29:
You need first to stop the instance running with the pm2 process manager:

With something like this

```
sudo pm2 list
sudo pm2 stop <IDOFPROCESS>

# Then npm run start 
```

gabbyhan, 2019-04-29:
Thanks! It worked until I went to localhost:8080 on the browser. This is the last couple of lines that were output: 

```
[video4linux2,v4l2 @ 0x47e970] Cannot open video device /dev/video1: No such file or directory
/dev/video1: No such file or directory
Mon Apr 29 16:03:03 2019 Could not open output feed file './feed1.ffm'

```
I have double checked the file path many times. My video's file path is ~/libwsclient/liblo-0.29/json-c/darknet-net/Hamburg_Hall.mp4
The path that I put in "PATH_TO_YOLO_DARKNET" is "/home/nvidia/libswclient/liblo-0.29/json-c/darknet-net"
In YOLO.js (server/processes), I put the file name as "Hamburg_Hall.mp4"

gabbyhan, 2019-04-29:
I believe I forgot to comment out the lines of code for running on a webcam in YOLO.js

gabbyhan, 2019-04-29:
I commented out the lines of code and I still get:

> open-data-cam@0.2.3 start /home/nvidia/libwsclient/liblo-0.29/json-c/darknet-net/lab-open-data-cam
> PORT=8080 NODE_ENV=production node server.js

Please specify the path to the raw detections file
Process YOLO initialized
Process WebcamStream initialized
> Ready on http://localhost:8080
> Ready on http://128.2.70.240:8080
WebcamStream: Sub-Process Stream webcam started
WebcamStream: Sub-Process FFServer started

.....................
[video4linux2,v4l2 @ 0x47e970] Cannot open video device /dev/video1: No such file or directory
/dev/video1: No such file or directory

 ....................
ffserver.conf:42: Setting default value for video buffer size = 128000. Use NoDefaults to disable it.
Mon Apr 29 16:11:28 2019 Could not open output feed file './feed1.ffm'

gabbyhan, 2019-04-29:
The instructions only say to uncomment the lines for running the program on an existing file (which I had already done). However, in the comments, it says to also comment the lines for running on a webcam, so I just did that as well.

gabbyhan, 2019-04-29:
I just checked and I have ffmpeg version 2.8.15 installed so I don't know why I am getting the` please specify the path to the raw detections file`

tdurand, 2019-04-29:
Could you copy paste your all the YOLO.js file code here ?

gabbyhan, 2019-04-29:
```
const forever = require('forever-monitor');
const config = require('../../config.json');
const simulation8FPSDetectionsData = require('../../static/placeholder/rawdetections8FPS.json');
const WebSocketClient = require('websocket').client;
const fs = require('fs');

let YOLO = {
  isRunning: false,
  isInitialized: false,
  process: null,
  simulationMode: false,
  simulationInterval: null
};

module.exports = {
  init: function(simulationMode) {

    YOLO.simulationMode = simulationMode;

    // On webcam
    // ./darknet detector demo cfg/voc.data cfg/yolo-voc.cfg yolo-voc.weights -c 1 -address "ws://localhost" -port 8080
    // Comment the following lines to run on a file directly
//    YOLO.process = new (forever.Monitor)(['./darknet','detector','demo','cfg/voc.data','cfg/yolo-voc.cfg','yolo-voc.weights','-c','1', '-address','ws://localhost','-port','8080'],{
  //    max: 1,
  //    cwd: config.PATH_TO_YOLO_DARKNET,
  //    killTree: true
  //  });

    // On file
    // ./darknet detector demo cfg/voc.data cfg/yolo-voc.cfg yolo-voc.weights -filename YOUR_FILE_PATH_RELATIVE_TO_DARK_NET_FOLDER.mp4 -address "ws://localhost" -port 8080
    // YOUR_FILE_PATH_RELATIVE_TO_DARK_NET_FOLDER.mp4 -> relative to darknet directory (set up in config.json), if outside this directory do:
    // ../file.mp4 or ../videos/file.mp4

    // Uncomment the following lines to run on a file directly

    YOLO.process = new (forever.Monitor)(['./darknet','detector','demo','cfg/voc.data','cfg/yolo-voc.cfg','yolo-voc.weights','-filename', 'Hamburg_Hall.mp4', '-address','ws://localhost','-port','8080'],{
      max: 1,
      cwd: config.PATH_TO_YOLO_DARKNET,
      killTree: true
    });

    YOLO.process.on("start", () => {
      console.log('Process YOLO started');
      YOLO.isRunning = true;
    });

    YOLO.process.on("stop", () => {
      console.log('Process YOLO stopped');
      YOLO.isRunning = false;
    });
   console.log('Process YOLO initialized');
    YOLO.isInitialized = true;

    // TODO handle other kind of events
    // https://github.com/foreverjs/forever-monitor#events-available-when-using-an-instance-of-forever-in-nodejs
  },

  startYOLOSimulation: function() {
    let client = new WebSocketClient();

    client.on('connectFailed', function(error) {
        console.log('Connect Error: ' + error.toString());
    });

    client.on('connect', function(connection) {
        console.log('WebSocket Client Connected');
        connection.on('error', function(error) {
            console.log("Connection Error: " + error.toString());
        });
        connection.on('close', function() {
            console.log('echo-protocol Connection Closed');
        });

        let detections = simulation8FPSDetectionsData;
        let frameNb = 0;

        // Simulate YOLO 15s booting time
        setTimeout(() => {
          // Simulate a 8 FPS detections
          YOLO.simulationInterval = setInterval(sendDetection, 125)
        }, 2000)

        function sendDetection() {
            if (connection.connected) {
                // Convert detection coordinates to float
                // 1920 => 1
                // detection.x, detection.w => ?
                // 1080 => 1
                // detection.y, detection.h => ?
                let detection = detections[frameNb].map((detection) => {
                 return {
                    ...detection,
                    class: detection.name,
                    x: detection.x / 1920,
                    y: detection.y / 1080,
                    w: detection.w / 1920,
                    h: detection.h / 1080
                  }
                })
                connection.sendUTF(JSON.stringify(detection));

                if(detections[frameNb + 1]) {
                  frameNb++;
                } else {
                  console.log('Reset simulation counter')
                  // Infinite
                  frameNb = 0;
                  // clearInterval(YOLO.simulationInterval)
                }
            }
        }

    });

    client.connect('ws://localhost:8080/');
  },

  start: function() {
    if(YOLO.simulationMode) {
      this.startYOLOSimulation();
    } else {
      if(!YOLO.isRunning) {
        YOLO.process.start();
      }
    }
  },

  stop: function() {
    if(YOLO.simulationMode && YOLO.simulationInterval) {
      clearInterval(YOLO.simulationInterval)
    } else {
      if(YOLO.isRunning) {
        YOLO.process.stop();
      }
    }
  }
}

```

tdurand, 2019-04-29:
You configuration seem correct, did you run `npm run build` after changing the code ? You need to recompile before doing `npm run start`

Also maybe try to restart your jetson because ffserver leave some ghost behind when failing multiple time

If it still doesn't work, send me the actual error it gives you.

gabbyhan, 2019-04-29:
I had already recompiled with npm run build but I just restarted the Jetson, ran `npm run build` again, and had to stop the instance running again with `sudo pm2 stop ID`, then finally ran `npm run start`

It still gave me this error:

```
> open-data-cam@0.2.3 start /home/nvidia/libwsclient/liblo-0.29/json-c/darknet-net/lab-open-data-cam
> PORT=8080 NODE_ENV=production node server.js

Please specify the path to the raw detections file
Process YOLO initialized
Process WebcamStream initialized
> Ready on http://localhost:8080
> Ready on http://128.2.70.240:8080
WebcamStream: Sub-Process Stream webcam started
WebcamStream: Sub-Process FFServer started
ffmpeg version 2.8.15-0ubuntu0.16.04.1 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.10) 20160609
  configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
[video4linux2,v4l2 @ 0x47e970] Cannot open video device /dev/video1: No such file or directory
/dev/video1: No such file or directory
ffserver version 2.8.15-0ubuntu0.16.04.1 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.10) 20160609
  configuration: --prefix=/usr --extra-version=0ubuntu0.16.04.1 --build-suffix=-ffmpeg --toolchain=hardened --libdir=/usr/lib/aarch64-linux-gnu --incdir=/usr/include/aarch64-linux-gnu --cc=cc --cxx=g++ --enable-gpl --enable-shared --disable-stripping --disable-decoder=libopenjpeg --disable-decoder=libschroedinger --enable-avresample --enable-avisynth --enable-gnutls --enable-ladspa --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libgme --enable-libgsm --enable-libmodplug --enable-libmp3lame --enable-libopenjpeg --enable-libopus --enable-libpulse --enable-librtmp --enable-libschroedinger --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libssh --enable-libtheora --enable-libtwolame --enable-libvorbis --enable-libvpx --enable-libwavpack --enable-libwebp --enable-libx265 --enable-libxvid --enable-libzvbi --enable-openal --enable-opengl --enable-x11grab --enable-libdc1394 --enable-libiec61883 --enable-libzmq --enable-frei0r --enable-libx264 --enable-libopencv
  libavutil      54. 31.100 / 54. 31.100
  libavcodec     56. 60.100 / 56. 60.100
  libavformat    56. 40.101 / 56. 40.101
  libavdevice    56.  4.100 / 56.  4.100
  libavfilter     5. 40.101 /  5. 40.101
  libavresample   2.  1.  0 /  2.  1.  0
  libswscale      3.  1.101 /  3.  1.101
  libswresample   1.  2.101 /  1.  2.101
  libpostproc    53.  3.100 / 53.  3.100
ffserver.conf:42: Setting default value for video bit rate = 64000. Use NoDefaults to disable it.
ffserver.conf:42: Setting default value for video bit rate tolerance = 32000. Use NoDefaults to disable it.
ffserver.conf:42: Setting default value for video rate control equation = tex^qComp. Use NoDefaults to disable it.
ffserver.conf:42: Setting default value for video max rate = 128000. Use NoDefaults to disable it.
ffserver.conf:42: Setting default value for video buffer size = 128000. Use NoDefaults to disable it.
Mon Apr 29 16:54:45 2019 Could not open output feed file './feed1.ffm'
```

tdurand, 2019-04-29:
That is weird seems ffmpeg still wants to open the webcam, but please also try to install ffmpeg v3: see in the readme

```
sudo add-apt-repository ppa:jonathonf/ffmpeg-3
# sudo add-apt-repository ppa:jonathonf/tesseract (ubuntu 14.04 only!!)
sudo apt update && sudo apt upgrade
sudo apt-get install ffmpeg
```

gabbyhan, 2019-04-29:
I installed it again and I get the same error. I checked my ffmpeg version again and it is version 2.8.15

tdurand, 2019-04-29:
I think the issue comes from the fact you have not ffmpeg v3  installed, I'm not sure how to help you further, maybe you need to uninstall ffmpeg first, here is the complete article on how to install it: http://ubuntuhandbook.org/index.php/2017/05/install-ffmpeg-3-3-in-ubuntu-16-04-14-04/

And maybe even try ffmpeg v4 ( we didn't test with it though) : https://tecadmin.net/install-ffmpeg-on-linux/

tdurand, 2019-04-29:
also: https://blog.programster.org/ubuntu-16-04-install-ffmpeg-from-ppa

gabbyhan, 2019-05-06:
Hi,

I have spent the past week working with someone experienced in linux and posting on ffmpeg and ubuntu forums to try to get version 3 of ffmpeg but I have so far been unsuccessful. Is there any way I can get Open Data Cam to work with my older version off ffmpeg?

Thank you!

tdurand, 2019-05-06:
Infortunately not, but Open data cam v2.0 won't require ffmpeg. You can already try it we release the beta.1 : https://github.com/moovel/lab-opendatacam/tree/v2#-get-started-quick-setup

There is not documentation to run from a file yet though.. But in 2 weeks we will have a way to do it with v2.0

gabbyhan, 2019-05-06:
Since I already have everything flashed from version 1, can I just update to Ubuntu 18.04? I have all my packages installed from my Jetson TX2 3.2.

tdurand, 2019-05-07:
No, you need to flash it again to jetpack 4.2 , because we need the new versions of Nvidia software like CUDA , etc etc

tdurand, 2019-05-07:
BTW with v2 there is just 1 installation step after flashing to 4.2, we run from a docker image... Or you can run it without docker with a simpler setup : https://github.com/moovel/lab-opendatacam/blob/v2/doc/RUN_WITHOUT_DOCKER.md

gabbyhan, 2019-05-08:
Hi,

So I was able to manually install ffmpeg version 4.1.1 but I now get this error when I try to launch localhost:8080 from my browser after `npm run start`:

```
> open-data-cam@0.2.3 start /home/nvidia/libwsclient/liblo-0.29/json-c/darknet-net/lab-open-data-cam
> PORT=8080 NODE_ENV=production node server.js

Please specify the path to the raw detections file
Process YOLO initialized
Process WebcamStream initialized
> Ready on http://localhost:8080
> Ready on http://128.2.70.240:8080
WebcamStream: Sub-Process Stream webcam started
events.js:183
      throw er; // Unhandled 'error' event
      ^

Error: spawn ffserver ENOENT
    at Process.ChildProcess._handle.onexit (internal/child_process.js:190:19)
    at onErrorNT (internal/child_process.js:362:16)
    at _combinedTickCallback (internal/process/next_tick.js:139:11)
    at process._tickCallback (internal/process/next_tick.js:181:9)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! open-data-cam@0.2.3 start: `PORT=8080 NODE_ENV=production node server.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the open-data-cam@0.2.3 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

```

tdurand, 2019-05-08:
It might be, didn't test on ffmpeg v4. opendatacam isn't able to launch it... Maybe some ghost is laying around, try to restart your jetson.

I really advise you to install the opendatacam v2 beta version as it brings much more improvements (speed, UX, features...), and we get rid of ffmpeg dependency.

gabbyhan, 2019-05-08:
That sounds great. I just was hoping to run Open Data Cam on a file before the week ends and if there isn't yet support for that in v2.0 I didn't want to go through the whole process of flashing the jetson again.

tdurand, 2019-05-08:
for now it is supported, but not documented, I will put out a quick documentation for you by the end of the week

gabbyhan, 2019-05-08:
Hi,

Thank you so much for all the support. I just wanted to try one more time as through the manual installation process I am able to get ffmpeg version 3.4.2. I now no longer get an error but it still is trying to connect to a webcam I believe:

```
> open-data-cam@0.2.3 start /home/nvidia/libwsclient/liblo-0.29/json-c/darknet-net/lab-open-data-cam
> PORT=8080 NODE_ENV=production node server.js

Please specify the path to the raw detections file
Process YOLO initialized
Process WebcamStream initialized
> Ready on http://localhost:8080
> Ready on http://128.2.70.240:8080
WebcamStream: Sub-Process Stream webcam started
WebcamStream: Sub-Process FFServer started
ffmpeg version 3.4.2 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.11) 20160609
  configuration: 
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
[video4linux2,v4l2 @ 0x19cb3b0] Cannot open video device /dev/video1: No such file or directory
/dev/video1: No such file or directory
ffserver version 3.4.2 Copyright (c) 2000-2018 the FFmpeg developers
  built with gcc 5.4.0 (Ubuntu/Linaro 5.4.0-6ubuntu1~16.04.11) 20160609
  configuration: 
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
ffserver.conf:42: Setting default value for video bit rate = 64000. Use NoDefaults to disable it.
ffserver.conf:42: Setting default value for video bit rate tolerance = 32000. Use NoDefaults to disable it.
ffserver.conf:42: Setting default value for video rate control equation = tex^qComp. Use NoDefaults to disable it.
ffserver.conf:42: Setting default value for video max rate = 128000. Use NoDefaults to disable it.
ffserver.conf:42: Setting default value for video buffer size = 128000. Use NoDefaults to disable it.
Wed May  8 17:26:57 2019 Codecs do not match for stream 0
Wed May  8 17:26:57 2019 [ffm @ 0x182fe50]Using AVStream.codec to pass codec parameters to muxers is deprecated, use AVStream.codecpar instead.
Wed May  8 17:26:57 2019 FFserver started.
Wed May  8 17:27:01 2019 127.0.0.1 - - [GET] "/webcam.jpg?1557336420088 HTTP/1.1" 200 63
Wed May  8 17:27:02 2019 127.0.0.1 - - [GET] "/webcam.jpg?1557336421088 HTTP/1.1" 200 63
Wed May  8 17:27:03 2019 127.0.0.1 - - [GET] "/webcam.jpg?1557336422088 HTTP/1.1" 200 63
Wed May  8 17:27:04 2019 127.0.0.1 - - [GET] "/webcam.jpg?1557336423088 HTTP/1.1" 200 63
Wed May  8 17:27:05 2019 127.0.0.1 - - [GET] "/webcam.jpg?1557336424088 HTTP/1.1" 200 63
^C
```

I just wanted to let you know since I have followed all the instructions for running on an existing video file, as you know, but I still can't seem to get the video to show up. If even with ffmpeg v3 the issue cannot be resolved I suppose I have no choice but to try v2.0. 

Again, thank you so much for all the support and advice.

tdurand, 2019-05-08:
This error means it tries to pickup the webcam that isn't connected (as you run from a file)

Could you make sure the YOLO.js file contain the changes and that you have did a npm run build afterwise ?

gabbyhan, 2019-05-08:
Yes, I checked my YOLO.js and I have not changed it since I copied it above. I also restarted, stopped the process, and then ran npm run build again before starting.

tdurand, 2019-05-08:
Mmmm ... It shouldn't pick up the webcam if this line is commented out... Could you double check (sorry but I don't see any other potential problems) ?

tdurand, 2019-05-08:
If you go the darknet directory and run

./darknet detector demo cfg/voc.data cfg/yolo-voc.cfg yolo-voc.weights -filename Hamburg_Hall.mp4.mp4 -address "ws://localhost" -port 8080

What does happen ?

gabbyhan, 2019-05-08:
It kept running but I stopped with ^C:

```
videoPath:./output.avi, namedPipe:0, writeVideo:0, showWindow:0
using comm protocol ws
address:ws://localhost, port:8080 
setting up ws
ws onerror: (-2): Could connect to any address returned by getaddrinfo
comm setup done
Demo
layer     filters    size              input                output
    0 conv     32  3 x 3 / 1   416 x 416 x   3   ->   416 x 416 x  32
    1 max          2 x 2 / 2   416 x 416 x  32   ->   208 x 208 x  32
    2 conv     64  3 x 3 / 1   208 x 208 x  32   ->   208 x 208 x  64
    3 max          2 x 2 / 2   208 x 208 x  64   ->   104 x 104 x  64
    4 conv    128  3 x 3 / 1   104 x 104 x  64   ->   104 x 104 x 128
    5 conv     64  1 x 1 / 1   104 x 104 x 128   ->   104 x 104 x  64
    6 conv    128  3 x 3 / 1   104 x 104 x  64   ->   104 x 104 x 128
    7 max          2 x 2 / 2   104 x 104 x 128   ->    52 x  52 x 128
    8 conv    256  3 x 3 / 1    52 x  52 x 128   ->    52 x  52 x 256
    9 conv    128  1 x 1 / 1    52 x  52 x 256   ->    52 x  52 x 128
   10 conv    256  3 x 3 / 1    52 x  52 x 128   ->    52 x  52 x 256
   11 max          2 x 2 / 2    52 x  52 x 256   ->    26 x  26 x 256
   12 conv    512  3 x 3 / 1    26 x  26 x 256   ->    26 x  26 x 512
   13 conv    256  1 x 1 / 1    26 x  26 x 512   ->    26 x  26 x 256
   14 conv    512  3 x 3 / 1    26 x  26 x 256   ->    26 x  26 x 512
   15 conv    256  1 x 1 / 1    26 x  26 x 512   ->    26 x  26 x 256
   16 conv    512  3 x 3 / 1    26 x  26 x 256   ->    26 x  26 x 512
   17 max          2 x 2 / 2    26 x  26 x 512   ->    13 x  13 x 512
   18 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024
   19 conv    512  1 x 1 / 1    13 x  13 x1024   ->    13 x  13 x 512
   20 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024
   21 conv    512  1 x 1 / 1    13 x  13 x1024   ->    13 x  13 x 512
   22 conv   1024  3 x 3 / 1    13 x  13 x 512   ->    13 x  13 x1024
   23 conv   1024  3 x 3 / 1    13 x  13 x1024   ->    13 x  13 x1024
   24 conv   1024  3 x 3 / 1    13 x  13 x1024   ->    13 x  13 x1024
   25 route  16
   26 conv     64  1 x 1 / 1    26 x  26 x 512   ->    26 x  26 x  64
   27 reorg              / 2    26 x  26 x  64   ->    13 x  13 x 256
   28 route  27 24
   29 conv   1024  3 x 3 / 1    13 x  13 x1280   ->    13 x  13 x1024
   30 conv    125  1 x 1 / 1    13 x  13 x1024   ->    13 x  13 x 125
   31 detection
mask_scale: Using default '1.000000'
Loading weights from yolo-voc.weights...Done!
video file: Hamburg_Hall.mp4

FPS:9098.3
Objects:

1280 x 720
ws onerror: (-12): Error during send in libwsclient_send

FPS:6.4
Objects:

1280 x 720
ws onerror: (-12): Error during send in libwsclient_send

FPS:6.2
Objects:

1280 x 720
car: 36%
ws onerror: (-12): Error during send in libwsclient_send

FPS:6.2
Objects:

1280 x 720
car: 36%
ws onerror: (-12): Error during send in libwsclient_send

FPS:6.2
Objects:

1280 x 720
car: 24%
car: 37%
ws onerror: (-12): Error during send in libwsclient_send

FPS:6.2
Objects:

1280 x 720
car: 24%
car: 38%
ws onerror: (-12): Error during send in libwsclient_send

FPS:6.3
Objects:

1280 x 720
car: 39%
ws onerror: (-12): Error during send in libwsclient_send

```

gabbyhan, 2019-05-08:
I checked my path in config.json and it is the absolute path to darknet:

```
{
  "PATH_TO_YOLO_DARKNET" : "/home/nvidia/libswclient/liblo-0.29/json-c/darknet-net",
  "VALID_CLASSES": ["*"]
}
```

tdurand, 2019-05-08:
This means that darknet (the neural network) is working on the file.

This also means that this code from YOLO.js should be working

```
YOLO.process = new (forever.Monitor)(['./darknet','detector','demo','cfg/voc.data','cfg/yolo-voc.cfg','yolo-voc.weights','-filename', 'Hamburg_Hall.mp4', '-address','ws://localhost','-port','8080'],{
      max: 1,
      cwd: config.PATH_TO_YOLO_DARKNET,
      killTree: true
    });
```

The only problem I can see if that the node app isn't running this code. Please try to replace your YOLO.js with this, and stop the node process,  "npm run build" , "npm run start" (without restarting the jetson):

```javascript
const forever = require('forever-monitor');
const config = require('../../config.json');
const simulation8FPSDetectionsData = require('../../static/placeholder/rawdetections8FPS.json');
const WebSocketClient = require('websocket').client;
const fs = require('fs');

let YOLO = {
  isRunning: false,
  isInitialized: false,
  process: null,
  simulationMode: false,
  simulationInterval: null
};

module.exports = {
  init: function(simulationMode) {

    YOLO.simulationMode = false;

    YOLO.process = new (forever.Monitor)(['./darknet','detector','demo','cfg/voc.data','cfg/yolo-voc.cfg','yolo-voc.weights','-filename', 'Hamburg_Hall.mp4', '-address','ws://localhost','-port','8080'],{
      max: 1,
      cwd: config.PATH_TO_YOLO_DARKNET,
      killTree: true
    });

    YOLO.process.on("start", () => {
      console.log('Process YOLO started');
      YOLO.isRunning = true;
    });

    YOLO.process.on("stop", () => {
      console.log('Process YOLO stopped');
      YOLO.isRunning = false;
    });

    console.log('Process YOLO initialized');
    YOLO.isInitialized = true;

    // TODO handle other kind of events
    // https://github.com/foreverjs/forever-monitor#events-available-when-using-an-instance-of-forever-in-nodejs
  },
  startYOLOSimulation: function() {
    let client = new WebSocketClient();

    client.on('connectFailed', function(error) {
        console.log('Connect Error: ' + error.toString());
    });

    client.on('connect', function(connection) {
        console.log('WebSocket Client Connected');
        connection.on('error', function(error) {
            console.log("Connection Error: " + error.toString());
        });
        connection.on('close', function() {
            console.log('echo-protocol Connection Closed');
        });

        let detections = simulation8FPSDetectionsData;
        let frameNb = 0;

        // Simulate YOLO 15s booting time
        setTimeout(() => {
          // Simulate a 8 FPS detections
          YOLO.simulationInterval = setInterval(sendDetection, 125)
        }, 2000)

        function sendDetection() {
            if (connection.connected) {
                // Convert detection coordinates to float
                // 1920 => 1
                // detection.x, detection.w => ?
                // 1080 => 1
                // detection.y, detection.h => ?
                let detection = detections[frameNb].map((detection) => {
                  return {
                    ...detection,
                    class: detection.name,
                    x: detection.x / 1920,
                    y: detection.y / 1080,
                    w: detection.w / 1920,
                    h: detection.h / 1080
                  }
                })
                connection.sendUTF(JSON.stringify(detection));

                if(detections[frameNb + 1]) {
                  frameNb++;
                } else {
                  console.log('Reset simulation counter')
                  // Infinite
                  frameNb = 0;
                  // clearInterval(YOLO.simulationInterval)
                }
            }
        }

    });
    client.connect('ws://localhost:8080/');
  },

  start: function() {
    if(YOLO.simulationMode) {
      this.startYOLOSimulation();
    } else {
      if(!YOLO.isRunning) {
        YOLO.process.start();
      }
    }
  },

  stop: function() {
    if(YOLO.simulationMode && YOLO.simulationInterval) {
      clearInterval(YOLO.simulationInterval)
    } else {
      if(YOLO.isRunning) {
        YOLO.process.stop();
      }
    }
  }
}
```

gabbyhan, 2019-05-08:
I replaced the code, the process was already stopped, did npm run build and npm run start. I still get the same output as before unfortunately.

tdurand, 2019-05-08:
Do you have a webcam to connect to the jetson ? Just to see if it works with a webcam

tdurand, 2019-05-08:
I'm afraid  I can't help you much more than this... Running from a file was very experimental in v1 and it might be a different ffmpeg version... not sure bugs can come from everywhere.... Also I can't invest time in reproducing the problem as we are commiting all our effort in releasing the v2 _(besides all my jetson have the v2 dependencies)_

I think your best shot is to flash the jetson with a clean jetpack 4.2 and run v2 , I will be able to help you  much more to run from a file with v2. (you will need for now to edit the YOLO.js also.. not much more work)

gabbyhan, 2019-05-08:
Okay, that works for me. Sorry for taking your time. I just wanted to make sure I tried as well as I could with this version. I will get started on the process of running v2!

tdurand, 2019-05-08:
no worry ! You will see v2 setup is less painful

Please follow this guide **after flashing to jetpack 4.2** :  https://github.com/moovel/lab-opendatacam/blob/v2/doc/RUN_WITHOUT_DOCKER.md , as with the docker setup you can't run from a file for now.

I'll make sure to put a little doc for you on how to run with a file

gabbyhan, 2019-05-11:
Hi,

I set up v2. I was just wondering where I could find the file to run from a video file?

Thanks!

tdurand, 2019-05-13:
On Saturday I released the support to run from a file with v2.0.0.beta-2, you can follow the quick startup guide:

https://github.com/moovel/lab-opendatacam/tree/v2#2-install-and-start-opendatacam-3-min-

And then read how to run from a file

https://github.com/moovel/lab-opendatacam/blob/v2.0.0-beta.2/doc/CONFIG.md#run-opendatacam-on-a-video-file-instead-of-the-webcam

gabbyhan, 2019-05-14:
Thanks so much it is finally working! However, I can't seem to find the export.csv. It is not in any of the directories. Is a new one supposed to show up every time I stop recording? Thanks!

tdurand, 2019-05-14:
In the data tab you can download JSON data, I'm still working on the csv exports for v2, you can follow progress here:  https://github.com/moovel/lab-opendatacam/issues/70 . I expect to have it working by the end of the week and release new version.

gabbyhan, 2019-05-14:
Thanks. Where is the data tab?

tdurand, 2019-05-14:
You should have some buttons on top (Counter, LiveView .... ) and there is
one named "Data"