opendatacam/opendatacam #57: Including a waitbar other than "Initializing Neural Network"

yannikw93, 2018-11-02:
I found it a bit difficult to figure out what is happening in the tool with just the info "Initializing Neural Network" after pushing the "Start Tracking Button".
A lot of times it gets stuck on that part (it just continues "loading" forever) and it's impossible to know whether or not it will start the tracking soon or not. It can be a bit frustratinbg not knowing where it gets stuck in that part. Is there an error message handler or a different kind of waitbar (with percentages) you can think of?

Maybe if it always got stuck at say 75% of the initialization then the user would know that there must be an issue, because usually it would go past that point really quick when everything is set up correctly.

tdurand, 2019-01-09:
Hello @yannikw93 , thanks for your feedback, yep it's true the UI doesn't give much feedback other than being loading forever.. For now as a general rule if it is still stuck after 45 second you can consider that there was an error and you need to restart. 

Good news is that we are investing more time on this project, doing some bug fixes and hopefully improve the reliability of the project, so we'll try to include better error handling on a future version.

tdurand, 2019-01-29:
Next version will have a console that will give more feedback on potential bugs

closed 2019-01-29 by tdurand