opendatacam/opendatacam #59: Upcoming Opendatacam update

tdurand, 2019-01-17:
I have a first working version on branch 0.3. cc @b-g 

📹 **Here is a demo video:** [https://youtu.be/dc5u0xeTNT4](https://youtu.be/dc5u0xeTNT4) 📹 

Working on Jetson TX2 with VOC weight and Yolov2, with this implementation of YOLO: https://github.com/AlexeyAB/darknet

[image]

You can see it runs at 10 FPS 🎉 with webcam stream

### Current limitations

1. Webcam resolution needs to be set to a 16/9 ratio  in order to have the front-end displaying correctly.. By default OpenCV will pick 640x480 (4/3 ratio). Here is how we do it: we need to modify two line of C++ in the YOLO implem and compile (this is the only change we do need), see more here: https://github.com/AlexeyAB/darknet/issues/2228 

   We should also fix the front-end to be able to handle any video ratio, right now everything is hardcoded to have a canvas to draw - video display - counting logic to 16/9... 

   We didn't run into this issue with the previous version because we were streaming the webcam with ffserver which was streaming a 16/9 ratio... but I suspect we were counting with the detections of YOLO with the wrong ratio and maybe had some accuracy issue while counting because of this... _we should have the same ratio when drawing the counting lines and what we feed to YOLO input._

2.  The live stream multipart HTTP connection can hangs if network isn't fast enough, when I did test first I was connecting the jetson tx2 to the router via Wifi and connecting the client (my computer) via wifi too, and was a bit far away from the router..  And this caused the stream to be max 864x480 resolution to avoid most of the issues.

    When I did test on jetson xavier which is connected via ethernet cable I didn't have the issue.. And my computer was still connected via wifi so I guess on our default setup we shouldn't have issue but I didn't try on hotspot mode on TX2 yet as real world as I did flashed the TX2 and it's not configured for this yet.

   Anyway, this is something I should watch for.. but don't worry too much because with anyway I've set a timer that restore the HTTP MJPEG connection every 5s... so in the worst case it will hang only 5s... There is no way to exactly know when it hangs infortunately.

### Up next

- **Discuss UX with live stream:**

   Raw detection from YOLO on livestream, maybe remove and add tracker ones? only when counting ? which style? Maybe just have the path visualization on the raw stream ? 

   We now also have the possibility to do more fancy stuff like animation when something cross a lines etc etc ...

   When we have the live stream + the counters on top, UI is pretty cluttered... 

  Is the path visualization usefull now ? 
    
- Investigate Docker image for easy install

  Making the mesos darknet fork obsolete save us 3-4 dependencies to install (not much)

- Refactor front-end code when we have the new flow defined.. Upgrade npm package version... Make it more robust to errors and edges cases, make it more easy to run from file ... Add a debug mode ?

- Update technical and setup doc (wait for the docker investigation)

### Random idea

- Beeing able to plug opendatacam on any webcam stream source ? And deploy a version on Amazon so the https://opendatacam.moovellab.com/ has an interactive demo ?  I imagine people beeing able to count stuff from any public cams... For example from the timesquare one https://www.earthcam.com/cams/newyork/timessquare/?cam=tsrobo3 ...

syedmustafa54, 2019-01-17:
There should be a heatmap like hot zone Traffic going on this side has hot zone something like that i can help with python but i am new to node. 
And path visualization  is very useful. But its fading after some time can we disable that fading the lines should never fadeout we can as option in the webpage. Drawing path some thing like this https://lh5.googleusercontent.com/proxy/UKP4VOLsSxHM7Rm3fB-5tE9hxmaTks-yFcccOKHH85JDaHz3sOBI8PQsoczYJApDxScZXJH2dqjDQCS7IIrZGMpSRMOe21L6Dm-vxiARY19rIi2VwfB5aUuXzeLCJACfu6refpmCH2s0odEHyfU=w496-h296-nc

syedmustafa54, 2019-01-17:
How can i test 0.3 i have a 0.2 version if i just clone 0.3 will it work if possible can you add small doc how to install and use 0.3

tdurand, 2019-01-17:
Thanks for your inputs , I’ll add a doc next week for now it’s really
alpha. Follow this thread for updates ;-)

tdurand, 2019-01-18:
@b-g :  Some more material for the UI/UX rethinking

Here is a more HD demo of the live stream prototype (sorry should have shut down the sound): https://my.pcloud.com/publink/show?code=XZqmSw7ZDLXmY082LpYNIdbkKS6YSRJY9bEy

More thoughts:

   - black screen at the beginning is expected, we need to put a loading screen because in this new opendatacam implem, we start YOLO at the beginning, and once it's started the webcam feed is available
   - when we click start tracking, now it is instant because YOLO is already started
   - in this screencast the UI is 1280x720 something, but we should keep in mind that this feels more cluttered in mobile as we have no space with the counters (look at the initial video or resize your screen with the live demo link below)
   - I also noticed we can have contrasts issue with the white text when the webcam stream is bright, this should be fixed somehow, previously we had a dark opacity on top of the webcam frame
  
You can also feel the "old flow" by opening this link of the old experience simulated: https://open-data-cam-jvircjlvae.now.sh/

syedmustafa54, 2019-01-22:
@tdurand 
1.I tried branch 0.3 cloned [AlexeyAB](https://github.com/AlexeyAB/darknet) in the same dir.
2.`npm run build`
3.`npm start`
this is the error i am getting 
```
`Something went wrong: connect ECONNREFUSED 192.168.1.191:8070
/media/bintex/ubuntu/ai-project/lab-opendatacam-0.3/node_modules/eventemitter2/lib/eventemitter2.js:290
          throw arguments[1]; // Unhandled 'error' event
          ^

Error: Cannot stop process that is not running.
    at /media/bintex/ubuntu/ai-project/lab-opendatacam-0.3/node_modules/forever-monitor/lib/forever-monitor/monitor.js:357:26
    at process._tickCallback (internal/process/next_tick.js:61:11)
`
```

closed 2019-01-29 by b-g