opendatacam/opendatacam #62: Docker

tdurand, 2019-01-24:
Was able to make it work 🎉 (on tx2), cc @b-g , but it has been a painful journey 🤒, good news is that we can run YOLO inside a Docker without any perf hits, runs 9-10 fps same as outside docker.. So we will be able to finally provide an easy install way for opendatacam which should be just consist of:

- Flashing Jetson
- Install docker & run docker image _(giving we succeed to make the wifi hotspot config from command line also)_

## Docker on Jetson TX2 specificities:

Since Jetpack 3.2  (we are at Jetpack 3.3), docker can run out of the box.. which is good news because before you had to recompile the kernel and it was much more painful...

The challenge is that out of the box docker doesn't have access to the GPU, so it's useless to run darknet in it.. Nvidia released on an official [nvidia-docker](https://github.com/NVIDIA/nvidia-docker)  base image setup all the nvidia GPU software (CUDA, CUDNN) for us, but no support for Jetson now... (it might come in the future: https://github.com/NVIDIA/nvidia-docker/issues/214 ) 

The good news is that in the meantime there is a workaround to give docker access to the GPU and the jetpack installed CUDA, CUDNNN software , which is documented here: https://github.com/Technica-Corporation/Tegra-Docker#giving-docker-access-to-gpu

We are using this method, the downside are that it seems you can't do any compilation work that need the GPU (or CUDA stuff) when building the Docker image, you need to include pre-compiled binaries: [Some learning along the way](https://github.com/Technica-Corporation/Tegra-Docker/issues/9)

This means you need to pre-compile darknet with the jetson architecture config (different for Xavier and TX2, see: https://github.com/AlexeyAB/darknet/blob/master/Makefile#L10 ) and an opencv version that need to be the same as the one you will be running in your Docker.

You see me coming, I had a hard time figuring out that the opencv version built-in with the Jetpack 3.3 was not the same as the one I was trying to install in the Docker container: https://github.com/jcjimenez/darknet-docker/issues/1#issuecomment-457186788  . (And then a hard time figuring out how to compile your own opencv, install it on Jetpack ... opencv takes something like 1 hour to compile on jetson tx2 ⏰💤)

## How to build a jetson GPU "darknet" Docker image:

_This repo: https://github.com/jcjimenez/darknet-docker/tree/master/jetson-tx2 helped making progress.. It is more advanced than https://github.com/Technica-Corporation/Tegra-Docker_

🚨 this is still WIP, will improve change / things 🚨

1. Compile opencv on jetson using this script: https://github.com/jcjimenez/darknet-docker/blob/master/jetson-tx2/build-opencv.sh  . This take around 1h on the jetson tx2.

2. Remove default jetpack pre-installed opencv ( [source](https://devtalk.nvidia.com/default/topic/1026044/jetson-tx2/newbie-question-how-to-install-opencv-and-setup-the-path-/post/5220636/#5220636) ): 

```
sudo rm -f /usr/lib/libopencv*
sudo rm -rf /usr/include/opencv
```

3. Install opencv 3.4.1 by extracting the tar file produced in 1. at root level /

`sudo tar -xvzf opencv-3.4.1.tgz`

4. Compile darknet with `make`

5. Copy paste darknet folder and opencv compiled tar file in some empty folder we will use for image creation

6. Create a Dockerimage file and copy paste this:

```Dockerfile
FROM arm64v8/ubuntu:16.04

RUN apt-get update && apt-get install -y apt-utils bzip2 curl wget git sudo unp cmake && apt-get clean && rm -rf /var/cache/apt
RUN apt-get -y autoremove && apt-get -y autoclean
RUN rm -rf /var/cache/apt

# Opencv dep
RUN apt-get update
RUN apt-get install -y \
    libglew-dev \
    libtiff5-dev \
    zlib1g-dev \
    libjpeg-dev \
    libpng12-dev \
    libjasper-dev \
    libavcodec-dev \
    libavformat-dev \
    libavutil-dev \
    libpostproc-dev \
    libswscale-dev \
    libeigen3-dev \
    libtbb-dev \
    libgtk-3-dev \
    libgstreamer1.0-0 \
    libwayland-client0 \
    libwayland-egl1-mesa \
    libwayland-cursor0 \
    cmake \
    pkg-config

RUN apt-get install -y python-dev python-numpy python-py python-pytest -y
RUN apt-get install -y libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev
RUN apt-get install -y ffmpeg

COPY darknet/ darknet/

# install opencv
ADD opencv-3.4.1.tgz /

WORKDIR /darknet
```

7. Run docker build command : `sudo docker build -t dockerimagename .`

This should build the image successfully

Extra info:

- Other repo to build opencv on tx2 : https://github.com/jetsonhacks/buildOpenCVTX2  , but installing the .deb file didn't work on Docker, but seems cleaner to overwrite the Jetpack opencv version

- same for Xavier: https://github.com/jetsonhacks/buildOpenCVXavier

## How to run the docker image

1. Need to copy this script: https://github.com/jcjimenez/darknet-docker/blob/master/jetson-tx2/darknet-docker

2. Run it: `sudo ./darknet-docker run --rm -it dockerimagename`  (run and enter the docker instance to try things)

3. Try to run darknet, for example: `./darknet detector demo cfg/voc.data cfg/yolo-voc.cfg yolo-voc.weights -c 0 -ext_output -dont_show -json_port 8070 -mjpeg_port 8090`

## Alternative idea

Also might be even simpler than docker to install for end user,  provide a full Jetson image with everything installed:  https://elinux.org/Jetson/TX2_Cloning  .. But people would need to erase everything on there jetson

## TODO List

- [ ] Complete docker image with node install, run project etc etc ... 
- [ ] Write and test a "classic" Dockerimage to deploy to a non jetson device, ie a Nvidia GPU, using https://github.com/NVIDIA/nvidia-docker , in order to provide an easy way to test opendatacam without a Jetson 
- [ ] Document in Readme both how to build the image and how to run it
- [ ] Build a specific docker image for xavier by compiling darknet with correct flags.

tdurand, 2019-03-30:
_Update on this (note to myself mostly)_

After Jetpack 4.2 release lots of the stuff I had figured out changed  (but for the best 😉)

After lots of struggle I finally managed to have a docker image working again, I even tried to pull it and run it on xavier recently flashed (with only one command) and it worked 🎉 (not optimized as we will need to generate a different docker image for tx2, xavier).

Still todo + things of main issue

- Figure out some weird bug with webcam access that doesn't work sometimes from docker + webcam resolution set by default to maximum
- Add mongodb 
- Add env variable to avoid having a docker branch
- Investigate how to start docker image on startup
- Figure out how to simply edit config.json without having to rebuild the docker image: https://dantehranian.wordpress.com/2015/03/25/how-should-i-get-application-configuration-into-my-docker-containers/
- Figure out wifi hotspot situation, looks we still need to do some stuff: https://elinux.org/Jetson/TX1_WiFi_Access_Point , but maybe can put that in a clean .sh script : https://seravo.fi/2014/create-wireless-access-point-hostapd , https://github.com/offlinehacker/docker-ap or https://github.com/fgg89/docker-ap

Hopefully next week I'll have a first beta to try !

tdurand, 2019-05-02:
Next steps:

- [ ] Work on making opendatacam docker image configurable at runtime: https://github.com/moovel/lab-opendatacam/tree/v2#%EF%B8%8F%EF%B8%8F%EF%B8%8F-opendatacam-settings
- [ ] Easy run from files
- [ ] Easy switch of YOLO weights ( v3 COCO, v3 SPP , v2 VOC, v3 TinyYOLO )
- [ ] See if we can setup the wifi hotspot automaticaly : https://elinux.org/Jetson/TX1_WiFi_Access_Point , https://www.youtube.com/watch?v=YqWuRwKbJKc or via docker: https://github.com/offlinehacker/docker-ap, https://github.com/fgg89/docker-ap

vielmetti, 2019-05-05:
Of interest to you, @DieterReuter

tdurand, 2019-05-10:
For reference, I'm currently facing an issue to have access to the raspberry PI camera form the docker container, I wrote a summary on the nvidia forums: https://devtalk.nvidia.com/default/topic/1051653/jetson-nano/access-to-raspberry-cam-nvargus-daemon-from-docker-container/  . 

It works great when running opendatacam outside the docker container though.

tdurand, 2019-05-11:
Following up on this,

For what someone answered on the nvidia forum: https://devtalk.nvidia.com/default/topic/1051653/jetson-nano/access-to-raspberry-cam-nvargus-daemon-from-docker-container/post/5338069/#5338069,  seems it won't be possible to run with a raspberrycam with the docker setup, it will be a limitation, you will need to install opendatacam without docker to do so. Will be documented

tdurand, 2019-05-13:
Regarding the raspberry pi cam, someone pointed me another potential solution to be able to run it from docker: https://devtalk.nvidia.com/default/topic/1051653/jetson-nano/access-to-raspberry-cam-nvargus-daemon-from-docker-container/post/5338479/?offset=10#5338482

tdurand, 2019-05-17:
Issue solved, we can now access the raspberry cam from the docker container, the solution consist on:

- Starting a process that proxys the Raspberry cam into an usb cam 
- Picking this "fake" usb cam from the docker container

More on this: https://devtalk.nvidia.com/default/topic/1051653/jetson-nano/access-to-raspberry-cam-nvargus-daemon-from-docker-container/post/5338140/#5338140

tdurand, 2019-05-17:
Now documented & made easy to setup with scripts and cron job that start on boot: https://github.com/moovel/lab-opendatacam/blob/v2/doc/JETSON_NANO.md#run-opendatacam-container-with-raspberrypi-cam

BTW keeping this issue open because I still need to improve a bit the Docker images but all the technical issues have been solved ✅

tdurand, 2019-05-23:
Can improve docker wrapper script for jetson with this: https://github.com/rbonghi/jetson_stats/blob/master/scripts/jetson-docker , much cleaner.

Also there is some nice scripts for jetsons in this repo: https://github.com/rbonghi/jetson_stats

tdurand, 2019-05-23:
@florianporada could we have a 20 min chat on some docker improvements idea you have ? for example monday next week or whenever it is good for you, let me know

florianporada, 2019-05-24:
@tdurand Sure. What is your timezone at the moment? :) Monday works for me.

tdurand, 2019-05-24:
2:30 pm for you would work ? I think it can be pretty quick

tdurand, 2019-06-04:
Closing this one in favor of https://github.com/moovel/lab-opendatacam/issues/62

closed 2019-06-04 by tdurand