opendatacam/opendatacam #65: CSI-2 cameras

shams3049, 2019-02-02:
I working with https://www.e-consystems.com/jetson-tx2-ultra-low-light-camera-board.asp camera to allow tracking traffic at night. However without particular success, can any one guide me to the right configuration to set up CSI-2 input instead of USB camera.

tdurand, 2019-02-07:
Hi, can't help much as we didn't try on something else than usb cam, from a quick search it seems that you will need to recompile OpenCV with GStreamer support and change this line of Opendatacam to use Gstreamer.

https://github.com/moovel/lab-opendatacam/blob/master/server/processes/YOLO.js#L24

change this:

`./darknet detector demo cfg/voc.data cfg/yolo-voc.cfg yolo-voc.weights -c 1 -address "ws://localhost" -port 8080`

To something like:

`./darknet detector demo cfg/voc.data cfg/yolo-voc.cfg yolo-voc.weights "nvcamerasrc ! video/x-raw(memory:NVMM), width=(int)640, height=(int)480, format=(string)I420, framerate=(fraction)30/1 ! nvvidconv flip-method=2 ! video/x-raw, format=(string)I420 ! videoconvert ! video/x-raw, format=(string)BGR ! appsink"`

Reference: https://devtalk.nvidia.com/default/topic/1014006/jetson-tx2/opencv-videocapture-with-onboard-csi-camera-/

Also I'm not sure YOLO was trained with night footage, so don't know if it will work well..

closed 2019-02-07 by tdurand

jm677zz, 2019-02-08:
Please write your query to techsupport@e-consystems.com. They will provide the response for your query.