opendatacam/opendatacam #67: Jetson nano

tdurand, 2019-03-19:
New jetson : https://devblogs.nvidia.com/jetson-nano-ai-computing/

Claim to have 25 FPS with Tiny YOLO ... Let's watch more writing on it, maybe would be able to run opendatacam

tdurand, 2019-03-19:
cc @b-g

b-g, 2019-03-19:
@tdurand Nice find! Very interesting ... but I think we need at least YOLO VOC with around 12 FPS to replace the TX2, or? Waht is the FPS of YOLO tiny on the TX2?

tdurand, 2019-03-19:
I need to test this again, but from my early test with Opendatacam v1 I think TX2 was running tiny yolo around ~25 FPS . 

YOLO VOC with more than 8-9 FPS is the "compat" level yes 😉

tdurand, 2019-03-22:
Out of curiosity just tested quickly Tiny Yolo v3 on TX2 and I get 31 FPS... Direct extrapolation would give 8 FPS of nano for YOLO VOC, but yeah, architecture and hardware not the same... 

I guess we should wait and see when more benchmarks comes out... Would be amazing be able to run opendatacam on it.

tdurand, 2019-03-28:
Was able to order one today.. If delay are right should get it mid april.

b-g, 2019-03-29:
Great! Where? Sparkfun?

tdurand, 2019-03-29:
Directly on nvidia.com but in the U.S as I'm delivering to Uruguay.

tdurand, 2019-03-30:
Some things to consider when buying one, it doesn't come with power supply or memory, so you need to buy a microSD and a power supply.

I bough the 5V - 4A power supply in order to be able to run it on the max perf mode 😉

See the "What you will need section": https://elinux.org/Jetson_Nano

b-g, 2019-04-01:
Thanks + good spot! Just ordered a Nano as well ...

tdurand, 2019-04-15:
Before benchmark, hardware wise in order to power through the 5V - 4A and not the microUSB port you'll need to put a jumper in the J48 pin and nvidia doesn't provide this jumper: https://www.youtube.com/watch?v=jq1OqBe267A&t=127s , and here is a link to buy some jumpers: https://www.amazon.com/2-54mm-Standard-Computer-Jumper-100pack/dp/B00N552DWK/ref=sr_1_3?keywords=jumper+electronics&qid=1555349546&s=gateway&sr=8-3

tdurand, 2019-04-15:
[image]

tdurand, 2019-04-15:
Ok first benchmark of darknet on jetson nano:

Neural network | FPS
-- | --
YOLOv2 + VOC | ~4.5 FPS
Tiny YOLOv3 | ~17 FPS

_Note : Nvidia claims 25 FPS with tiny YOLOv3 but with the version using TensorRT https://github.com/NVIDIA-AI-IOT/deepstream_reference_apps , which is not implemented in darknet (https://github.com/AlexeyAB/darknet/issues/2019#issuecomment-444812039) I think that using deepstream we would reach something like 7 FPS, untested yet as the implemtation of yolo in this framework doesn't seem to give easy detection stream, live stream or webcam access as https://github.com/alexeyab/darknet does_ 

It is a bit a disappointment at first (but previsible 😁 as jetson nano is similar to TX1 when you read around), but I then tried nano with opendatacam anyway an here is the result:

**Running YOLOv2 (VOC) at 4 FPS:**

https://youtu.be/qBxkP73NYL0

_You see it can count, but the livestream is really slow, and the detections / mjpeg stream aren't really in sync.._

**Running Tiny YOLOv3 (COCO) at 17 FPS:**

This one is interesting, tiny YOLO seems to have smaller targets but they seems to be pretty accurate frame after frame.. so counting seems to be good. (You also see the path recording works great)

https://youtu.be/5_ajmTtyxWw

**Conclusion**

- TinyYOLO v3 seems to work pretty great in terms of tracking on this video, but we would need to test it is more footage to see what are the results.. _(In a perfect world we would have 5-10 benchmark videos and run "formal" test on them with different combinaison, which could also help to improve the tracker, but this is pretty time consuming)_

- My feeling is that it might be interesting for the project to feature a jetson nano compatibility anyway.. We reached the point that there is so many combinaison possible of YOLO version + weights ( tinyYOLO, COCO in 3 flavours, Openimages, VOC ) and also of hardware ( Nano, TX2, Xavier...) ... that we should just give the possibility to switch between them... 

- This connect with the idea of giving the possibility to run tensorflow.js models (on node + gpu) that could make the nano even more relevant. I'll work on this benchmark also this week

b-g, 2019-04-16:
@tdurand Many thanks for the Nano investigations! Fully agree with your conclusion. 👍 Please go ahead. Have a nice week, B

tdurand, 2019-05-02:
Next steps for jetson nano:

- [ ] Benchmark TinyYOLO on more footage
- [ ] Try to run it with a classic powerbank , I have an anker one that feeds 3A I think
- [ ] Figure out wifi with this [dongle](https://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/in/wireless_adapters_n150/ew-7811un/): 
- [ ] Try with [Raspberry Pi cam](https://www.amazon.com/Raspberry-Pi-Camera-Module-Megapixel/dp/B01ER2SKFS/ref=sr_1_3?keywords=raspberry%2Bpi%2Bv2%2Bcamera&qid=1554831689&s=electronics&sr=1-3&th=1)

for reference for ecosystem and DIY Jetbot robot by nvidia: https://github.com/moovel/lab-opendatacam/issues/74

tdurand, 2019-05-08:
Gstreamer "magic" line to run from the Raspberry CSI cam:

```
"nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink"
```

tdurand, 2019-05-08:
How do you like this 165 USD opendatacam full mobile setup ? @b-g @mmmmmmmmmmmmmmmm @florianporada   ?  

[image]
[image]

It works ! You can power the nano with a "normal" powerbank to charge cellphones, and it has enough power to run darknet with tinyYOLO at 17 FPS 🖖 + raspberry pi cam + usb wifi dongle as a hotspot

List of hardware

- Jetson nano (100 USD)
- Raspberry Pi Camera module (25 USD)
- Anker 10000 mAh powerbank (30 USD)
- Wifi dongle: Edimax EW-7811Un (10 USD)
- Edimax EW-7811UN Nano 
- Electric tape 😎

Night was coming and couldn't benchmark the accuracy very well (would need to by an Infra red cam module 🙃), but it seemed to work very well.

Tomorrow I'll let it run for a while to see if it crashes at some point cause of not enough power...

b-g, 2019-05-09:
@tdurand Mega! So it is's ~$150 without the powerbank! Yay!!

tdurand, 2019-05-17:
I have an issue after playing on the nano today, now it crashes when trying to run on powerbank for any config.

I suspect that the setup with the raspberry cam proxy caused to add more background process... not sure... (I tried everything on AC plug via jack) ... 

For now, everything is crashing on powerbank:

On Docker

-> with usb cam, crashing ==> not sure if this one was working
-> with raspberry cam, crashing

On "bare metal" node.js app run

-> with usb cam, crashing  ==> I'm sure this one was working
-> with raspberrycam , crashing  ==> I'm sure this one was working

I'll setup a clean jetpack install next week sort this out properly. (might also be mongodb ??)

tdurand, 2019-05-17:
Ok, after setuping everything again (and fixing some documentation along the way), found out that I needed just to set the jetson nano to 5W power mode  with this command: `sudo nvpmodel -m 1` (it defaults to 10W).

And every configurations are working from the powerbank supply again 🎉

florianporada, 2019-05-21:
Hi @tdurand, 
I'm finally able to start the testing with the nano (tx2 to come).
One question regarding the installation/setup of the nano: Have you ever encountered issues with pulling the container from the docker registry? I tried it several times but it will fail at some point with a timeout error.

I will try it in a different network and also with a wired connection. Right now it's connected via WiFi USB dongle.

tdurand, 2019-05-21:
great to know I'll soon have feedback from you... No.. weird.. maybe docker hub has some issues...

I always tested while connected to an ethernet connexion.. but this shouldn't be the problem..

What is the exact error from docker ?

florianporada, 2019-05-21:
Thanks your fast reply :) ...
The errors are not consistent but these are the latest ones:
- `net/http: request canceled while waiting for connection (Client.Timout exeeded while awaiting headers)`
- `Error response from daemon: Get https://registry-1.... TLS handshake timeout`

... I have the feeling that it might be a problem with the internet connection (either via the USB Dongle or the network in general here in the office) Just wanted to check if you had similar issues. I will test it via wired connection once I get home.

tdurand, 2019-05-21:
from what I googled it is a problem with a proxy at moovel office I think.... : https://forums.docker.com/t/error-response-from-daemon-get-https-registry-1-docker-io-v2-net-http-request-canceled-while-waiting-for-connection-client-timeout-exceeded-while-awaiting-headers/54848/2

tdurand, 2019-05-21:
https://stackoverflow.com/questions/23111631/cannot-download-docker-images-behind-a-proxy

florianporada, 2019-05-21:
I also saw those posts. But the behavior is, that it starts downloading for some minutes and then stops with one of the errors.
(also we are currently located in a co-working space which makes things more complicated with network related issues)
I'll start another try at home and see what result i get :)

florianporada, 2019-05-23:
@tdurand so the issue was apparently the network. 
Nevertheless: Pulling the odc image via the WiFi dongle is not recommendable since it took ~30min for me. (less than 5mins via ethernet)
I'm currently testing tx2 and nano - I will post my findings in: https://github.com/moovel/lab-opendatacam/issues/70

tdurand, 2019-05-23:
Great ! Feel free to modify the documentation if you see something to add: https://github.com/moovel/lab-opendatacam/tree/v2/doc , and also please be aware that it is not feature complete yet: https://github.com/moovel/lab-opendatacam/issues/60#issuecomment-488832532

b-g, 2019-05-27:
@tdurand I would like to buy a 2nd set for testing ... Could you help to compile the shopping list? + We should add such a hardware list as well to the repo. So far I have noted:

ODC 2.0 Shopping List
- 1x Jetson nano (100 USD)
- 1x [Raspberry Pi Camera](https://www.amazon.com/Raspberry-Pi-Camera-Module-Megapixel/dp/B01ER2SKFS/ref=sr_1_3?keywords=raspberry%2Bpi%2Bv2%2Bcamera&qid=1554831689&s=electronics&sr=1-3&th=1) module (25 USD)
- 1x Wifi dongle: [Edimax EW-7811Un Nano](https://www.edimax.com/edimax/merchandise/merchandise_detail/data/edimax/in/wireless_adapters_n150/ew-7811un/) (10 USD)
- 1x Samsung 128GB 100MB/s MicroSD card (20 USD)
- 1x Power supply?
- 1x [2.54mm Standard Computer Jumper](https://www.amazon.com/2-54mm-Standard-Computer-Jumper-100pack/dp/B00N552DWK/)

Optional
- Power supply (mobile): Anker 10000 mAh powerbank (30 USD)
- Housing: [Naturebytes IP55 Weatherproof Wildlife](https://www.amazon.de/gp/product/B07LFJTBYM?pf_rd_p=71cb17e9-f468-4d3f-94d5-a0de44c50a7e&pf_rd_r=37VYVZGF001W1H816VC5) (50 USD)

tdurand, 2019-05-27:
Great , looking forward to get feedback from you

The minimum setup for 5W Power is:

- 1 Jetson nano
- 1 Raspberrycam module v2 : https://www.amazon.de/Raspberry-Kamera-Modul-mit-Filter/dp/B01ER4FA9U
- 1 Wifi dongle , this one is compatible out of the box: https://www.amazon.de/EDIMAX-EW-7811UN-Wireless-Adapter-IEEE802-11b/dp/B003MTTJOY
- 1 MicroSD card (at least 32 GB and 100 MB7s)
- 1 Power supply: 5V⎓2A Micro-USB adapter ( Either a powerbank with 2A output or a converter from house electricity plug : https://www.adafruit.com/product/1995 ) 

For 10W power mode ( this is good for desktop use when you plug the screen, the mouse, the keyboard, it draws powers from the periferics)

- Power supply: 5V⎓4A DC barrel jack adapter, 5.5mm OD x 2.1mm ID x 9.5mm length, center-positive: https://www.amazon.de/PremiumCord-Netzteil-230V-5V-4A/dp/B07NSSD9RJ
- 1x [2.54mm Standard Computer Jumper](https://www.amazon.com/2-54mm-Standard-Computer-Jumper-100pack/dp/B00N552DWK/) This is used on the J48 Pin when supplying power from the jack entry instead of the microUSB. It tells the jetson to by-pass the microUSD power entry.

For setup:

- 1 usb mouse
- 1 usb keyboard
- 1 screen (HDMI or Displayport)
- And for faster connexion, a ethernet cable to your router

florianporada, 2019-05-29:
@tdurand I'm playing around with the raspi camera as well.
Mounting the camera inside the docker container is still kind of hacky right? I tried several approaches but I was not able to mount the camera in the container. Any tips on that maybe :) ?

akrolic, 2019-05-29:
I wish there was a better more broadly applicable solution than using the tegra-docker wrapper. I am looking into a solution for this problem aswell for another project, but so far @tdurand is the only guy diligently working towards what could be a solution. More updates around this topic would be super helpful :) keep up the good work.

tdurand, 2019-05-29:
@florianporada , yes I already wrote documentation on this: https://github.com/moovel/lab-opendatacam/blob/v2.0.0-beta.4/doc/JETSON_NANO.md#run-opendatacam-container-with-raspberrypi-cam , it's hacky but works. The idea is to proxy the raspberry cam into a "fake" usb cam dev/video2 and pick this one from the docker container

@akrolic  , what are you looking to achieve exactly ? The docker wrapper for jetson works pretty well for us meanwhile nvidia release a compatible docker for ARM64 platform ... BTW there is a cleaner tegra docker wrapper here: https://github.com/rbonghi/jetson_stats/blob/master/scripts/jetson-docker

florianporada, 2019-05-29:
I got it working for some time. Super hacky indeed :D ... but this is something we have to deal with. Thanks a lot! :)

tdurand, 2019-05-29:
@florianporada , what do you mean with "for some time" ? it doesn't work anymore ?

akrolic, 2019-05-29:
@tdurand I am working on implementing project redtail (https://github.com/NVIDIA-AI-IOT/redtail) on the Jetson Nano. It uses Yolo and another DNN TrailNet. I will be using gstreamer to get the camera input. 
I am pretty novice, and have learned so much during this project. Community members like yourself make it possible for me to continue making progress on this project.

tdurand, 2019-05-29:
@akrolic glad the knowledge built with opendatacam helped you !

Fabri73, 2019-06-04:
Hello everyone, I am having a little issue with tiny Yolo and Jetson Nano. Hope anyone can help.
I have set up the JN, installed Darknet and weights. However, using the pretrained weights (for tiny) it takes 0,6ish seconds for 1 jpeg. Even though Cuda is working properly, I tried to compile the examples and it seems to work, there is no difference in performance when I set the compile with GPU=1/0 and CuDnn =1/0. It seems it is not running on GPU. Anything you believe I am missing?
On my laptop it runs properly...
Thanks

tdurand, 2019-06-04:
Hi, 

You can have a look at this documentation: https://github.com/moovel/lab-opendatacam/blob/v2/doc/RUN_WITHOUT_DOCKER.md#2-install-darknet--neural-network-framework-running-yolo-

specificaly:

```
# Set these variable to 1:
GPU=1
CUDNN=1
OPENCV=1

# Uncomment the following line
# For Jetson TX1, Tegra X1, DRIVE CX, DRIVE PX - uncomment:
ARCH= -gencode arch=compute_53,code=[sm_53,compute_53]
```

Best

florianporada, 2019-06-04:
@tdurand the 'for some time' was because the service still stops after a certain amount of time. Also the cronjob which starts the proxyscript for the raspberrycam will not be created if you are not logged in with root. 
.. maybe we should create a dedicated issue for raspi cam related stuff.

Fabri73, 2019-06-04:
@tdurand Thanks, I will follow the advice and let you know.
Cheers
F

tdurand, 2019-06-04:
@florianporada yes this issue is getting a bit crazy, I think you can create an issue "Setup improvements" and list some of the stuff we have been discussing

tdurand, 2019-07-05:
Closing this, jetson nano got his own documentation page with all the knowledge built around it: https://opendatacam.github.io/opendatacam/documentation/jetson/JETSON_NANO.html

closed 2019-07-05 by tdurand

adrianosantospb, 2019-11-06:
> ~17 FPS

I could hear with the same FPS as you, but the video doesn't look fluid. Do you have some idea that what could it be?

tdurand, 2019-11-08:
Running on a jetson nano could feel slow if you have the jetson nano connected to a monitor for example.