opendatacam/opendatacam #68: Mapping of Yolo classes to OpenMojis

b-g, 2019-03-21:
Hi @tdurand @mmmmmmmmmmmmmmmm!

Here are all OpenMojis which are currently available: 
[openmoji-black-svg.zip](https://github.com/moovel/lab-opendatacam/files/2992598/openmoji-black-svg.zip)

I think I would just vote to simply add the mapping of yolo classes to icons to the config file and copy the emoji to an assets folder.

```js
{
	"PATH_TO_YOLO_DARKNET": "/darknet",
	"VALID_CLASSES": ["*"],
	"DISPLAY_CLASSES": ["bicycle", "car", "bus", "motorbike", "truck", "person"]
  	"DISPLAY_CLASSES_ICONS": ["1F6B2.svg", "1F697.svg", "1F68C.svg", "1F3CD.svg", "1F69B.svg", "1F468.svg"]
}
```

or

```json
{
	"PATH_TO_YOLO_DARKNET": "/darknet",
	"VALID_CLASSES": ["*"],
	"DISPLAY_CLASSES": [
	  { "class": "bicycle", "icon": "1F6B2.svg"},
	  ...
	]
}
```
Both fine for me.

tdurand, 2019-03-21:
Great thanks + agree with config 👍 .

tdurand, 2019-03-29:
Do we have also a openmoji-white for the dark / light variation ?

Or maybe I could hack this by styling the color on runtime with currentColor: https://css-tricks.com/cascading-svg-fill-color/  and inlining the svgs in the code: https://github.com/gilbarbara/react-inlinesvg

tdurand, 2019-04-30:
ok done, documented here: https://github.com/moovel/lab-opendatacam/blob/v2/doc/CONFIG.md#display-custom-classes

commit 9dd57211 (2019-04-30): Evo: Map YOLO classes to openmojis fix #68
closed 2019-04-30 by tdurand