opendatacam/opendatacam #69: APIs

tdurand, 2019-03-29:
List of current APIs of the node.js opendatacam app

### Webcam: `GET /webcamstream`

Available after YOLO is started, this endpoint streams the webcam as a MJPEG stream. (streams the sequence of JPEG frames over HTTP).

_The TCP connection is not closed as long as the client wants to receive new frames and the server wants to provide new frames_

_Only support one client at a time, if another one connect, the first HTTP connection is closed_

### Webcam: `GET /videoresolution`

Returns webcam resolution, example:

```json
{
  "w": 1280,
  "h": 720
}
```

### Tracker:  data frame by frame: `GET /tracker/sse`

From the browser, you can open a SSE connection: `new EventSource("/tracker/sse")` . Then it works like websocket but only the server can push data. [(What are SSE : Server Side Events)](https://medium.com/axiomzenteam/websockets-http-2-and-sse-5c24ae4d9d96)

Opendatacam pushes each frame this data:

```javascript
{
        trackerDataForLastFrame: { someData },
        counterSummary: { someData },
        trackerSummary: { someData },
        videoResolution: { someData }, 
        appState: {
          yoloStatus: { someData },
          isListeningToYOLO: { someData },
          recordingStatus: { someData }
        }
}
```

TODO document data

_Only support one client at a time, if another one connect, the first SSE connection is closed_

### Console: `GET /console`

Send the last 3000 characters of the console stoud.

### Recording:  Start recording `GET /recording/start`

Make Opendatacam start recording.

### Recording: Stop recording `GET /recording/stop`

Make Opendatacam stops recording.

### Recording: Get recordings history `GET /recording/history`

TODO implement pagination as for now it returns the last 30 recording

TODO document JSON return

### Recording: Get single recording tracker data  `GET /recording/:id/trackerhistory`

TODO document JSON return, can be a very large json 

### Recording: Get single recording counter data  `GET /recording/:id/counterhistory`

TODO document JSON return

### Counter: Register counting areas  `POST /counter/areas`

TODO document post data

tdurand, 2019-03-29:
cc @b-g , an overview of the current APIs .

tdurand, 2019-04-25:
Should use : http://apidocjs.com/

tdurand, 2019-04-26:
Integrated apidocjs to generate docs on endpoint from comments

Did a first documentation of all endpoints but will improve continously as we close the gap to v2 release.

It looks like this now: https://apidoc-dzchmakjlf.now.sh/ 

cc @b-g

b-g, 2019-04-29:
@tdurand Thanks + nice! Just ... https://apidoc-dzchmakjlf.now.sh/ looks quite like a "big" software project. It might be enough to simply generate with documentation.js a markdown file and use the default github style.

@mmmmmmmmmmmmmmmm @tdurand 
How would you guys imagine the API docs view in the UI be integrated? Simply with an iframe and default styles?

tdurand, 2019-04-29:
I think document.js is more aimed at documenting a node / javascript API and not an REST API. I documented the "REST" API of opendatacam as the goal isn't in my opinion to be a library... But we can talk about this 😉

For integration I was more thinking of  a link "_blank" in the menu to the doc... 

Let's discuss this on next meeting 😎

b-g, 2019-04-29:
OK :) Wasn't aware of the REST aspect ...

tdurand, 2019-05-09:
Some improvements to make to the API:

- _GET /recording/:id/counter_ should have also a specific "GET /recording/ongoing/counter" endpoint to get the current recording data.

- We should add a _GET /counter/areas_ to get the configuration of the counting lines 

- We should add a _GET /opendatacam/status_ endpoint that send the latest status (is currently recording or not, time started, FPS ...) 

This will make the API much more useful to handle use cases like: https://github.com/moovel/lab-opendatacam/issues/79

tdurand, 2019-05-22:
- Document how to configure the mongodb adress so people can also use a remote DB: https://github.com/moovel/lab-opendatacam/issues/83

tdurand, 2019-06-04:
Added more api endpoints to make it easier to build your own layer on top of opendatacam , ideas from #79  , Also added mongodb url in the config file so people can persist data on their own remote instance

Documented them in the API docs also, here is the latest version: https://apidoc-ydewpxduvy.now.sh

tdurand, 2019-07-03:
Ok, improved further the API and API docs for v2 release: https://opendatacam.github.io/opendatacam/apidoc/

closed 2019-07-03 by tdurand