opendatacam/opendatacam #70: Open data cam 2.0.0-beta.1 🎉

tdurand, 2019-04-04:
cc @b-g @mmmmmmmmmmmmmmmm  @florianporada 

### 2.0.0-beta.1 ready 🎉:

After lots of "docker" pain, I managed it ! To have a one line command install & run for opendatacam v2.

[See Install Experience (30s)](https://www.youtube.com/watch?v=NwXrXHHGSgk)

[image]

It actually takes only 3 min on a good connexion to get opendatacam ready to use... what a change from v1 😉

### For @florianporada :

Would be helpful if you can try to run the 2.0.0-beta.1 on the TX2 and give me some feedback . You will need first to re-flash the jetson with jetpack 4.2, see README: https://github.com/moovel/lab-opendatacam/tree/v2

### For @b-g @mmmmmmmmmmmmmmmm  :

If you want to play with the UI, I added a section on the README on how to run opendatacam in simulation mode on your local machine: https://github.com/moovel/lab-opendatacam/tree/v2#run-simulation-mode

### Next steps

Still lots to do, but feel great to already be able to release a beta of this new install process.

- Complete the app on both design and features + fix bugs
- Complete and update the doc, already did started with lots of install and docker workflow: https://github.com/moovel/lab-opendatacam/tree/v2
- Work on making opendatacam docker image configurable at runtime: https://github.com/moovel/lab-opendatacam/tree/v2#%EF%B8%8F%EF%B8%8F%EF%B8%8F-opendatacam-settings
- Wait for Jetson nano delivery to see if we can do something to be able to run on it
- Investigate node.js tensorflow to be able to run other neural nets then YOLO
- ...

tdurand, 2019-04-04:
Know issue:

Already seen that it can crash after a while, will need to investigate

```bash
{
 "frame_id":18393, 
 "objects": [ 
  {"class_id":8, "name":"chair", "relative_coordinates":{"center_x":0.817841, "center_y":0.739374, "width":0.341478, "height":0.508410}, "confidence":0.879339}, 
  {"class_id":8, "name":"chair", "relative_coordinates":{"center_x":0.097692, "center_y":0.740505, "width":0.181534, "height":0.506625}, "confidence":0.368568}
 ] 
}
SyntaxError: Unexpected token , in JSON at position 528
    at JSON.parse (<anonymous>)
    at IncomingMessage.<anonymous> (/lab-opendatacam/server/Opendatacam.js:490:46)
    at IncomingMessage.emit (events.js:189:13)
    at addChunk (_stream_readable.js:284:12)
    at readableAddChunk (_stream_readable.js:265:11)
    at IncomingMessage.Readable.push (_stream_readable.js:220:10)
    at HTTPParser.parserOnBody (_http_common.js:122:22)
    at Socket.socketOnData (_http_client.js:442:20)
    at Socket.emit (events.js:189:13)
    at addChunk (_stream_readable.js:284:12)
2019-04-04T21:27:52.998+0000 I NETWORK  [conn4] end connection 127.0.0.1:47762 (0 connections now open)
==== HTTP Stream closed by darknet, reset UI, might be running from file and ended it or have troubles with webcam and need restart =====
Pheew ....! Something unexpected happened. This should be handled more gracefully. I am sorry. The culprit is:  Error: Cannot stop process that is not run
```

b-g, 2019-04-05:
@tdurand Wow + mega! 🎉

florianporada, 2019-04-05:
@tdurand 
Thanks for the update :) 
It's really an improvement to the old installation process 😁
I'll give you feedback as soon as possible

mmmmmmmmmmmmmmmm, 2019-04-05:
@tdurand Wow, super cool! Thanks!

b-g, 2019-04-26:
Just document the scope of 2.0 quickly:

**2.0**
- Demo mode for online "playback" demos in website
- Emphasis on Nano
- Docker support for: Nano, TX2, Xavier and docker nvidia
- docker nvidia -> enables "cloud support"
- maybe: external ip cam input

**Out** (maybe later)
- Support of additional third party neural nets via tensorflow.js in node.js
- License plate detection

tdurand, 2019-04-26:
Thanks @b-g !

I added a first docker image compatible with Jetson nano for testing, see README of v2: https://github.com/moovel/lab-opendatacam/tree/v2#2-install-and-start-opendatacam-3-min-

For now it will run slow at 4 FPS as it is still running YOLOv2 + VOC , but next week I'll work on making the neural network runned configurable.

tdurand, 2019-05-02:
Ok I've reorganized a bit the repo issues, now we have a v2 milestone: https://github.com/moovel/lab-opendatacam/milestone/1 , and this issue is now for feedback from testing the First beta on jetson.

shams3049, 2019-05-03:
I have three tx2 for my project, and I am mainly using this application on them to count cars and trucks. what kind of test results are you interested in, maybe I can be of some assistance.

tdurand, 2019-05-03:
Hi @shams3049 , many thanks for your message. Mainly just looking to test it on other jetsons than mine to see if it works, and maybe feedback on new the install process...

If you want to try the new version just follow this guide: https://github.com/moovel/lab-opendatacam/tree/v2#-get-started-quick-setup . You will need to flash one of your jetson with the new jetpack 4.2.

shams3049, 2019-05-06:
Installing and running the application works perfectly. No issues. 

I still have to figure out why it happens. But running the application for longer then 4 to 5 hours tends to crash or have no response what so ever. 
[image]

tdurand, 2019-05-06:
@shams3049 , great ! thanks for feedback and the long test, I still need to do those myself.... 

Two things you can try if you want:

- When it crashes / freeze, try to have a look at the logs: I think with `sudo docker logs` 
- More time consuming, but try to install it without docker and see if it has the same problem (maybe it is docker related): https://github.com/moovel/lab-opendatacam/blob/v2/doc/RUN_WITHOUT_DOCKER.md

shams3049, 2019-05-06:
I am running two Jetson TX2 with Jetpack 4.2 one with docker and one without. Will leave them running over night and compair the results 2mro morning. So far both looks okay. 

However I have to add, that the device with Docker is running a Webcam, where as the device running without the docker is running a CSI-2 camera sensor from e-con

tdurand, 2019-05-06:
@shams3049 Awesome , many thanks, let's see what comes out of it...

Just curious, the CSI-2 camera works out of the box with opendatacam ?

shams3049, 2019-05-06:
No totally not, they are one of the tedious and fragile part of the whole setup. I want to generate data in night and day, so I dont have much choice of low light sensors.

So far both the vendors that manufacturer CSI-2 camera censors for Jetson TX2 board are behind on every aspect regarding drivers and interfacing.

tdurand, 2019-05-06:
Ok ! and did you notice any tracker accuracy difference with night footage ? I'm not sure if YOLO is trained for low light environments.

shams3049, 2019-05-06:
YOLO is pretty good at it actually. The trained model are looking for feature set (shape and all). The low light sensor are good at contours. So win/win.

My biggest problem is indefinite run time.

Pretty much out of options on how to make the system stable for longer and longer runs, as the systems are deployed at remote locations,

tdurand, 2019-05-06:
Ok, let's see if we can improve this, but yes it isn't the goal of this project, I mean we want it to be able to run at least a day without crashing and more if possible, but we do not test it much more than this...

With v1 how long did it last before crashing ?

shams3049, 2019-05-06:
V1 crashed more often actually. 

I totally understand as its not the goal of this project. 

As most of the requirements for my project are met thanks to this project. So I have much time to work on the stability of this project. 

So in case you want me to look at a specific part, that you think can improve stability, I would be glad to do so. Feel free to tag me along any performance issue that need in depth analysis.

tdurand, 2019-05-06:
Thanks, well, let's see what comes ou of those test, I guess if it is the core YOLO implem: https://github.com/AlexeyAB/darknet that is crashing we can't do much.. But maybe as now the software we built around it is much better than v1 we could have a way to detect a crash and restart it automaticly after this and restore the current counting areas configuration...

tdurand, 2019-05-10:
FYI install process is broken for v2.0.0-beta.1 as I'm working on v2.0.0-beta.2 which will introduce configurable neural network weights, choose input video (file, difference webcam types...)

v2.0.0-beta.2 Should be ready by the end of the week, of early next week.

tdurand, 2019-05-11:
I just released v2.0.0-beta.2 💪

It features:

- Opendatacam app improvements ( see https://github.com/moovel/lab-opendatacam/issues/60#issuecomment-488832532 ) 
- Configurable settings (neural network run, run from files), See config documentation to get a sense of what is configurable https://github.com/moovel/lab-opendatacam/blob/v2.0.0-beta.2/doc/CONFIG.md
- Docker setup improvements to be able to change settings
- Lots more documentation 

See new README quick start on how to install this new version: https://github.com/moovel/lab-opendatacam/tree/v2#2-install-and-start-opendatacam-3-min-

For now I just did docker builds for jetson nano and jetson tx2 : https://hub.docker.com/r/opendatacam/opendatacam/tags

Any feedback is useful if you test v2.0.0-beta.2 😎

b-g, 2019-05-13:
@tdurand Yay!

TacHuynh, 2019-05-19:
Error when install on jetson TX2 
docker: Error response from daemon: error while creating mount source path '/usr/lib/aarch64-linux-gnu/libGL.so': mkdir /usr/lib/aarch64-linux-gnu/libGL.so: file exists.

nvidia@nvidia-TX2:~/openDataCam$ sudo ./install-opendatacam.sh --platform tx2
Installing opendatacam docker image
Stop any running docker container...
Installing opendatacam v2.0.0-beta.2 for platform: tx2 ...
Download run script for docker ...
--2019-05-18 17:09:06--  https://raw.githubusercontent.com/moovel/lab-opendatacam/v2.0.0-beta.2/docker/run-jetson/run-docker.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.188.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.188.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2837 (2.8K) [text/plain]
Saving to: ‘run-docker.sh’

run-docker.sh       100%[==================>]   2.77K  --.-KB/s    in 0s      

Last-modified header missing -- time-stamps turned off.
2019-05-18 17:09:07 (6.19 MB/s) - ‘run-docker.sh’ saved [2837/2837]

Download config file ...
--2019-05-18 17:09:07--  https://raw.githubusercontent.com/moovel/lab-opendatacam/v2.0.0-beta.2/config.json
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.188.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.188.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1287 (1.3K) [text/plain]
Saving to: ‘config.json’

config.json         100%[==================>]   1.26K  --.-KB/s    in 0s      

Last-modified header missing -- time-stamps turned off.
2019-05-18 17:09:07 (24.2 MB/s) - ‘config.json’ saved [1287/1287]

Create the directory to run on files ...
Download demo video ...
WARNING: timestamping does nothing in combination with -O. See the manual
for details.

--2019-05-18 17:09:07--  https://github.com/moovel/lab-opendatacam/raw/v2.0.0-beta.2/static/demo/demo.mp4
Resolving github.com (github.com)... 192.30.255.113
Connecting to github.com (github.com)|192.30.255.113|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/moovel/lab-opendatacam/v2.0.0-beta.2/static/demo/demo.mp4 [following]
--2019-05-18 17:09:08--  https://raw.githubusercontent.com/moovel/lab-opendatacam/v2.0.0-beta.2/static/demo/demo.mp4
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.188.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.188.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4659459 (4.4M) [application/octet-stream]
Saving to: ‘opendatacam_videos/demo.mp4’

opendatacam_videos/ 100%[==================>]   4.44M  6.95MB/s    in 0.6s    

2019-05-18 17:09:09 (6.95 MB/s) - ‘opendatacam_videos/demo.mp4’ saved [4659459/4659459]

Replace config file with platform default params ... (you can change those later)
NEURAL_NETWORK : yolov2-voc
VIDEO_INPUT : usbcam
Download, install and run opendatacam docker container
Unable to find image 'opendatacam/opendatacam:v2.0.0-beta.2-tx2' locally
v2.0.0-beta.2-tx2: Pulling from opendatacam/opendatacam
bf7c45c81bd7: Pull complete 
5396f07cded3: Pull complete 
e0ceb566582d: Pull complete 
bdb3982847b2: Pull complete 
33ec523b04b9: Pull complete 
581a6de92705: Pull complete 
551288a6a4c6: Pull complete 
e2b3b3dbaea7: Pull complete 
55d980ad84d2: Pull complete 
d1b3d681f78a: Pull complete 
ee359eeb5970: Pull complete 
575abbe0f4f0: Pull complete 
733674d5e32f: Pull complete 
0b75b06c3106: Pull complete 
bd3cb3ae8377: Pull complete 
b4c809d87feb: Pull complete 
a46b207e42e5: Pull complete 
Digest: sha256:82aafdd99acc61b7550d89515290c71132cbbbfdb56df6ad94c737ed8c03a96c
Status: Downloaded newer image for opendatacam/opendatacam:v2.0.0-beta.2-tx2
69d03d167df3132c419a6be0aa3672840efc8a9d53a9075262ebf58e6ca12414
docker: Error response from daemon: error while creating mount source path '/usr/lib/aarch64-linux-gnu/libGL.so': mkdir /usr/lib/aarch64-linux-gnu/libGL.so: file exists.

tdurand, 2019-05-20:
Hum.. And it works when you comment out this line ? Do you have a clean Jetpack 4.2 ?

TacHuynh, 2019-05-20:
Yes, it is a clean 4.2 Jetpack installation.
Worked fine with “file” option, but hangs at usbcam and raspberrycam.
Thanks

tdurand, 2019-05-20:
1. Weird you had to edit the run-docker script for a clean jetpack 4.2 install... I'll try to reproduce this myself later this week

2. Make sure you connect the usbcam before starting the jetson , and that only one of the camera is connected to it... (Raspberry Pi isn't supported yet with docker installation but will be soon) . You can run this command `ls dev/video*` to see if you have a video0 appearing which means you're camera is detected. Also you can do a `sudo docker container list` & `sudo docker logs <containerID>` and look at the logs and send me the logs.

florianporada, 2019-05-23:
Findings from testing the odc on tx2 and nano:

- When flashing the Jetson I had some issues caused by the SDK Manager. Make sure that `python` is installed. Did the trick for me. (`sudo apt-get install python`)

- When I tried to run the `install-opendatacam.sh` script after a clean setup. The script failed with: 
```
Installing opendatacam docker image
Stop any running docker container...
"docker stop" requires at least 1 argument.
```
Since there are no docker containers to stop this error will appear. Quickfix: Run `docker run hello-world` to have a container in the list.

- If the account is used, which is created during the flashing, I think putting it in the docker group from the beginning makes things easier so the whole `sudo` prefixing will disappear. (`sudo usermod -aG docker $USER` + restart/re-login). 

- Unless you are logged in, the services does not seem to work. After logging in via ssh or directly (not with root) and checking if the container is running via `docker ps` the service is available via browser. (Which account did you use? @tdurand) 

- After the `install-opendatacam.sh` script finished, the usb cam was not recognized by the default config (`"usbcam": "v4l2src ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink"`) . Adding `device=/dev/video1` (or the path where the camera is reachable) after `v4l2src` solved the issue for me. (Possible Improvement: Introduce a device variable which can be modified in the config file as well.)

- It seems that the JSON data for the path feature generates non valid JSON: 

```
{
 "frame_id":8383,
 "objects": [
  {"class_id":14, "name":"person", "relative_coordinates":{"center_x":0.013917, "center_y":0.808672, "width":0.027029, "height":0.171619}, "confidence":0.317677},
  {"class_id":14, "name":"person", "relative_coordinates":{"center_x":0.278547, "center_y":0.735929, "width":0.041525, "height":0.156780}, "confidence":0.275129},
  {"class_id":6, "name":"car", "relative_coordinates":{"center_x":0.432426, "center_y":0.803255, "width":0.122990, "height":0.129827}, "confidence":0.770144},
  {"class_id":6, "name":"car", "relative_coordinates":{"center_x":0.215096, "center_y":0.890960, "width":0.157477, "height":0.144858}, "confidence":0.642485},
  {"class_id":6, "name":"car", "relative_coordinates":{"center_x":0.575315, "center_y":0.742808, "width":0.095191, "height":0.093791}, "confidence":0.561879},
  {"class_id":6, "name":"car", "relative_coordinates":{"center_x":0.062624, "center_y":0.936735, "width":0.119383, "height":0.140294}, "confidence":0.523762},
  {"class_id":6, "name":"car", "relative_coordinates":{"center_x":0.745875, "center_y":0.935939, "width":0.103746, "height":0.110257}, "confidence":0.452738},
  {"class_id":6, "name":"car", "relative_coordinates":{"center_x":0.866210, "center_y":0.785294, "width":0.106178, "height":0.172627}, "confidence":0.367681},
  {"class_id":6, "name":"car", "relative_coordinates":{"center_x":0.630859, "center_y":0.959244, "width":0.181323, "height":
SyntaxError: Unexpected end of JSON input
    at JSON.parse (<anonymous>)
    at IncomingMessage.<anonymous> (/lab-opendatacam/server/Opendatacam.js:490:46)
    at IncomingMessage.emit (events.js:189:13)
    at addChunk (_stream_readable.js:284:12)
    at readableAddChunk (_stream_readable.js:265:11)
    at IncomingMessage.Readable.push (_stream_readable.js:220:10)
    at HTTPParser.parserOnBody (_http_common.js:122:22)
    at Socket.socketOnData (_http_client.js:442:20)
    at Socket.emit (events.js:189:13)
    at addChunk (_stream_readable.js:284:12)
Error with msg send by YOLO, not valid JSON
0.085623}, "confidence":0.311012},
  {"class_id":6, "name":"car", "relative_coordinates":{"center_x":0.524645, "center_y":0.747425, "width":0.088093, "height":0.107299}, "confidence":0.295245},
  {"class_id":6, "name":"car", "relative_coordinates":{"center_x":0.649867, "center_y":0.721911, "width":0.074840, "height":0.071174}, "confidence":0.271292},
  {"class_id":6, "name":"car", "relative_coordinates":{"center_x":0.871647, "center_y":0.794196, "width":0.162976, "height":0.306448}, "confidence":0.268356},
  {"class_id":5, "name":"bus", "relative_coordinates":{"center_x":0.168569, "center_y":0.656027, "width":0.204142, "height":0.246275}, "confidence":0.257718}
 ]
}
SyntaxError: Unexpected token } in JSON at position 8
    at JSON.parse (<anonymous>)
    at IncomingMessage.<anonymous> (/lab-opendatacam/server/Opendatacam.js:490:46)
    at IncomingMessage.emit (events.js:189:13)
    at addChunk (_stream_readable.js:284:12)
    at readableAddChunk (_stream_readable.js:265:11)
    at IncomingMessage.Readable.push (_stream_readable.js:220:10)
    at HTTPParser.parserOnBody (_http_common.js:122:22)
    at Socket.socketOnData (_http_client.js:442:20)
    at Socket.emit (events.js:189:13)
    at addChunk (_stream_readable.js:284:12)
Error with msg send by YOLO, not valid JSON
```
Could this be a streaming issue?

- I tried to adjust the resolution for the camera but it failed for all resolutions I tried. (I used `lsusb -s 001:002 -v | egrep "Width|Height"` to get the available resolutions)

```
VIDEOIO ERROR: V4L: device v4l2src device=/dev/video1 ! video/x-raw, framerate=30/1, width=1280, height=720 ! videoconvert ! appsink: Unable to query number of channels
```

[... cut ...]
- __Nano only__: The live stream of the camera has also a big delay (IMO: can also be ignored since the detections are good and just don't match the camera stream. Which is also just for calibration and initial setup)
- __Nano only__: Also the nano crashes if the voltage is not set to 5W via command line. (assuming that the power supply only provides 5W)

Besides those little bugs, the installation process is super easy and straight forward. Including the feature update it is a super huge improvement to v1 👍👍😃

tdurand, 2019-05-23:
@florianporada  , many thanks for this , very helpful

- Python issue with sdk manager : Could you add a line in "Common issues" in https://github.com/moovel/lab-opendatacam/blob/v2/doc/FLASH_JETSON.md ? 

- For the docker stop yes I already fixed this but didn't release a new beta yet

- Interesting that you can avoid the sudo ... Not sure I quite understand everything.. For now you just don't log in, the docker hub images of opendatacam are public.. You say that if you do not login to a docker account the services of auto-restart doesn't work ? Maybe we can have a quick call to discuss docker improvements

- Yes I think we should try to detect which device usbcam is available, for now we assume it will be 0 .. Did you had more than one thing connected .. not sure why it was on video1 .. Also the gstreamer pipeline is `"v4l2src device=/dev/video1 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink"` for alternative video devices... I'll document this in the next beta I'll release.. 

- Yes darknet sends not valid JSON on each tick but I also fixed this mess by doing some more hacking around the JSON streams lately, will also be released in next beta 

- To adjust the camera resolution, you need to change it in the Gstreamer pipeline (one of the reasons we use gstreamer now) .. Here is how you do it: `"v4l2src device=/dev/video1 ! video/x-raw, framerate=30/1, width=1280, height=720 ! videoconvert ! appsink"` ... Need to document this.. but the idea is to limit it to a low resolution so the video stream is fast, and YOLO just needs  400x400 in input.

- Yes this is a limitation of opendatacam, only one client working at a time.... Will see in the future if we want to invest time on this

- Yes nano takes almost 30s to load YOLO and start..

- This is not normal, maybe related to the non valid json, shouldn't have stream lag with the detections

- For the power setting for nano, yes, need to document this also.

I'll release a new beta by the end of the week so you can test some improvements

florianporada, 2019-05-23:
- Concerning `Interesting that you can avoid the sudo ... Not sure I quite understand everything.. `

`sudo usermod -aG docker $USER` only adds the current logged in user to the docker linux group and gives it permission to use the **docker** without `sudo`. - So this has nothing to do with the docker registry :). A proposal would be to add this command: `$ sudo usermod -aG docker $USER` to the `install-opendatacam.sh` script. 

- the Auto restart of the container on reboot works. But I have to ssh on the jetson (and apparently run a docker command like: `docker ps`) in order to be able to use the opendatacam via the browser. Maybe docker is stuck or something - are you able to start the the opendatacam without ssh-ing onto the hardware?

- Camera resolution: I tried adjust the resolution as described by you but it results in the `VIDEOIO ERROR: V4L: device v4l2src device=/dev/video1 ! video/x-raw, framerate=30/1, width=1280, height=720 ! videoconvert ! appsink: Unable to query number of channels
` error. I curious if higher resolution leads to better results in this case.

- I'll update the docs accordingly

tdurand, 2019-05-23:
- Ok, will try this, good suggestion !

- Hum... yes, docker container starts on boot and start the node.js app... On the nano it takes time though, but after 5 min you should be able to open your browser at IPOFJETSON:8080 which sshing into it..

-  Ah yes sorry, I need to write more doc on this, but you need to specify a width and height supported by the webcam :  in order to find out which resolution are supported, try this:
```
sudo apt-get install v4l-utils
v4l2-ctl --list-formats-ext
```

- Higher resolution won't help with better detections as YOLO resize the input to 412x412 px anyway.. But the default for the height is lower than that so I should put at least a bit more.

tdurand, 2019-05-23:
- BTW the command that start the docker container is: `sudo ./run-docker.sh run -d --restart unless-stopped opendatacam/opendatacam:$VERSION-$PLATFORM`  if you want to investigate whever the container is starting at boot

florianporada, 2019-05-23:
FYI:
The status of docker was `Inactive` on startup (`$ systemctl show --property ActiveState docker`)
I think for me, running `$ sudo systemctl enable docker` did the trick.
After restarting, I was able to connect via the browser to the jetson board. ✌️

tdurand, 2019-05-23:
Ok, good to know, maybe you cam add this to the TROUBLESHOOTING ;-)

tdurand, 2019-05-28:
I've tagged a v2.0.0-beta.4 and released Jetson Nano and TX2 images.

Still need to work on some setup improvements discuss here, and this release make me thing that we need an upgrade script, for now need to re-run the install script, which should work

PS: I may be half broken, just after pushing the docker image, I found out that the data page in the app was crashing... will investigate next week.

tdurand, 2019-06-04:
FYI beta.4 was half broken because I didn't update my config.json file after updating the docker container, I added a note about this (upgrading from v2.x to v2.x): https://github.com/moovel/lab-opendatacam/tree/v2#2-bis-optional-upgrade-opendatacam-from-v2x-to-another-v2x-version

tdurand, 2019-07-05:
Closing this in favor or https://github.com/opendatacam/opendatacam/issues/100

closed 2019-07-05 by tdurand

MeTaXaS4, 2019-07-05:
I am trying to use the opendatacam with nvidia docker and a webcam and I am getting countless of errors "Video-stream stopped!" can you do a simple change there with a counter or something to not have 100.000 same errors on log? so the log is unreadable after all of those and I cannot find out why the stream don't open

tdurand, 2019-07-05:
Hi, yes I feel you.. We would need add some kind of filter...

If you have lots of errors this means that YOLO seems to start...

Please try to restart the docker container, open the app and immediatly click on the show details > download log button so we see the first part of the log.

tdurand, 2019-07-05:
Also please comment directly on #100 or create a dedicated new issue, I closed this issue 😉