opendatacam/opendatacam #71: Tensorflow.js with Node.js

tdurand, 2019-04-16:
@b-g spent some more time evaluating this, summarizing relevant finding here for future work on it

### Tensorflow.js on node challenges

Tensorflow.js project is still much oriented to run inside browser, this means

- No direct webcam access via browser APIs
- Not much examples yet for node : https://github.com/tensorflow/tfjs-examples , most of the example feature how to train but not how to run the trained model with node.js
- Need to workaround some optimized browser things (model weights split in several files etc etc, image handling)  
- Need to use @tensorflow/tfjs-node-gpu to access CUDA and GPU performance instead of WebGL

That said this seems to be evolving toward more [node.js friendliness](https://github.com/tensorflow/tfjs/issues?q=is%3Aissue+node.js+label%3Acomp%3Anode.js+is%3Aopen) with issues opened labeled "compat:nodejs"

### Our use case

For our use case, we want to be able to run on a webcam stream some of the pre-built tensorflow models, like posenet: https://github.com/tensorflow/tfjs-models/tree/master/posenet

While this is [easy with tensorflow.js on the browser](https://github.com/tensorflow/tfjs-models/blob/master/posenet/demos/camera.js#L313), with node this requires:

1. A way to get the webcam stream without webcam browser APIs
2. A way to feed each frames into tensorflow.js without DOM Images and Canvas APIs

**1. access webcam stream in node.js**

_idea 1:_

This node.js package: https://www.npmjs.com/package/node-webcam  is a wrapper around fswebcam to directly get an image as a buffer or base64 in node.js

_idea 2:_

There is several github repos which offers some bindings to ffmpeg or gstreamer to setup a IP server from the webcam for example: https://github.com/sepehr-laal/livecam , and then we get new frames from this from the node.js process that would handle the tensorflow part

This also could be done outside node with some ubuntu package: https://motion-project.github.io 

_idea 3:_ 

Use some opencv node.js binding to use opencv webcam apis: https://github.com/justadudewhohacks/opencv4nodejs

**2. feed frames into tensorflow.js**

I built a working example for this: https://github.com/tdurand/tensorflow-node-experiments/blob/master/posenet.js , basically it uses some canvas node.js library to feed into tensorflow.js the same thing as the DOM Apis would allow. 

There is an outgoing issue on the tensorflow-node official repo to facilitate this: https://github.com/tensorflow/tfjs/issues/1416

Another way to do it is to directly convert images data to Tensor3D : https://dev.to/ibmdeveloper/machine-learning-in-nodejs-with-tensorflowjs-1g1p

**feasibility of our use case**

It's not that strait-forward, but it should be pretty easy to use those blocks to make something like posenet work on a webcam stream 

### Problem with @tensorflow/tfjs-node-gpu on ARM64 devices 

Now it's when it gets more problematic. I went on playing directly on the jetson to assemble this prototype and it turns out that the package that gives tensorflow.js on node access to GPUs via CUDA stuff is not working on Jetson (and ARM64 devices): 

- https://github.com/tensorflow/tfjs/issues/765
- https://github.com/tensorflow/tfjs/issues/1311

At the same time nvidia has released an official tensorflow compat package with jetson nano: https://devtalk.nvidia.com/default/topic/1048776/jetson-nano/official-tensorflow-for-jetson-nano-/1 , but this is for the python stuff and for tensorflow.js node gpu we need some C/C++ tensorflow libs: https://devtalk.nvidia.com/default/topic/1048711/jetson-nano/tensorflow-libs-for-c-c-integration-on-jetson-nano/ . 

It seems we can compile ourself those things: https://collaborate.linaro.org/display/BDTS/Building+and+Installing+Tensorflow+on+AArch64, but I didn't tried it yet

For now this is a blocker to use tensorflow.js node on jetson.

### Conclusion

Definitely not a low hanging fruit 🍎, but things move fast and at the tensorflow conference in march, they announced that making embedded platforms compatible with tensorflow.js was on the todo list : https://youtu.be/x35pOvZBJk8?t=1041 .. 

Also I think it's totaly doable to figure out ourself how to make things works now, but unclear how many road block we will need to clear on our way 😉.

b-g, 2019-04-17:
@tdurand Many thanks for the exploration and the nice write up! Make sense + 💯 agree! More in our next call. Thanks!

tdurand, 2019-04-25:
_Note: Regarding the compiling issue of tensorflow gpu node on nano, someone did post a gist of a working method: https://gist.github.com/sdeoras/533708a95a3de5ff6de1c6b018cabf6d_

tdurand, 2019-04-30:
Keep updating with pointers, some example of using tensorflow node on CPU (not GPU) for object detection: https://github.com/lucleray/object-detection/ , https://object-detection.now.sh/