opendatacam/opendatacam #75: YOLO on iOS

tdurand, 2019-04-30:
cc @b-g @mmmmmmmmmmmmmmmm 

Since we were talking about one of the potential future milestone would be running opendatacam directly on mobile devices, I though I would share this app (someone retweeted it in my feed):  https://itunes.apple.com/us/app/idetection/id1452689527

It is running the full YOLOv3 SPP weights and claims 13 FPS on latest iPhones (XS I think) . 

On my iPhone SE it just runs at 2 FPS but I guess Tiny YOLO would already run at a good enough... So I think the performances are already there to do it

Their work is opensource: https://github.com/ultralytics/yolov3   , the "challenge" is to port YOLO on CoreML, the APIs for machine learning in iOS.

They have also a nice transfer style app: https://itunes.apple.com/app/id1445737240

tdurand, 2020-03-30:
Just a quick update on this one, seems that it would be now possible to run YOLO easily from recent iOS devices, the app I mentionned a year ago is still getting updated to get an idea of the performance you can get: https://apps.apple.com/us/app/idetection/id1452689527   .. Seems you can run full yolo (not tiny) at 30 fps + on latest iPhone 11+ .. I have an old iPhone SE still so I can't really test hehe

And how to use YOLO from iOS is even documented in the official apple developer docs: https://developer.apple.com/machine-learning/models/ , sure there is plenty of example in github on how to use this.

Maybe and idea for OpenDataCam could be to release an "OpenDataCam" app which would work on latest iPhone devices 😁..  cc @b-g @mmmmmmmmmmmmmmmm

tdurand, 2020-03-30:
Their YOLO tiny (https://apps.apple.com/us/app/idetection/id1452689527) runs at 9-10 fps even on my iPhone SE.. impressive

laem, 2020-03-31:
Hi, awesome project !

I was wondering if you've explored the possibility of running opendatacam on a relatively modern Android phone.

I was worried about performance, but I tried [this app](https://play.google.com/store/apps/details?id=hash.tf.objectdetection&hl=ln), which looks like a simple tenser flow mobile template app (there are similar apps with a different name on the store) and it looks like it could run fast enough to monitor trafic on my S10e Exynos 9820 (released in 2019). Sorry, I'm confined, and no window directed to a road 😄 ). It runs the above app's inference on my hand moving at ~10FPS

Why Android phones ? The idea came from the fact that I have a S8 at the bottom of a drawer, with a half-broken screen, rough to the fingers but fully working. ~8FPS.

Android phones are probably the most common integrated processing units (chip + wifi + screen + good camera) one can find at home and in enterprise fleets.

tdurand, 2020-03-31:
Hi @laem !

Many thanks for the input, just renamed this issue to "on Mobile", so people know we don't have any preference regarding iOS / Android..  Historically iOS was a bit ahead of the game regarding Deep Learning stuff on mobile.

Indeed https://play.google.com/store/apps/details?id=hash.tf.objectdetection&hl=ln seems to be running at 13-15 FPS on my Xiaomi mi 8... So it is also promising, I guess it is running YOLO under the hood also, but don't know which weights. Also couldn't find if the implementation if published somewhere.

Building a proof of concept of OpenDataCam on Mobile is becoming very tempting ;-) , I would definitely be a big step towards accessibility and ease of use of the project.

Please share any further findings !

laem, 2020-03-31:
This 1 year old blog poste could be interesting https://blog.francium.tech/real-time-object-detection-on-mobile-with-flutter-tensorflow-lite-and-yolo-android-part-a0042c9b62c6

This could be the source of those Play Store apps : https://github.com/szaza/android-yolo-v2

tdurand, 2021-01-11:
Hi @laem , OpenDataCam on mobile is happening 🙌 , please see: https://github.com/opendatacam/opendatacam/issues/365

closed 2021-01-11 by tdurand