opendatacam/opendatacam #76: Documentation

tdurand, 2019-05-02:
We start to have several markdown files:

- https://github.com/moovel/lab-opendatacam/blob/v2/doc/CONFIG.md
- https://github.com/moovel/lab-opendatacam/blob/v2/doc/CREATE_DOCKER_IMAGE.md
- https://github.com/moovel/lab-opendatacam/blob/v2/doc/FLASH_JETSON.md
- https://github.com/moovel/lab-opendatacam/blob/v2/doc/RUN_WITHOUT_DOCKER.md
- TODO one on architecture

Should we use the Wiki tab of github ? In my experience it is nice to have the contextual menu

Another way to go would be to generate a website from markdown file like with  something like read the doc : https://docs.readthedocs.io/en/stable/intro/getting-started-with-sphinx.html  ... But not sure we want this.

Let's see as we add more stuff to the README is some change is needed

b-g, 2019-05-02:
Hi @tdurand, True! Either way is fine. I personally would favor simply to have a TOC in the README listing the topics and linking to the other files. That way everything incl. the tocs is nicely contained in our repo. The wiki tab feels at least for the slightly disconnected.

tdurand, 2019-05-02:
Yes, for now will just work on content, and when we have all the content let's see what makes sense.

tdurand, 2019-05-17:
- [ ] Work on a CODEOFCONDUCT.md section

tdurand, 2019-05-27:
- [ ] Work on a CODEOFCONDUCT.md section
- [ ] Add a LICENCE , ping @b-g

tdurand, 2019-06-26:
- [ ] Work on a CODEOFCONDUCT.md section
- [ ] Add a LICENCE.md
- [ ] Work on a simple technical architecture diagram

tdurand, 2019-07-02:
cc @b-g , first shot at a general technical architecture schema.. Feel free to edit here: https://docs.google.com/drawings/d/1ew5ux-wcMdxqHVwois21zYpXn63UNfM0r48wZcCl84o/edit?usp=sharing

[image]

tdurand, 2019-07-02:
Also assigned you for Ethics + Licence + Proof read (see first post of this thread)

tdurand, 2019-07-11:
See this other tool to generate a website from technical doc: https://www.docsy.dev/  (by Google) , with https://docs.readthedocs.io/en/stable/intro/getting-started-with-sphinx.html these would be too ways to improve our technical doc experience as we grow them in the future. _(for example many place in the doc I would like to repeat something or do a condition to make it more digest, but with raw markdown I can't)_

shams3049, 2019-07-11:
I personally thing the user device tobe just be a remote control, access and storage point, sort of conveys the true potential for this application. 

https://purepng.com/photo/25166/clipart-dedicated-server

closed 2019-08-28 by tdurand