opendatacam/opendatacam #79: Discrete Data Files

shams3049, 2019-05-06:
First of all, its a great project. It has saved so much of my time to be started with my project. 

I wanted to ask, how can I approach the data files to be saved at 10 min intervals and start a new data counter.

tdurand, 2019-05-06:
Let me get back to you with pointer on this on this by the end of the week, I'll likely work on it 2 days later this week...

I think this might be an interesting use case to include... We persist all the data in DB so I think it would be more to have a way to export the data with some time clustering each X min of the counted data and not directly have feedback on the UI (this would require more work and hardcode a specific use case)

If you can share more on your particular use case for this let us know.

cc @b-g

tdurand, 2019-05-06:
With v2 we have an API endpoint to export the counter data, and you have in it an array of all counted items with a timestamp associated (look into "counterHistory"): https://apidoc-dzchmakjlf.now.sh/#api-Recording-Counter_data . 

For now with the v2-beta.1 you can export this data in the "data" tab, and click on download button on the counter part of the recording.

Then you'll need to build something that parse this JSON and cluster each X mins...

shams3049, 2019-05-06:
I my use case, we have multiple Jetson tx2 deployed, around a harbor,  each one keeping a local count, 

We sink to a central dashboard, that parses individual location data in a meaningful way at a **fixed interval**, and update the overall harbor status based on the count, if the traffic in normal or not, and based on long term collection of this data, if we can predict a future situation.

tdurand, 2019-05-09:
Ok, great to learn about some real world usage of opendatacam, I've added some API endpoints to make that will enable you to build around the v2 API to solve your use case: https://github.com/moovel/lab-opendatacam/issues/69#issuecomment-491073779

tdurand, 2019-05-09:
I'll close this for now, feel free to reopen once you try out the APIs.

closed 2019-05-09 by tdurand