opendatacam/opendatacam #80: how to use deepstream on jetson nano

BlackLionZhou, 2019-05-08:
I want to use DeepStream on the jetson nano ,but i cant find webset to download it .i know this https://developer.nvidia.com/embedded/deepstream-on-jetson-downloads 
but there is nothing for nano .when i search it  ,i find someone is using  DeepStream on the jetson nano . can anybody tell me how to do it?

tdurand, 2019-05-08:
Hi, we do not use deepstream with opendatacam, the best place to ask this would be on the nvidia forum directly: https://devtalk.nvidia.com/default/board/371/jetson-nano/

closed 2019-05-08 by tdurand

BlackLionZhou, 2019-05-08:
Ok ,thank you !

| |
zzqfang
邮箱：zzqfang@163.com
|

Signature is customized by Netease Mail Master

On 05/08/2019 18:26, Thibault Durand wrote:

Closed #80.