opendatacam/opendatacam #81: 3D Tracking with Zed (stereo camera)

tdurand, 2019-05-09:
FYI the darknet fork we use now support Zed stereo camera: https://www.stereolabs.com/zed/

This means that you can have with the bbox the distance to the camera.

[image]

No sure if this could be useful for some opendatacam application but it is worth sharing and documenting.

cc @b-g

b-g, 2019-05-09:
@tdurand Wow! Yes we should document that and at least try it out. Do you know the pricing of such a stereo cam? Model?

tdurand, 2019-05-09:
I think it's 449 USD ,  https://store.stereolabs.com/products/zed/

tdurand, 2019-05-09:
There are other brand with lower price (200 USD), but unclear it would be hard to support it with this darknet fork: https://store.intelrealsense.com/buy-intel-realsense-tracking-camera-t265.html

b-g, 2019-05-10:
Thanks! I think we are actually using the ZED one for our "Who wants to be a self driving car?" buggy setup :) And yes, we definitely should document that it is possible to use it with a stereo cam in the docs.

tdurand, 2019-05-10:
Ah great, so if you have one in the office we might be able to test, the integration with the Zed camera needs this ( documented here: https://github.com/AlexeyAB/darknet#how-to-compile-on-linux , last bucket point)

- Install the ZED SDK on device
- Compile darknet with ZED_CAMERA=1
- Then run darknet with `LD_LIBRARY_PATH=./:$LD_LIBRARY_PATH ./uselib data/coco.names cfg/yolov3.cfg yolov3.weights zed_camera``
- Would need to investigate where is the 3D info in the detections stream and make some changes to the tracker to "forward" them..

uvaaland, 2020-07-14:
Hi!

I have successfully been working with a Logitech C920 usb-camera in opendatacam and it works great! I recently got hold of a Zed 2 camera and started to explore if it would be possible to use it with opendatacam. As a first step, I have Zed 2 working with Yolo v4, but it doesn't follow that it will work in the opendatacam framework.

I am wondering if you could give me an idea of what modifications are needed in opendatacam to be able to use a Zed 2 camera? I don't need any 3D features (like 3D tracking, etc.), I just want to be able to use my Zed 2 with opendatacam in place of my Logitech C920.

Thanks! :)

tdurand, 2020-07-15:
I don't have a Zed camera so I'm not sure how is it supported by darknet now... 

But if you can make darknet work with your zed camera it should work with OpenDataCam.. We just maintain a very light diff between darknet code base and our fork: https://github.com/opendatacam/darknet/pull/6

uvaaland, 2020-07-15:
It requires me to use the `./uselib` executable instead of `./darknet` which is not the one that is currently used by opendatacam. However, I saw that there was a time when you looked at using `./uselib` as the executable (opendatacam v3.0.0-beta.1). Even though this idea is deprecated, do you think it would with a Jetson Xavier? Should I use the darknet version from the following pull request?

https://github.com/opendatacam/darknet/pull/5

uvaaland, 2020-07-15:
Hi again!

I was able to find a solution which requires minimal modifications. By changing a single line in the config.json file I got the desired result:

```bash
"usbcam": "v4l2src device=/dev/video0 ! videobox right=2208 ! videoconvert ! appsink"
```

That is, remove the `video/x-raw, framerate=30/1, width=640, height=360` part of the usbcam instruction and replace it with a crop (videobox). Then it runs with the Zed camera on the latest docker container. The crop is needed because the output will otherwise be two images concatenated due to the Zed camera's two 'eyes'.

tdurand, 2020-07-15:
okay glad you figured it out .. yes ./uselib isn't compatible out of the box with opendatacam, you would need to modify the JSON stream I think.. so better for you to just make the Zed camera work with ./darknet

Denizzje, 2021-01-08:
Hello, 

I am recieving a Zed camera hopefully early next week. I would like to get the distance of objects measured, which is now natively supported in Yolo. Is it somehow possible that ODC passes through this information so it becomes a part of the tracker? Because that would be very useful for my test project. I need to have that data in combination with the already existing parts in ODC (timestamp, direction etc).

tdurand, 2021-01-08:
If now `./darknet` supported the ZED camera (which I don't know) and include some 3D info in the JSON stream.. should be easy to change things in the tracker so it keeps the data.

let me know when you have the camera and test with darknet and I can point to the changes in the code to be made

uvaaland, 2021-01-08:
I had a look at this last year and found that the Zed functionality in YOLO was available using the binary runfile `./uselib` that can be compiled by setting `LIBSO=1` in the Makefile. However, it is not available with the standard `./darknet` executable, which is the one that ODC uses. As far as I can tell from the Darknet documentation, this is still the case.

I played with the idea of modifying ODC to run with the `./uselib` executable, since there was an old branch of ODC that had looked at this, but since my knowledge of JavaScript is limited I decided against it and to wait and see if the Zed functionality would be included as part of the `./darknet` executable in the near future.

vsaw, 2021-01-08:
@uvaaland Have you looked at the current development branch? We made the way we launch darknet more flexible. Potentially allowing it to call different executables (see YoloSimulation as an example) Maybe this makes it easier for you :-)

uvaaland, 2021-01-10:
@vsaw Thanks for the pointer! I like the direction that the project is moving.

I should mention that what I originally wanted to do was to run ODC with the Object Detection module that comes with the ZED SDK (https://www.stereolabs.com/docs/object-detection/). However, at the time ODC was more closely tied to running YOLO through darknet as the backend. Therefore, using ZED as part of darknet was an option that seemed more attainable with ODC at the time (e.g. running ODC with the `./uselib` executable).

I know there has been talks in the ODC community about making it more backend agnostic (using Deepstream, etc.). For now, using the ZED camera in ODC without any 3D features is sufficient for my use-case. What I would ultimately want to do is to run ODC with an executable that uses the Object Detection module in the ZED SDK as backend, and it seems that with the changes you are making in the development branch, it is getting closer to a point where this is attainable 👍