opendatacam/opendatacam #84: OpenDataCam in the wild: testimonials, use cases and examples?

b-g, 2019-05-23:
# 📣🥁 Dear people using OpenDataCam, please get in touch with us and show us your examples of OpenDataCam in the wild!

While preparing the layout and contents for the upcoming new website for OpenDataCam v2 ... we realized that it would be nice to know more about "OpenDataCam in the wild" e.g. testimonials, use cases and real world examples. 

cc @tdurand

tdurand, 2019-05-23:
👌️ Maybe @shams3049 would be interested to collaborate with us on this

shams3049, 2019-05-23:
Hi, @tdurand @b-g Defiantly I am interested on collaboration on this one. 

One of my use case is using this application for collecting traffic data from multiple locations (24/7) and pushing them to a single server.

I am still running test on remote locations and trying to keep the service live and have better remote diagnostics.

In relation to the website, I want to work on the use-case to have the possibility on the webpage to push data to a desired location (local or cloud).

Let me know what you think!

Looking forward to

b-g, 2019-05-27:
Hi @shams3049, many thanks for the interest!

Could you share with us "what" are you doing with ODC non technical? What and why are tracking certain things? (If this is a business secret ... could you generalize it on an abstact level)

We decide deliberately not to deal with any data storage or cloud connection. Basically the message is: you own your data. We don't want and have to deal with this. German data privacy laws are very strict and hence we decided not to touch this topic at all, every user has to find a pragmatic solution on his/her own :) Sorry! I hope you understand.

shams3049, 2019-05-27:
Hi @b-g , 

Its an open project with public funding, Here is some insight in to the project. 

We have a dashboard designed in specific for the needs of Naus Harbour, in NRW region. Following link should work do some basic level. https://logisticsdashboard.traffgoroad.com/, In case it does not work, here is a screen shot. 

[image]

As you can see in the screenshot, where there are camera icons, are basically the locations where the TX2 will go. The idea is that we count the number and type of vehicles and trains at each location of the camera, and stream the data to the dashboard where we further analyze and update the dashboard for current traffic situation. In addition, we want to learn from the data some projections such as future traffic changes, and test hypothesis on data in regards to infrastructure changes. 

As for the German law is concern, By using ODC, we are only interested lebled class data, which is practice does not violate any privacy concerns, as we are not aware of any techniques that can give individual insights. So streaming the number of vehicles, time of encounter, location of encounter is so far a safe bet as not to get into any legal problems.

Would be glad to explain further about the project in case you have more questions.

rraapphhaaeell, 2019-07-02:
Hey @shams3049,

> Would be glad to explain further about the project in case you have more questions.

I'd really like to chat with you about the details on how you're coming along. Would you mind jumping on the phone with me? I'll dm you. :)

tdurand, 2020-05-15:
Refreshing this 😎 . Call for contribution !

We would love to feature the diversity of projects based on OpenDataCam (mobility and non mobility related) that are "privacy respecting" for v3.0.0 release.  We believe showing that you can do plenty of different things with object detections and tracking can bring more diversity to the project and enable people to try to thinker around their use case with OpenDataCam.

It doesn't need to be a "finished", "super polished", "production" project 😉

I'll start first, I am currently doing a project using OpenDataCam and YOLO to detect stains in a small cardboard factory near to where I live.. the goal is to be able to auto discard bad sheets and also store statistic of defects to have an idea of the quality level of the cardboard for each production.

[image] 

From the people active / participating on the repository I remember sharing some details (sorry for the one I forgot, please answer with your project):

- @Denizzje : building a system that enables to count fish and build statistics on fish migration for scientist . 

- @rantgithub : building a project around traffic counting on highways in a district of the USA

- @shams3049 : working on a project for Neuss Harbour to monitor traffic around it (we are already in touch 👍)

- Your project 😎

If you'd like to share a bit more about what you are working on based on OpenDataCam _(it can be very short and does not need to be hyper detailed)_ please ping me on my email tibo.durand [at] gmail.com or here.

haviduck, 2021-03-14:
heya,, thanks for sharing opendatacam with the world. im building a fish counter and classifier controlled by a handheld device for aquaculture. i have a py version inplace, but issues with zmq and chunking is taking too much time so using odc as inferencing rig / mjpeg gen behind a custom gui. added realtime charting and report generation based on your json 

the transition took 1 day due to your friendly api. prototype in place and tested as we speak. 

nice to haves:
1 param for additional data stored along with the recording (location, userid that sort of stuff. 1 column accepting json in mongodb for example). this would allow us to really use the reclist. 

2 classbased roi colors

3 manually set params for darknet detector demo

4 hook for new counts. 

[image]

tdurand, 2021-03-15:
Hi, 

Many thanks for the feedback, sounds a very cool project , i think @Denizzje was working on something similar

> 1 param for additional data stored along with the recording (location, userid that sort of stuff. 1 column accepting json in mongodb for example). this would allow us to really use the reclist.

Could you elaborate on this, not sure how we can make this configurable ?

> 2 classbased roi colors

Not sure what you mean

> 3 manually set params for darknet detector demo

You can already overwrite the .cfg and the weights file, what are you missing ? 

> 4 hook for new counts.

tracked here: https://github.com/opendatacam/opendatacam/issues/258

> 5 iconfonts for class view

Not sure what you mean ? You can already use custom emoji for the classes: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#display-custom-classes

haviduck, 2021-03-15:
> Hi,
>
> Many thanks for the feedback, sounds a very cool project , i think @Denizzje was working on something similar
>
>
> Could you elaborate on this, not sure how we can make this configurable ?
>
>
> Not [...]

its been alot of fun building and researching. :) again, thanks for being awesome and sharing your hard work with the rest of us. ill have to check that out @Denizzje :)

1) well right now you store limited metadata for a recording. in my case i would need location and operator id since they have multiple locations and multiple operators that work the fish farms. im implementing it, just wanted to share my thoughts :)

2) all the bboxes are white, would be sweet to factor in specific colors per class. easier to spot the different classes that way imo.

3) pretty much -thresh and some of the other darknet params

4) sweet!

5)) totally missed that, i was looking at hex icons xD

tdurand, 2021-03-15:
>  again, thanks for being awesome and sharing your hard work with the rest of us

👍

> well right now you store limited metadata for a recording. in my case i would need location and operator id since they have multiple locations and multiple operators that work the fish farms. im implementing it, just wanted to share my thoughts :)

Ok, well if you see a way to do it configurable and want to do a pull request feel free .. Also please note that the development branch has lots of improvements from master

> all the bboxes are white, would be sweet to factor in specific colors per class. easier to spot the different classes that way imo.

not a big fan of this but you could do it pretty easily by tweaking around here: https://github.com/opendatacam/opendatacam/blob/master/components/canvas/engines/LiveViewEngine.js#L33

 > pretty much -thresh and some of the other darknet params

feel free to submit a PR against the development branch to expose the params in the `config.json`  .. would be nice to have indeed.

haviduck, 2021-03-16:
absolutely :) if it works ill share, but  these things have a tendency to bite me in the ass.  have a good one!

tdurand, 2021-05-22:
Making another 📢📢📢 to the community to share the projects they have worked on to feature them on the v4 release !

shams3049, 2021-05-22:
Hey @tdurand I wrote a paper recently using OpenDataCam (its related to my Logistics.NRW peoject). At the moment its peer reviewd will know more by second week of June. If it gets accepted, we can refrance it here to motivate usecases.

willytrueba, 2021-06-22:
After two months without turning off, the Jetson Nano is having a fantastic result. Our Jetson Nano is inside a panel/box, and in the upper part there is a module with the camera, which we use to count vehicles and bicycles for 16 hours every day, and depending on the density we send a message to the panel, storing in a external data base the counter in real time using 4G.

During these two months, both the temperature and the performance of the Nano have been optimal, and there have been no errors in the 4G Nano - BD - Panel communication.

The only problem that we have detected and we are looking for a solution is that bicycles / motorcycles are counted as people. The camera focuses on the road head-on, recording how cars approach head-on, making the bikes look like people (tall and narrow).

If anyone has any ideas on how to solve the problem, we appreciate it.

I leave you a couple of photos. 

[image]
[image]

tdurand, 2021-06-22:
@willytrueba many thanks for this, this is in spain ? are you doing this as a contractor for the local road network administration ?

It is so nice to see what OpenDataCam can enable.. if you have some picture featuring the nano inside the panel it would be awesome..

regarding your problem.. yes this is a know "issue" of the default trained YOLO model.. I'm afraid there is no easy solution.. I guess the solution would be to retrain a custom model with a few hundred (thousands) pictures of bikes + people on bikes to help refine this part, which is definitely not an easy task.. maybe future version on YOLO trained on something else than COCO will bring improvement to this..

Are you running the jetson nano with tiny yolo v4 or with full weight yolov4 ? which other message do you display on the panel depending on how much bicycle ahead ?

geologic, 2021-06-22:
> 
> The only problem that we have detected and we are looking for a solution is that bicycles / motorcycles are counted as people. The camera focuses on the road head-on, recording how cars approach head-on, making the bikes look like people (tall and narrow).
> 

Did you try facing camera 45º instead head on?

haviduck, 2021-06-22:
the easiest way without doing any changes to your app is as @geologic says to change the camera angle.

next id try using other datasets as coco isnt optimized for head-on vehicles.
this for example (in their backup dir)
https://github.com/achen353/Taiwanese-Traffic-Object-Detection

im sure there are alot of pretrained weights available for this.

willytrueba, 2021-07-01:
> @willytrueba many thanks for this, this is in spain ? are you doing this as a contractor for the local road network administration ?
>
> It is so nice to see what OpenDataCam can enable.. if you have [...]

1039 / 5000
Resultados de traducción
Thank you first for your time and help. The panel is in Spain, but at the moment it is only a prototype that we are testing, but the idea is to develop it together with the local highway administration.

As soon as I can, I will show you how the Jetson Nano looks inside the panel. as well as the messages that are displayed. We are going to mount another panel in the opposite direction with another Jetson Nano on a stretch of road that is 8 kilometers long and one panel in each direction.

The idea we have is that the two ODCs are sending the records to the database and taking into account the number of bicycles / cars in each direction, calculate the traffic density and ease of overtaking based on the traffic in the opposite direction to show a message or other.

As for entering YOLO I see it as a second phase, since first I would need to obtain thousands of images or videos to feed.

Right now I am using ODC with tiny yolo 4, based on Balena, which allows me to remotely monitor the status of each jetson nano.

willytrueba, 2021-07-01:
> Did you try facing camera 45º instead head on?

I will try it, but I need time to move the box up the panel.

haviduck, 2021-07-01:
i hope i can share some photos of the end product soon, albeit i scratched odc for my own py /darknet / electron variant, its foundation and idea very much come from this place . im working on an open source version to share and maybe contribute something back along the way :)

napking, 2021-07-15:
Hi!

I'm a transportation engineering graduate from the University of Alberta. I was previously working with a community group here that wanted to compare air quality with nearby traffic levels and they found ODC. We were working towards an experimental setup when I had to step back to focus on my studies.

I since bought my own hardware for ODC for personal uses: I wanted to work with raw traffic count data, to learn about computer vision tech for transportation uses and to learn about the traffic on my street. It's been a bit of a journey to get my setup running, but lots of great learning. I got so far as setting up a Mongo Server on my local machine and turning the ODC on-off manually every day for a week or so. 

I can't remember what exactly, but I hit a roadblock and decided to do a complete re-install. I'd love to be able to make some tweaks to the code (specifically #388 ) but my experience building containers is stalling me further. Current plan is to just get the ODC running, collect some data and focus on the data analysis side.

Here's a screen-cap of my camera view. Not ideal with the side-angle, but placement of counting bars at the screen edges tries to maximize the time objects are in view. The parking lot across the street really hits hard on the tracker data stream though.
[image]

Cheers!

haviduck, 2021-07-15:
> Hi!
>
> I'm a transportation engineering graduate from the University of Alberta. I was previously working with a community group here that wanted to compare air quality with nearby traffic levels and [...]

hey man! im a bit confused about what you are asking. do you need help setting up the conrainer? or do you need help tweaking the source?

regardless, get odc running and use its api to collect data for insight and analysis.
if you want to get down and dirty with yolo, i suggest compiling darknet from scratch and use its python binaries to really get control over whats going on.  

anywah, let us know how we can help and welcome to the wonderful world of computervision :) where nothing is what it seems, rgb is bgr and everyone is dying for a cold beer.

napking, 2021-07-16:
> hey man! im a bit confused about what you are asking. do you need help setting up the conrainer? or do you need help tweaking the source?

I'm not even sure _if_ I'm asking for help. Still evaluating where I'm going next with my device: a) tweak source code to what I need or b) get data for some analysis.

I've dug through the ODC source code and build pathway to learn how it's done. Where I realized I have a steep learning curve is in building my own docker image (aka: writing and executing the Dockerfile). More specifically, I was thinking of tweaking [MongoDbManager.js](https://github.com/opendatacam/opendatacam/blob/development/server/db/MongoDbManager.js) amongst other base-level code. But, well, my brain went to mush following the code both top-down (following the commands in the docker/build/ directory) and bottom-up (follow opendatacam.js and server.js to see what they did). I'm not a heavy programmer, never touched javascript, so I worry about breaking my forked code and not having a solid development environment for debugging.

>if you want to get down and dirty with yolo, i suggest compiling darknet from scratch and use its python binaries to really get control over whats going on.

I don't think I'd ever mess with yolo and the computer vision part. I'll take it for what it is and work with the data that comes out.

>regardless, get odc running and use its api to collect data for insight and analysis.

For sure. I had it running earlier this year but hit some networking issue that I couldn't resolve. I'm probably going in this direction to just get the data collection going and continually updating my local Mongo server. Learning lots!

Cheers!

haviduck, 2021-07-16:
> I'm not even sure _if_ I'm asking for help. Still evaluating where I'm going next with my device: a) tweak source code to what I need or b) get data for some analysis.
>
> I've dug through the ODC [...]

i mean we all kinda operate differently and your learning path is your own. but i will say that compiling darknet these days is a breeze compared to what it was. odc utilize darknet "out of the box" and put js on top of it to perform tracking and counting.  if all you want is raw data i highly recommend compiling darknet, adding the json flag and voila you have a continous stream of detections. as far as i can tell you use pretrained weights for your usecase, and that would still apply here.

that being said, odc is up and running in no time and has its api youve already studied. id recommend moving away from altering odcs spurce but instead write your own handler for its data. as i recall it even has webhooks. reason behind this is mainly cause they are releasing their 4 version shortly and we can expect rapid bugfixes following that.

glad to hear you are learning and enjoying the wonderful, frustrating world of object detection.  hope you get your network issues fixed and able to move on quickly. let us know if you need anything and im sure the experts will come in and help :)

haviduck, 2021-07-26:
[image]

i cant show a whole lot of video of it but thought id share my approach

tdurand, 2021-07-26:
so cool ! if I understood correctly this is custom software based on OpenDataCam or rewrote from the ground up ?

haviduck, 2021-07-26:
> so cool ! if I understood correctly this is custom software based on OpenDataCam or rewrote from the ground up ?

thank you! with darknets recent development its working really well. 
its rewritten using darknets api, but i wouldnt have gotten there if not for odc's approach. was when i saw your tracker js i "got" it. 

i mean, heres the pitch that won.something familiar there, surely. 😄 
[image]

tdurand, 2021-07-27:
Nice ! so glad to know that our work empower other people to solve other use cases even if this is not OpenDataCam directly 😉

haviduck, 2021-07-27:
> Nice ! so glad to know that our work empower other people to solve other use cases even if this is not OpenDataCam directly 😉

additional items in our scope forced me to rewrite but a couple early units is still in use  (image is from early prototyping) and pushing data to our hub perfectly :) have a good summer! 
[image]

zeljkomarinkovic, 2021-08-17:
@haviduck 
Det var kult, god jobb!! 
What kind of gear is your setup running?

haviduck, 2021-08-17:
> @haviduck
> Det var kult, god jobb!!
> What kind of gear is your setup running?

thank you!  :) 
for the nodes, each of them consists of a tiny server with a gpu, ipcams from hikvision and  2 in 1 tablets for operators. the actual hw changes but thats the general hw stack

parthapp, 2021-11-09:
For traffic signal operations, traffic volume counts for through and turning movements are necessary. I plan to use cameras installed at traffic signals to record videos and process them using ODC to get counts and then retime signals. 

Other application will involve Incident management. Suppose there is an incident on the highway and folks are getting on the service road. We would "like" to have a system where a camera is looking at an approach, by matching vehicle ID at line A and B, calculates the speed. If the average speed drops below a threshold over a period of time (eg below 10mph averaged over 3 mins for all vehicles crossing lines A and B), some sort of email/text/notification gets sent out to people managing the signal and they manually trigger special signal timing plans to flush out traffic on service road.

willytrueba, 2021-11-11:
> 1039 / 5000 Resultados de traducción Thank you first for your time and help. The panel is in Spain, but at the moment it is only a prototype that we are testing, but the idea is to develop it [...]

After 160 days working and two cameras counting vehicles in two directions in same road, I give you my impressions.

- 166 days running 24 hours a day
- 470,000 objects counted in one direction per camera
- The day with the most objects was 8150.
- 412,000 cars, 42,000 people / bikes, 5,300 trucks, etc. Sound good.
- Also 1270 umbrellas, 30 TV monitors, 18 airplanes and 17 microwaves. I think that some of its came from night period, because we don't have a good cams for night or bad weather.

The cameras work 24 hours for 3/4 days until the memory is full (CPU over 60% and temperature over 60º are ok) and we have to stop the opendatacam service with balena and restart it.

Currently we count the bicycles (person + bicy objects) that enter on each side of the road section and perform a calculation in real time to send a message to a panel on the road indicating whether low, medium or high density of cyclists.

I'm working on a CRON job to stop/start the opendatacam service if memory is full (Jetson Nano 4GB) and improve the ratio between pedestrians, cyclists and motorists because it almost always takes it as the same type of object

We will start with other idea about animals crossing a road, with a camara on a camera in a straight and define two lines on both sides to send a message if someone enters the road.

nagajavisetty, 2021-12-03:
> After 160 days working and two cameras counting vehicles in two directions in same road, I give you my impressions.
>
> * 166 days running 24 hours a day
> * 470,000 objects counted in one direction per [...]

Hi, I am working on opendatacam project for counting buses, cars, persons, etc. But it is not tracking and counting objects .Do you know the reason?  .

haviduck, 2021-12-04:
> Hi, I am working on opendatacam project for counting buses, cars, persons, etc. But it is not tracking and counting objects .Do you know the reason? .

hey man :) that sounds something odc would fit well for. but i think youll get more answers if you make a separate ticket for this and add some more info. like what model are you using, what does your config look like, how are you deploying it..