opendatacam/opendatacam #85: Credit

shams3049, 2019-06-04:
I have a keynote speech on Thursday at a university in Netherlands. As part of my project relies on this project I want to credit where credit is due. 

Is the content of this [link](https://opendatacam.moovellab.com/#credits) uptodate or any other references that goes you would like to be mentioned.

tdurand, 2019-06-04:
thanks @shams3049 .., yes you can use those, ping @b-g in case he has another recommandation .. Sorry after v2 release everything will be a bit more clear 😉

b-g, 2019-06-05:
@shams3049 That's great! Congrats!

credits ODC v1
```
**Concept & Idea**
Benedikt Groß, Markus Kreutzer, Raphael Reimann

**Software Engineering**
Thibault Durant, Thomas Derleth, Marco Biedermann, Joey Lee, Florian Porada, MESO Digital Interiors

**Product Design**
Olivier Brückner

**Video Documentation**
Raphael Reimann, Thomas 
```

and for ODC v2 probably something like ... still tbd.
```
**Concept & Idea**
Benedikt Groß, Markus Kreutzer, Raphael Reimann, Thibault Durant

**Software Engineering**
Thibault Durant, Florian Porada

... more to come for video and website etc.
```

closed 2019-06-17 by shams3049