opendatacam/opendatacam #88: Auto restart YOLO when running on a file

tdurand, 2019-06-04:
Currently, if we run opendatacam on a file, it will crash after the video has finished... It would be nice to have a way to auto-restart YOLO for example...

legeiger, 2019-06-21:
Maybe this is an easy fix: when creating the docker container from the image there is already the restart=unless-stopped flag passed. If the container would stop with a non zero stop code once the yolo stream ends, the docker daemon would restart it. Maybe this is also a workaround for long time crashes mentioned in #70?  Of course counting lines would be lost..

tdurand, 2019-06-26:
Yes this could be an idea very simple to implement, but I think I will try to restart the YOLO process directly without restarting the whole container so we keep the counting lines and everything is smoother. I imagine when you work on a file you don't want to wait 2 min the docker restart... 

Regarding long time crashes yes, we need to see what we want to do, for now we also start the YOLO / Webcam process only when a first HTTP request (opening localhost:8080) is made.. maybe we want to change this behaviour as well.

Anyway also running from a file isn't a priority for us right now, it's a nice feature but running realtime on a webcam is our main focus..

tdurand, 2019-07-02:
Changed this issue into a more global "Improve run on file support"  , cc @b-g

tdurand, 2019-07-03:
More ideas

- When running from file, should have a frame index instead of timestamp
- Add a "startRecordImmediately" flag on the config.json so we can run Opendatacam on the whole file without missing some frames..

b-g, 2019-07-03:
@tdurand 
RE: startRecordImmediately
yes ... or for now just a basic command line tool/options and no investment in new UI features

tdurand, 2019-07-03:
Agree it is kind of patching something that isn't though for it.. Let's comment here some ideas we can have a decide what to do for v2.1 .

b-g, 2019-07-04:
I guess ... maybe simply?

`$ opendatacam opendatacam_videos/demo.mp4 --video-input=file --config config.json`

Will overwrite the value `VIDEO_INPUTS_PARAMS: file` in the config.json, everything else in config stays.

tdurand, 2019-07-04:
Yes, this would be an improvement of command line to avoid tweaking config.json directly.

tdurand, 2019-07-05:
More thoughs on this, once nvidia-docker support lands for Jetsons, we will be able to simplify a lot the setup and overhead we have right now and just build a nice command line interface to start/stop docker containers, maybe we will have one for Opendatacam with UI / and one for Opendatacam headless to run on a file...

tdurand, 2019-08-08:
Some update on this.. Turns out that darknet doesn't support running on a video file and outputting result as a json file. It does only support this for a single image or for a stream.

It didn't seem that hard to port some of the C code to support also this for a video, so I went ahead and made it: https://github.com/AlexeyAB/darknet/pull/3723 .

With this PR darknet has a new option ( -json_file_output ) to output the YOLO detections on a video file as json... (I'm not sure whether it will be merged as it, or at all, we will see, we have our own fork for stability anyway: https://github.com/opendatacam/darknet )

```
./darknet detector demo cfg/coco.data cfg/yolov3.cfg yolov3.weights file.mp4 -dont_show -json_file_output yolo-detections.json
```

This `yolo-detections.json` is the same format as the one we consume for the Stream (we use this for the simulation mode). It contains only the raw detections and no IDs.

Then I still need to add a specific parser on https://github.com/opendatacam/node-moving-things-tracker to be able to take this as input and output the detections with the IDs. Output would be like https://opendatacam.github.io/opendatacam/apidoc/#api-Recording-Tracker_data but with frame Ids instead of timestamp.

Once this will be done, we will be able to make a specific command that chain those two pieces together and output tracker data from a raw video as JSON... But I still feel we need to discuss what support for file we want as this would be some extra thing to maintain and we may want something more integrated to the current Opendatacam, this also wouldn't support counting things etc etc...

Anyway for now this work has been useful for making progress on : https://github.com/opendatacam/opendatacam/issues/87

b-g, 2019-08-11:
Many thanks for the updates and the great progress! Yes, let's discuss once we are both back in office. Happy 🏝! B

tdurand, 2019-09-18:
Hi @b-g , I've added a basic run on video file support that takes as input a video file and outputs a tracker.json file : https://github.com/opendatacam/opendatacam/blob/master/documentation/HEADLESS_RUN_ON_FILE.md

This is what I think to be most pragmatic way to support this right now (did this in ~1 day), but maybe not the simplest for the end user. Please check this out and let me know what you think.

To further improve this, I may have a way to integrate this into OpenDataCam that would not be that complex.

Right now the main issue is that when we launch darknet in opendatacam, we do no have a way to "hook" the node.js app to it at the very begining on the file processing, the way we hook to it is connecting to a stream opened by Darknet on port 8070, and it's a bit hacky because we don't know exactly when it is open, [we retry until we successfully connect](https://github.com/opendatacam/opendatacam/blob/master/server/Opendatacam.js#L562)  . And even if we connect every 20 ms for example, we sometimes get only the second of third frame.. it is not reliable at all.

But now that we added to darknet a feature to output all the detections in a json file at the end: https://github.com/AlexeyAB/darknet/pull/3723 , when we run on "file" mode, we could use this new feature on top of the current way things work, and when the processing of the file has finished (we detect this when YOLO crashes ...), we can display a dialog that ask the user if he wants to download the whole file processed frames as a json file.

The only downside I see to this is that it is a bit confusing that you do not need to hit "record" , and that this outputted json file wouldn't appear in the "Data" tab.. also we still have this very primitive usage that you need to restart OpenDataCam if you want to run on another file / restart run on the same file... 

In order to improve this we would need to further change how OpenDataCam works and do not start darknet when launching OpenDataCam UI on the current config, instead open a "home screen" gives you the choice to either work on a file or on some other input stream that are configured.. But this would mean that we need to fully manage the darknet process from the node.js app and be able to start / stop / restart it with different params.. Which add more complexity, but also makes OpenDataCam a bit more versatile.

b-g, 2019-09-19:
Hi @tdurand, OK. Got CUDA and NVCC running on my local machine! Yay! 

Quick feedback: Followed then the steps as documented in [HEADLESS_RUN_ON_FILE.md](https://github.com/opendatacam/opendatacam/blob/master/documentation/HEADLESS_RUN_ON_FILE.md) ... However now I think the `node-moving-things-tracker` is buggy, as all coordinates are 0 and bearing is null. (maybe a rounding bug?)

```js
// tracker.json
{"id":0,"x":0,"y":0,"w":0,"h":0,"confidence":97.71,"bearing":null,"name":"car","isZombie":false}
```
Will paste you a link with the example data set via DM.

tdurand, 2019-09-19:
Fixed this in 0.7.1 of https://github.com/opendatacam/node-moving-things-tracker/releases/tag/v0.7.1 , sorry for the bug ! 

More upcoming progress on the run on file end integrated to opendatacam

tdurand, 2019-09-19:
Worked what we discussed today, and was much faster than expected, already have a working version for you to test, in order to install it, you should go to some empty folder and

```
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1-alpha.1/docker/install-opendatacam.sh
chmod 777 install-opendatacam.sh
./install-opendatacam.sh --platform nvidiadocker_cuda_archbin_6_1
```

This will launch opendatacam on the demo.mp4 file as always, but then you can edit the config file and restart the docker container to run on another one

- The infinite looping of the file is working pretty well from my test

- When you hit record, it should trigger a restart of YOLO and immediately record, and then end recording after the file is finished (and re-restart)... But one weird thing from the process manager is that I can only force the restart one time... Then if you hit record again it will do nothing but way the next auto-restart to record the file... I'll try to fix this and if I can't we can just update the UI accordingly and say "recording will start when the file will loop" or maybe we can also have another button "Record file on restart".  This limitation also is happening for manual stopping and starting of the process.. so this is also a blocker for a file selector for example.

- I included the frameId in the recording and persisted data , but the limitation I added is to only persist from frame 50 of the file. As we have this technical limitation of not beeing able to get the very beggining of the stream

- When you hit record or when you have an auto restart, it puts back the screen "Initialising neural network" , we might want to display something slightly different

Let me know what you think... and I'll try to work on another iteration next week but I think the core things are already done.

b-g, 2019-09-20:
Hi @tdurand, Mega!! Just can't get it running! :) See error below ... any idea? Much looking forward to give it a try!

```
opendatacam@opendatacam:~/Documents/test-opendatacam-v2.1-alpha.1$ ./install-opendatacam.sh --platform nvidiadocker_cuda_archbin_6_1
Installing opendatacam docker image
index : 3
Stop any running docker container...
[sudo] password for opendatacam: 
ae3338f8bf1c
8af247b1c90f
618f9e4beeff
d5c38d220b3d
d56eb177475d
b03280e9f7e5
d76cad6dbb8c
26a9859d0f79
75da91411738
5bb6af538480
02d2f0f78482
a91071b791aa
Installing opendatacam v2.1-alpha.1 for platform: nvidiadocker_cuda_archbin_6_1 ...
Download run script for docker ...
--2019-09-20 11:07:44--  https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1-alpha.1/docker/run-nvidia-docker/run-nvidiadocker.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.248.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.248.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 452 [text/plain]
Saving to: ‘run-nvidiadocker.sh’

run-nvidiadocker.sh                             100%[====================================================================================================>]     452  --.-KB/s    in 0s      

Last-modified header missing -- time-stamps turned off.
2019-09-20 11:07:44 (129 MB/s) - ‘run-nvidiadocker.sh’ saved [452/452]

Download config file ...
--2019-09-20 11:07:44--  https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1-alpha.1/config.json
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.248.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.248.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 2240 (2.2K) [text/plain]
Saving to: ‘config.json’

config.json                                     100%[====================================================================================================>]   2.19K  --.-KB/s    in 0s      

Last-modified header missing -- time-stamps turned off.
2019-09-20 11:07:44 (75.0 MB/s) - ‘config.json’ saved [2240/2240]

Create the directory to run on files ...
Download demo video ...
WARNING: timestamping does nothing in combination with -O. See the manual
for details.

--2019-09-20 11:07:44--  https://github.com/opendatacam/opendatacam/raw/v2.1-alpha.1/static/demo/demo.mp4
Resolving github.com (github.com)... 192.30.253.112
Connecting to github.com (github.com)|192.30.253.112|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1-alpha.1/static/demo/demo.mp4 [following]
--2019-09-20 11:07:45--  https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1-alpha.1/static/demo/demo.mp4
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.248.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.248.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 4659459 (4.4M) [application/octet-stream]
Saving to: ‘opendatacam_videos/demo.mp4’

opendatacam_videos/demo.mp4                     100%[====================================================================================================>]   4.44M  11.7MB/s    in 0.4s    

2019-09-20 11:07:45 (11.7 MB/s) - ‘opendatacam_videos/demo.mp4’ saved [4659459/4659459]

Replace config file with platform default params ... (you can change those later)
NEURAL_NETWORK : yolov3
VIDEO_INPUT : file
PATH_TO_YOLO_DARKNET : /var/local/darknet
Download, install and run opendatacam docker container
Create a run-opendatacam.sh script for easy container start
Unable to find image 'opendatacam/opendatacam:v2.1-alpha.1-nvidiadocker_cuda_archbin_6_1' locally
docker: Error response from daemon: manifest for opendatacam/opendatacam:v2.1-alpha.1-nvidiadocker_cuda_archbin_6_1 not found: manifest unknown: manifest unknown.
See 'docker run --help'.

```

tdurand, 2019-09-20:
@b-g Oups, I forgot to push to docker hub, let me fix this.

tdurand, 2019-09-20:
Do, I think you can just retry `sudo ./run-opendatacam.sh`

b-g, 2019-09-23:
Hi @tdurand, Thanks! Was able to run it with the new docker image! Feedback and comments below:

> The infinite looping of the file is working pretty well from my test

Yes the looping was working with the `demo.mp4` file, but with `L1-Tokio.mp4` it nearly crashed my Ubuntu machine. The entire machine became mega slow and I had to ssh into it and stop the docker image. Maybe you can give the tokyo file a try as well (will post a link via DM)?

> When you hit record, it should trigger a restart of YOLO and immediately record, and then end recording after the file is finished (and re-restart)... But one weird thing from the process manager is [...]

Can confirm "the restart one time" bug. Let me know how it goes and whether we really need the work around.

> I included the frameId in the recording and persisted data , but the limitation I added is to only persist from frame 50 of the file. As we have this technical limitation of not beeing able to get the very beggining of the stream

Fine. We just have to mention it in the doc. As I guess that people will rather process long files this shouldn't be too bad.

> When you hit record or when you have an auto restart, it puts back the screen "Initialising neural network" , we might want to display something slightly different

Yes, a hint that the app is about to process `file-xyz.mp4" would be great!

**Bugs**
After I press "start recording" and the video loops, the video stream gets stuck ... but the bounding boxes keep moving.

[image]
[image]

tdurand, 2019-09-24:
Hi ! Many thanks for trying it out, some comments

- I had tried with another file ( 1 min long also), and didn't had this issue... If this file the 1 min one from beat the traffic or something longer ? Does it also lags with the previous version of opendatacam ? Or after a few loops ? It might be that darknet isn't cleaning up properly memory... I ran in some crashes while trying out things also.. I must admit I don't have many solution to this.

- Yes, I'll look if possible to better handle the process lifetime

- Bug of video stream freezing. does it happen all the time ? What is happening here is that when darknet restarts, the MJPEG stream closes and the front-end video stream freezes, but this shouldn't happen as we go back to the loading screen and show back the stream only when darknet it up again. I'll try to repro.

b-g, 2019-09-24:
Hi @tdurand,

> I had tried with another file ( 1 min long also), and didn't had this issue... If this file the 1 min one from beat the traffic or something longer ? Does it also lags with the previous version of [...]

After first loop. File is ca. 2:30 min. See DM.

> Bug of video stream freezing. does it happen all the time ? What is happening here is that when darknet restarts, the MJPEG stream closes and the front-end video stream freezes, but this shouldn't [...]

Yes happens all the time with demo.mp4 and the Tokyo one. Will check as well tomorrow once more. I think it could also be a conflict with another docker ODC container running.

tdurand, 2019-09-24:
Was able to dedicate some time to this this afternoon, and release a new version `v2.1-alpha.2` :

```
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1-alpha.2/docker/install-opendatacam.sh
chmod 777 install-opendatacam.sh
./install-opendatacam.sh --platform nvidiadocker_cuda_archbin_6_1
```

- Was able to fix the restart only working once.. Was a mistake on my end.. Now it works, and also this open the door to be able to potentially restart on any file and add a file control in the UI

- Added an UI indication when we are restarting on a file to start recording

- Was unable to reproduce the freezing of the video stream after a restart, on my end this works smoothly.. That said if the FPS speed is over 30 FPS, it might bug a bit and get out of sync but not freeze.

- Was able to reproduce the freezing of the host machine after running a while on the Tokyo file.. Actually it seems to have nothing to do with the restart as on my laptop it crashing after running 5 min or so..  Good news is that it doesn't seem to happen when we run without docker....  

Still need to investigate more and do a few tests:

- Try to use the new docker --gpu support and see if it crashes also
- Try to update darknet in the docker image to a newer version
- Try to run on this same file with opendatacam v2.0.0 and see if it also freeze.

b-g, 2019-09-27:
Hi @tdurand, just tested `v2.1-alpha.2`. Many thanks for this! Getting there:

- image is frozen also with the normal video file even before I press "start recording"
- hint on pre recording (Initializing View) could be improved by adding the filename. I would also change the text to "Restarting to process video file {name.mp4}"
- data view: would also be nice to add to the result packages the filename (or that the input was a webcam, maybe even the model which was used e.g. tiny-yolo)
- probably not related, but the install script gives me two errors (also there is a typo "automaticaly" and "Opendatacam" vs "OpenDataCam")
```
Opendatacam docker container installed successfully, it might take up to 1-2 min to start the node app and the webserver
wlan0: error fetching interface information: Device not found
eth0: error fetching interface information: Device not found
Opendatacam will start automaticaly on boot when you restart you jetson
If you want to stop it, please refer to the doc: https://github.com/opendatacam/opendatacam
```

b-g, 2019-09-28:
@tdurand just quickly: note `L1-Tokio.mp4` is 60FPS ... wasn't aware of it :)

tdurand, 2019-09-30:
- for the image frozen, how fast is the FPS indicator ? Because maybe your GPU is very fast and it can run YOLO at more than 30 fps and cause the stream to bug... 
- data view: yes I agree.. I'll try to add those metadata along with the recording
- yes, those error are not relate.. they work on jetson to get the IP of the device.. not sure why they don't work on your desktop
- I think the FPS of the file itself doesn't change a thing as darknet will process every frame one by one.. it will just be slower on screen.

I'm gonna investigate more this afternoon

tdurand, 2019-09-30:
Some more updates on this.. I found out what was causing OpenDataCam to freeze after a few minutes... It is memory leaking somewhere. Here is a summary of all the tests:

- Tried OpenDataCam v2.1-alpha without docker on a file with infinite restarts .. No leaks
- Tried OpenDataCam v2.1-alpha with docker on a file with infinite restarts.. Huge leak, docker container RAM usage keeps growing until computer freeze.

=> Then verified that recent changes weren't causing trouble

- Tried OpendataCam v2.0 without docker on a file (the tokio one).. no leak
- Tried Opendatacam v2.0 with docker on a file .. Still the same leak .. we didn't identified it previously as the demo.mp4 file was short and didn't reach computer crashing, but with a longer one it does leak and crash

=> So recent changes aren't causing the issue, I then tried to see if the new docker-nvidia version was improving things.. turns out same thing is happening

=> I then tried to see if different type of streams were causing different problems:

- Ran on remote_cam stream with both docker and non-docker ... No leaks for both
- Ran on a usbcam with both docker and non-docker .. the docker is causing leaks, the non-docker one isn't (seems memory usage increases also but much slower)

The fact that running on a remote_cam stream doesn't seem to leak inside docker made me realized that the potential issue is with OpenCV... and not docker or darknet.. And I just realized that in the docker install for nvidiadocker, the OpenCV version is 3.4.5 , whereas right now on my laptop I have 3.3.1 so maybe in 3.4.x there are issues with leaks for some video input like file / webcam..

So now I'll try to run with the same OpenCV version on the laptop than the docker one to see if I can reproduce the leak... Also it would be great to perform all those tests on a Jetson... On the docker image build of the jetson the OpenCV version if 3.4.3 I think.

b-g, 2019-09-30:
@tdurand Many thanks for the investigation! Then lets hope a different opencv version will make things better.

> for the image frozen, how fast is the FPS indicator ? Because maybe your GPU is very fast and it can run YOLO at more than 30 fps and cause the stream to bug...

With the `L1-Tokio.mp4` (60FPS) the FPS indicator seems to go sometimes to more than 30 eg. 34FPS. But I converted the video to 30FPS and have the same problems. Bounding box seemed to be streamed in correct time but the video stream is extremely slow e.g. 1 FPS ... and video image and bbox are extremely out of sync.

[image]

> yes, those error are not relate.. they work on jetson to get the IP of the device.. not sure why they don't work on your desktop

I think I know why now: My standard Ubuntu 18 LTS came not installed with `ifconfig` :) According to this blog post "[If you’re still using ifconfig, you’re living in the past](https://ubuntu.com/blog/if-youre-still-using-ifconfig-youre-living-in-the-past)" we should rather be using something like `ip address show` ...

tdurand, 2019-09-30:
- Yes, I'm currently compiling several opencv versions to do more tests... Let's hope we find something related to this.. it might also help with https://github.com/opendatacam/opendatacam/issues/95

- If you convert the video to 30 FPS it doesn't change a thing as infortunately it seems that is is not the FPS of the video that matter but how fast can darknet process.. for example for tiny YOLO I have 50 FPS.. But with yolov3 full weight it is max 13 FPS on my laptop

-  For the out of sync this is weird I can't reproduce , did you try with chromium also ? I will try with Firefox.. 

- hehe thanks for the search for ifconfig .. will have a look to replace it

b-g, 2019-10-01:
Hi @tdurand!

> Yes, I'm currently compiling several opencv versions to do more tests... Let's hope we find something related to this.. it might also help with #95

OK + Good luck! But also wondering whether the potential DeepStream implementation relies on opencv? Would be a pity to "waste" time on it and then having a total different way of dealing with video files and webcams. cc @anasBahou

>    For the out of sync this is weird I can't reproduce , did you try with chromium also ? I will try with Firefox..

Nice guess! Yes it works in Chrome, but not in Firefox :)

[image]

tdurand, 2019-10-01:
- Yes I know, it sucks to spend time on those low level issues, but I guess it's the most quick win path as I just was able to reproduce the memory leak by getting OpenCV 3.4.5 on my laptop, and I get the same issue than in docker.. So it's definitly the problem here... 

- Regarding deepstream v4, seems it uses opencv 3.3.1 as a dependency: https://docs.nvidia.com/metropolis/deepstream/plugin-manual/index.html#page/DeepStream_Plugin_Manual%2Fdeepstream_plugin_faq.html   so my next step is to compile opencv 3.3.1 with CUDA+Gstreamer support and see if we have a leak.. I think it will work as it was my also default version on ubuntu ... 

- Also on jetsons we currently use OpenCV 3.4.3  .. so maybe it has also this leak and we should fallback on opencv 3.3.1 and solve some reliability issues on our way... BTW The reason we used a custom opencv build on jetsons and not the default one in the first place was to have gstreamer support, and we did just use the latest version available at the time.. but seems it may have introduced new problems. Interesting also if you have a look at Deepstream v3.0 opencv dependency for dGPU it was 3.4.x , and they also fallback on 3.3.1 on v4...

- Will have a look at that firefox issue...

I guess if I verify that opencv 3.3.1 with gstreamer support is fixing this Leak, re-packaging Opendatacam for v2,1 with this version and improving reliability is still worth it, as migrating fully to deepstream require much more changes and work and would I think be more a Opendatacam v3..

b-g, 2019-10-01:
OK + Great! Glad that you could reproduce it! 3.3.1 seems the way forward.

tdurand, 2019-10-01:
Some news.. Unfortunately it is always more complex than expected.. 

Compiled OpenCV 3.3.1 with gstreamer and ran darknet with it .. But still had the same leak.... So problem didn't seem to come from Opencv version as I previously had on my ubuntu machine also version 3.3.1 and it didn't leak.

As I didn't fully remember how I did build/install this previous OpenCV version.. I had a look at which Video I/O "backend" you can build into OpenCV.. and there are a few options: https://vovkos.github.io/doxyrest-showcase/opencv/sphinx_rtd_theme/page_videoio_overview.html , one of them if Gstreamer, but there is also FFmpeg of course... (from what I understood, only these two can handle everything, reading file, webcam ...)

I tried to do a OpenCV build with FFmpeg  as Video I/O backend.. and leak seems to be fixed..  _(didn't try in a docker image yet.., but expect it to work as I'm able to reproduce all docker - non docker environments)_ ... This would mean that Gstreamer is part of the issue here... which I'm not really happy about as I did include gstreamer with OpenCV in the first place to be able to handle webcam resolution... and I think the pipeline for raspberrypi cam support on jetson requires gstreamer... And also Nvidia Deepstream has it as a dependency and not ffmpeg.. So it should work ;-) ...

Some further tests that would need to be done / in progress:

- (In progress) Compile OpenCV with both FFMPEG and GStreamer support.. and see if leaks is still happening when running on a file.. This would be great if that works out of the box... because we don't want to add an option to darknet to choose which Video I/O depending on which video input.. 

- See if this same leak is happening on the jetson, but I expect it to be there as it was reported by @anasBahou in the first place: https://github.com/opendatacam/opendatacam/issues/95#issuecomment-513714463   .. and if opencv with ffmpeg solves it as it does on Ubuntu Desktop + CUDA.

- Also why is this leak isn't happening when inputing a remote cam stream ? maybe because opencv isn't using Gstreamer to process theses.. 

If this does not conclude.. I'm not very sure the way to go.. Because getting rid of Gstreamer seems to be the opposite way to what Nvidia is doing with Deepstream...

Also this investigation is comforting me even more in the idea that the way forward for Opendatacam is to avoid taking care of these issues by using Deepstream which seem to be a level higher than this giving us directly a way to input a stream / file via Gstreamer .. and outputting a video stream - some detections without having to take care of how it is done.

tdurand, 2019-10-02:
Follow-up after compiling OpenCV with both FFmpeg && GStreamer support..

- It works, not leaks ! But I wasn't able to verify if the gstreamer pipelines still work as I have no webcam around.. so there is still this to check

I did package this in a new docker image, and also no leaks, so please try this alpha.3 :

```
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1-alpha.3/docker/install-opendatacam.sh
chmod 777 install-opendatacam.sh
./install-opendatacam.sh --platform nvidiadocker_cuda_archbin_6_1
```

Next steps is to:

- Verify it works on a usb cam gstreamer pipeline..
- Do the last fixes for the file support on the application side
- Then test this new way of compiling OpenCV on jetson.. First test if we have the same leaks .. and then see if it also fixes things a bit to be able to run on a file on jetson also

b-g, 2019-10-02:
@tdurand Great! Gave it a spin!

- in file mode it runs with 60FPS on my beefy desktop machine with the standard video file. It seems no longer to respect the FPS of the video. Which is great as much faster to process a file
- video stream works with chrome
- video stream works still buggy with firefox. After the video has looped for the first time the stream of the video is broken, but the bboxes are still runnning
- the usb cam didn't work out of the box for me with
```
"usbcam": "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink"
```

tdurand, 2019-10-02:
@b-g , great, I was actually writing a new update ;-) .. You didn't mention if it solves the biggest issue , does it run for a long time without crashing ?? 

More on this, with an v2.1-alpha.4, which should answer your other doubts

```
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1-alpha.4/docker/install-opendatacam.sh
chmod 777 install-opendatacam.sh
./install-opendatacam.sh --platform nvidiadocker_cuda_archbin_6_1
```

Added some application fixes:

- display filename on the restarting screen
- fixed bug stream refresh with firefox (I believe)
- persist confidence score

And also with this build memory shouldn't leak and you should be able to run indefinitely on the file, which would be great if you can confirm

With this we should be pretty close to what we wanted to reach in the first place for the run on file support.

BUT some limitations to know...

- For now this OpenCV build I do with both Gstreamer and FFmpeg support actually doesn't work with gstreamer .. So the bad thing is that you can use this container to run on a webcam or a remote cam .. I don't give up on this yet.. but it's pretty time consuming to try out different way to build OpenCV.. (As you have seen)

- That said, I did some test on jetson also, and the actual Jetson build of Opendatacam  doesn't seem to have this big leak ( ie: the build of openCV)... and can run on a file without the issue we encounter on Desktop ... Which is giving me hope of being able to build a non leaky OpenCV on desktop also...

To summarize right now if we were to release this 2.1 version of opendatacam:

- On Desktop : It would only support run on file 
- On Jetson: It would support everything

b-g, 2019-10-02:
Hi @tdurand! Many thanks for the new version and the summary! Will let it run overnight ... (+ Firefox bug seems to be fixed). Will post feedback tomorrow.

tdurand, 2019-10-02:
Ok great ! Ok, overnight I do not guarantee that it will work ;-) , but at least I think previously it was crashing after 20-30 min no ? On my laptop with the Tokyo video it was crashing after only 8 min I think.

BTW, I think I finally found the solution for the OpenCV build ! So we should have all the low level stuff sorted out... It may not help much will long term reliability on jetson though, as I think in the end that particular leak was something else than the potential leaks mentionned  in #95

anasBahou, 2019-10-02:
Hi @tdurand, @b-g . Sorry for not responding to your questions. I'll try to find some time to do so in the weekend or by next week. 
Thanks.

tdurand, 2019-10-02:
@anasBahou , thanks ! no problem ! I think this issue is mostly under control, and you can focus on feedback on this: https://github.com/opendatacam/opendatacam/issues/112

tdurand, 2019-10-02:
Ok, I did build another one, the v2.1-alpha.5  ..

The only change from alpha.4 is the OpenCV build, which now should support everything (webcam, remote stream ) without leaking when running on file ! Yay finaly found the way !! [Documented here how to compile OpenCV properly](https://github.com/opendatacam/opendatacam/blob/evo-runfromfile/documentation/USE_WITHOUT_DOCKER.md#optional-how-to-compile-opencv-with-gstreamer-support-on-desktop)

Keep in mind that if you want to run on a usb webcam from your desktop, [you should add this flag `--device=/dev/video0:/dev/video0`  in the run_nvidiadocker.sh script ... ](https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#run-from-an-usbcam)

```
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1-alpha.5/docker/install-opendatacam.sh
chmod 777 install-opendatacam.sh
./install-opendatacam.sh --platform nvidiadocker_cuda_archbin_6_1
```

Next steps to make this run on file feature happen in 2.1:

- Finalize feature .. I think I have everything we discussed done.. But maybe we can think of other improvements.. The next step would be to be able to change the file without the need to tweak the config.json and without restarting the docker container .. but this is not straightforward to do.. and I think this would maybe need to be part of a bigger change which would be to be able to reload darknet at runtime (via UI) with any of the settings possible, type of video input, yolo weights etc etc..

- The other one is try a fix of memory usage on Jetson, on desktop I don't see memory increase at all on the course of a 15 min run, but on the jetson I see memory increase a little bit like reported on #95 ... To try to solve this I have one thing I'd like to try is to also build a opencv 3.3.1 on jetson the same way we do it on desktop to see if it changes something, as right now we are using 3.3... if it doesn't I would just give up for now on this...

- Improve documentation about running on file.

b-g, 2019-10-03:
Hi @tdurand! Yay! This could be it! Feedback (Ubuntu 18 machine):

**v2.1-alpha.4**
- ran overnight for 17h on a video file. YAY! Everything was fine but a broken video stream (in Firefox)

**v2.1-alpha.5**
- webcam works
- file processing works
- would be nice to have the file name also reflected in data view
[image]
- would be nice to add a `download` attribute to the download links (now you always have to do a "save as")
- random mini detail: would be nice to make the download links consistent
[image]

👉 Will do an overnight test with `v2.1-alpha.5`! Feedback tomorrow. Fingers crossed! Looks very promising!

tdurand, 2019-10-04:
Ok you may try beta.1 .. Normally every features are in.. Also you could try to run on a usb cam a long time and see how it goes.. I've additional headache to make this work properly also on jetsons but getting there..

```
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v2.1.0-beta.1/docker/install-opendatacam.sh
chmod 777 install-opendatacam.sh
./install-opendatacam.sh --platform nvidiadocker_cuda_archbin_6_1
```

tdurand, 2019-10-14:
let's close this , and open new issue for further improvements

closed 2019-10-14 by tdurand