opendatacam/opendatacam #90: OpenDataCam at Google Coral?

b-g, 2019-06-06:
We should keep an eye on Google's new ML embedded device platform Coral ... would be interesting to know whether there is a docker-coral option as well, so that we can run ODC there "out of the box" as well.

Talk:
[Introducing Google Coral – Building On-Device AI (Google I/O'19)](https://www.youtube.com/watch?v=Jgm25QdF90A)

😂 Great demo:
https://twitter.com/teropa/status/1136166198733955072

tdurand, 2019-06-06:
Yes, I'm following it also, the architecture is different and no way to run darknet / cuda stuff for now. 

From what I know you can only run tensorflow stuff on it, but yeah YOLO has implementations in tensorflow.. but no quick win infortunately 😉

jogiji, 2021-02-11:
Are we doing this some time in future.. @tdurand I have a few coral devices with me ... from what i have seen we need a replacement for yolo based forever-monitor code for coral... If that is correct i would like to do a hacky build for this..

tdurand, 2021-02-11:
no planned @jogiji , so feel free to work on it.. Yes you need to switch darknet with the coral based YOLO and hook to OpenDataCam with an MJPEG stream & a JSONStream with the detections.