opendatacam/opendatacam #92: Run the software on a computer that is not. tx2

kevinjosue2326, 2019-06-19:
I would like to know if I can install it on a computer other than tx2 or nano, I have ubuntu 18.04 and GPU: RTX Titan 2080 TI. Cuda 10.0, and the specifications that a document in the document as Cudnn 7.3. and tensorRT 5.0, my question is is it possible to run it on a computer with my specifications and if so, what steps would I have to follow because of the architecture and the configurations that are in the repository? They are for another architecture.

My processor is I9-9900K Intel.
Ram: 64 GB.
GPU: RTX TITAN 2080TI

tdurand, 2019-06-26:
Yes it is possible:

- with docker-nvidia , I need to finish the documentation on this and publish a docker image: https://github.com/opendatacam/opendatacam/blob/master/documentation/DOCKER_CLOUDANDDESKTOP.md

- without docker, you need to install all dependencies yourself, and also this is not documented properly yet but it will be something similar to https://github.com/opendatacam/opendatacam/blob/master/documentation/RUN_WITHOUT_DOCKER.md

It is on my todo list for this week to document this, I'll ping you once it is done

tdurand, 2019-06-26:
Some more context on this https://github.com/opendatacam/opendatacam/issues/64

tdurand, 2019-06-26:
It will close this one in favor of #64 , just subscribe to the other one to be updated when ready.

closed 2019-06-26 by tdurand