opendatacam/opendatacam #93: PoseNet 2.0 with TensorFlow.js

b-g, 2019-06-24:
> We just released PoseNet 2.0 with TensorFlow.js, including a much more accurate ResNet-based model!

https://twitter.com/TensorFlow/status/1142111597344821248

shams3049, 2019-06-24:
Is it possible to do transfer learning? from the current models in YOLO to Posenet 2.0!!

b-g, 2019-06-24:
No I don't think so ... but one idea could be to make the detection framework e.g. YOLO configurable with a few prepared and selected ones like Posenet.

tdurand, 2019-06-26:
@b-g great, they don't tell much about the status of the node.js - gpu  but I guess in a few months it will be much better.

And in the browser with webgl it is very impressive

tdurand, 2020-05-05:
Closing this one to clean up and keeping: https://github.com/opendatacam/opendatacam/issues/71

closed 2020-05-05 by tdurand