opendatacam/opendatacam #95: Improve long term run reliability / performance

tdurand, 2019-07-01:
From users, seems after a few hours recording the probability of crash is very high. 

**Investigate:**

- Memory leaks: Tracker[ has one](https://github.com/opendatacam/node-moving-things-tracker/issues/11)

- Run opendatacam with while monitoring memory usage

**Improve restoring from crash**

Giving some crashes will be inevitable,  make restore from crash seemless:

- Docker container auto-restarts
- Opendatacam restarts
- Opendatacam restore all previous settings (counting lines etc etc..)

tdurand, 2019-07-01:
@shams3049 please comment directly on this issue when you get to investigate more this, thanks !

tdurand, 2019-07-01:
FYI I have fixed a memory leak (which I think was very large) in the piece of code we use for tracking: https://github.com/opendatacam/node-moving-things-tracker/issues/11 

It will be included in next beta version (will tag one this week), will be interested in some feedback on whether it improves things.. I have no time to test it myself infortunately

tdurand, 2019-07-11:
The first test that we will need to do is to run https://github.com/AlexeyAB/darknet alone and see if it crashed the same way.. This would mean it is not our wrapper app that has issues.

shams3049, 2019-07-11:
I'll do it on one of my Jetson's but without docker container overnight tonight, and see how long https://github.com/AlexeyAB/darknet lasts.

anasBahou, 2019-07-22:
Hi, I've tested https://github.com/AlexeyAB/darknet on my Jetson Nano without a docker and using a raspberry camera :   the memory usage kept increasing until it crashed in 15min.
I used this command to test: 
./darknet detector demo cfg/coco.data cfg/yolov3-tiny.cfg yolov3-tiny.weights “nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink” -ext_output -don_show -json_port 8070 -mjpeg_port 8090

Did anyone tested this and got the same results??

shams3049, 2019-07-22:
I ran https://github.com/AlexeyAB/darknet  on Tx2 for more then 4 hours, and it worked. Overnight run (longer then 4+ hours) always failed. In All cases, I find the system shut down in the morning, with no clear clue of what might have happened.  In there a system wide logger on TX2 to get an insight on to what happend?!

anasBahou, 2019-07-22:
In my case, it was a memory saturation problem. It might be a memory leak. But, what i can't understand is:  why does it work fine on my computer??

anasBahou, 2019-07-25:
@tdurand @shams3049 I'm trying to install [Deepstream and tensorrt-yolo-app](https://github.com/NVIDIA-AI-IOT/deepstream_reference_apps/tree/master/yolo) to test if they have the same memory problem as  https://github.com/AlexeyAB/darknet 

Do you have any experience with this 2 tools ?? If yes, do you think it would be better to replace darknet with one of them ??

tdurand, 2019-07-28:
@shams3049 So it seems the core darknet framework might be the root of the problem for long term run. Maybe this : https://github.com/rbonghi/jetson_stats could help you with system logging.

@anasBahou 

- Jetson's (and specificaly jetson nano) are more limited with memory so it might be the reason why it crashes on the nano and not on your computer.

- yes Deepstream and tensorrt yolo implem should be more reliable and faster on jetson's but infortunately last time I checked they didn't implement and easy way to run on a Gstreamer stream, and didn't provide a live detections information stream / video stream like https://github.com/AlexeyAB/darknet does. (look at the technical architecture schema on how we interface the darknet implem with our project: https://github.com/opendatacam/opendatacam/blob/master/documentation/DEVELOPMENT_NOTES.md#technical-architecture .)

- I seen you created an issue on https://github.com/AlexeyAB/darknet/issues/3633 on the darknet repo, let us know if you discover anything

Also we need to work on upgrading to Jetpack 4.2.1 which came out finaly.. I don't expect it to improve anything on reliability though.

_BTW we will be mostly inactive on opendatacam for the summer time... Most of us are in holidays, sorry for late reply._

tdurand, 2019-07-28:
@anasBahou BTW you should look into the new https://developer.nvidia.com/deepstream-sdk v4.0 which seems to provide more features... I'll try to find time to properly benchmark it...

anasBahou, 2019-07-28:
@tdurand Thanks a lot for the reply. I recently upgraded to Jetpack 4.2.1 but it didn't improve anything.
I'm testing deepstream v4.0 framework but as you said it's more difficult to use then darknet.
I let you know if I discovered anything.

_Have a nice holidays_

anasBahou, 2019-08-01:
@tdurand I've succefully tested deepstream v4.0 framework on yolov3-tiny with a CSI-camera.
I can save the detections(JSON format) to a directory .
Also, I can create a RTSP stream of the video with detections.
**TO DO :**
- Expose detections stream on "port_num" 
- Convert the RTSP stream to MJPEG stream.

If you have any suggestion about this, let me know.

muukkivi, 2019-08-01:
My set without docker on jetson tx2 jetback 4.2.1 using the logitec webcam in usb hub crashes in a few minutes now. On the other hand running just darknet yolo with yolov2 or yolov3 does not crash. I tried alexeyab without the small change from opendatacam that hides the coloured frames. I ran it overnight. Since i am new to linux and jetson i might have also missed something.

tdurand, 2019-08-02:
@anasBahou Great, I'll let you know when I'll have time to benchmark it myself.. Did you notice any performance boost ? reliability improvements ?

@muukkivi  You mean with the last darknet from alexeyab it ran without crashing overnight ? In the next update of opendatacam I'll update our "fork".

anasBahou, 2019-08-02:
> @anasBahou Great, I'll let you know when I'll have time to benchmark it myself.. Did you notice any performance boost ? reliability improvements ?
> 
> @muukkivi You mean with the last darknet from alexeyab it ran without crashing overnight ? In the next update of opendatacam I'll update our "fork".

I noticed a boost in performance : ~22 FPS (deepstream) compared to ~ 13 FPS  (darknet) using yolov3-tiny. Also, there is no memory saturation/leak. For more details on Deepstream performance check : https://docs.nvidia.com/metropolis/deepstream/4.0/dev-guide/index.html#page/DeepStream_Development_Guide%2Fdeepstream_performance.html%23wwpID0EVHA

muukkivi, 2019-08-02:
I ran just darknet alexeyab on its own to check if it works without opendatacam. I think it was the last one but have to check since i am away from my computer.

anasBahou, 2019-08-02:
@muukkivi did you noticed any increase in memory usage when testing darknet ??
You can use system monitor and track the memory usage of darknet.

muukkivi, 2019-08-02:
I can check that next week, but by its behaviour - the freeze of screen, mouse and disconnection of wifi, it looks like that.

tdurand, 2019-08-06:
> @tdurand I've succefully tested deepstream v4.0 framework on yolov3-tiny with a CSI-camera.
I can save the detections(JSON format) to a directory .
Also, I can create a RTSP stream of the video with detections.

@anasBahou Sorry I didn't comment this

- There is a ways to hook to a RTSP stream on the server side of opendatacam and expose MJPEG, but this might require to run ffmpeg and might cause some performance problems  .... _(Seen this project that does something slightly more advanced: https://github.com/kyriesent/node-rtsp-stream , it exposes a stream over websocket that the client should decode with another lib)_

- Did you see any other output from Deepstream for the video feed ? Maybe gstreamer has some plugin to expose directly a Mjpeg stream ?

- Is saving the detection to an output file the only way depstream provides ? We do not necesarly need to output to a JSONStream, it is what darknet was doing that's why we did hook to this, but any type of stream that node.js can hook to would work.

Seems deepstream v4.0 would be a good candidate to get a big performance boost + reliability.. But this requires to change the core "engine" of opendatacam, so will likely happen on another major release if we go this way.

anasBahou, 2019-08-06:
> @anasBahou Sorry I didn't comment this
>
> * There is a ways to hook to a RTSP stream on the server side of opendatacam and expose MJPEG, but this might require to run ffmpeg and might cause some [...]
@tdurand Thanks for the reply. 
I didn't find any plugin to expose directly a Mjpeg stream.
To save detection,  you  may use a message broker with a custom adapter for HTTP( u need to build it from scratch). Unfortunately, I couldn't do it. So , I tried a different approach.

Yesterday, I succeded to use [http_stream from darknet](https://github.com/AlexeyAB/darknet/blob/master/src/http_stream.cpp) to create only  a JSON Stream. 
I'm trying now to add that to deepstream-app source code. 
I think, I can create a MJPEG Stream the same way as JSON Stream( It may take some time to figure it out).
I let you know how it goes.

tdurand, 2019-08-06:
Sounds good, I probably only will be able to assess Deepstream v4 after v2 release , your work will be super usefull !

tdurand, 2019-08-06:
@anasBahou I created a dedicated issue for the Deepstream v4.0 SDK integration: https://github.com/opendatacam/opendatacam/issues/112 , I took the time to read some of the documentation this morning and run the basic YOLO demo on the nano and I'm very enthousiastic on this... It really sound a good candidate to replace the darknet piece of opendatacam. cc @b-g 

Timeline wise we won't have time to dedicate ourself to this evaluation before beg of september, but please keep us updated with your work on this.

I will let this issue open for discussion on how we can improve long term run with the current architecture.

shams3049, 2019-08-06:
I am working on https://www.balena.io/ for Jetson TX2 IOT (remotly deploy/manage) the docker container for OpenDataCam . Its too early to say anything but I think DeepStream 4.0 would be a good path too. I am going to include it in my path too, and see how it goes.

muukkivi, 2019-10-03:
I have tested the solution now and it freezes ubuntu 18 on my jetson tx2 (jetback 4.2 or 4.2.1) in less than 10 minutes. For testing i've just let the logitech camera to stream my room picture (not vehicles). 

Basically at the beginning of the initiation of localhost in chromium there is about 1500-2000 Mb of RAM and it starts to count down pretty quickly till its out of memory after 5-8 minutes. I've reinstalled jetback several times, tried docker and without docker yolo2 and tiny3 - results the same. In case i install a separate darknet from darknet site and test run only darknet yolo, everything is fine. I am quite new to linux but as far as i understand i go by the instructions and get the camera streaming and counting via opendatacam web but for less than 10 minutes. 

i run camera at framerate 20/1, darknet is compiled in a parallel folder to opendatacam app folder (but this probably changes nothing).

Any ideas? As far as i understand the system should run at least a few hours or a day before crashing.

tdurand, 2019-10-03:
Hi @muukkivi , and many thanks for you  comment and tests.

We are currently also investigating memory leaks problems with the Desktop GPU build here: #88 and  we identified a big leak caused by OpenCV that is very similar to the one you are decribing (freeze after a few minutes).. That said from quick tests, we didn't see that same big leak on Jetsons with the 2.0.0 build of Opendatacam.. So not sure if this would be the issue for you... but it is to consider

Some thoughs / questions about your case and further test you could do to help us pinpoint the problem:

- Do you use our fork of darknet: https://github.com/opendatacam/darknet , or the latest https://github.com/alexeyab/darknet ? 

- You say you don't see leaks when running darknet without opendatacam, could you tell me which OpenCV build you have installed on your Jetson when you compile darknet  ? The default one of Jetpack or our custom one ?  

- When you run "Raw" darknet without opendatacam, which exact command to you use to feed the usb cam as input ? With Gstreamer `"v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink"` or with the `-c 0` flag of the default way to do it ?

muukkivi, 2019-10-03:
Hei
I use your fork - installed as in your instructions. 
default opencv with jetback
"v4l2src device=/dev/video1 ! video/x-raw, framerate=20/1, width=640, height=360 ! videoconvert ! appsink" - quite sure as i recall, will check later at home, if different i will specify. Logitech is on video1 not 0, since 0 is the jetson onboard camera. Framerate 30/1 does not allow the camera streaming, basically it will not start to stream.

tdurand, 2019-10-03:
OK.. To make sure , at some point did you installed our builds of OpenCV (3.4.3) https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md#1-install-opencv-343-with-gstreamer , or not ? 

I'm investigating because they might be causing the memory leak problem.. So if you don't have leak with the default darknet compiled with the Jetpack opencv build.. It might confirm this.. 

But if you had the leak while never installing our OpenCV build ( I speak of running Opendatacam without Docker obviously) .. This would invalidate this..

Sorry if this is not very clear.. Basicaly Opendatacam depends on Darknet that in return depends on OpenCV that depends on FFmpeg / Gstreamer for Video I/O

muukkivi, 2019-10-03:
For sure i tested the docker version before installing "the without docker version". So the problem was there already. The first time i installed opendatacam on docker  ubuntu was just newly installed right after flashed with jetback 4.2.1. The first thing after connecting wifi i installed opendatacam on docker (https://github.com/opendatacam/opendatacam).

What i am not remembering is that did i later install the opencv for the "without docker version" or not. To my mind i skipped it (since i thought opencv is already there) but since after testing the without docker version i also tried to run some other counting-tracking solutions i might have  done it in the process for these solutions. 

In conclusion the leak seems to be there with the default opencv. I can re-test or try something if you need more answers.

tdurand, 2019-10-03:
Ok.. A test that could be interesting is:

- Compile standalone darknet and confirm it is running without Leak .

- Install Opendatacam without docker (but skip the OpenCV part).. But use that particular darknet compiled version as the darknet dependency (by specifying the path to darknet in the config.json). And see if running opendatacam leaks...

tdurand, 2019-10-03:
FYI I was able to reproduce the leak with this configuration on a jetson nano: 

- Jetpack 4.2.1
- Default opencv 3.3.1 from jetpack
- Darknet with branch https://github.com/opendatacam/darknet/tree/opendatacam

If I run darknet on my usbcam with this command ( Using a Gstreamer pipeline)

```
./darknet detector demo cfg/coco.data cfg/yolov3-tiny.cfg yolov3-tiny.weights "v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink" -dont_show -json_port 8070 -mjpeg_port 8090
```

Whereas if I use this command ( which input the camera with `-c 0` flag)

```
./darknet detector demo cfg/coco.data cfg/yolov3-tiny.cfg yolov3-tiny.weights -c 0 -dont_show -json_port 8070 -mjpeg_port 8090
```

Seems I don't get the leak... In this case, under the hood, OpenCV might not use Gstreamer to hook to the camera and might use directly the linux webcam I/O or FFmpeg interface.. not sure but we had a similar problem with #88 .

I'll try to produce a different build of OpenCV to see if I can fix this.

muukkivi, 2019-10-04:
ok, i got a separate darknet test going on for 15 hours and no leaks. But to test the without docker version. Not tested with opendatacam yet as you suggested

tdurand, 2019-10-04:
Great !

- Could you specify which command you used to run darknet ? And which OpenCV had you installed ? Did you monitor memory usage with something like `htop` ?

I'm currently in the process of adding support to input webcam into opendatacam with "-c 0" instead of the gstreamer pipeline .. As from my test this causes the leak.

muukkivi, 2019-10-04:
./darknet detector demo cfg/voc.data cfg/yolov2-voc.cfg yolo-voc.weights "v4l2src device=/dev/video1 ! video/x-raw, framerate=20/1, width=640, height=360 ! videoconvert ! appsink"

 watch -n 5 free -m

i also checked i had at some point installed 3.4.3 opencv. So if you need me to test with the default opencv on jetson then should i flash the jetson or can i somehow downgrade?

tdurand, 2019-10-04:
@muukkivi Ok thanks.. It is weird because In my tests I always get a memory leak  when using  "v4l2src device=/dev/video1 ! video/x-raw, framerate=20/1, width=640, height=360 ! videoconvert ! appsink" as an input .. 

And thanks for the info about OpenCV build , in my case I can reproduce with default jetpack build or our build.. So I think this doesn't change a thing.. The only helpful test would be to point opendatacam to use this instance of darknet that doesn't leak in your case.. and see if you have issues... Which would me it comes from Opendatacam.. but I really don't think so from the tests I'm doing.

I'm currently in the process of producing a 2.1.beta.1 build for Jetson TX2 with the other way to input the webcam.. I'll have it later today and it would be helpfull if you can install it and test.

muukkivi, 2019-10-04:
ok, i tried several times

memory on the other hand does not seem to leak (also not enough time to check before the process stops) but  it works for a few minutes and then gives following:
FPS:7.3
ObjeSend request to connect to YOLO JSON Stream
Something went wrong: connect ECONNREFUSED 127.0.0.1:8070
Too much retries, YOLO took more than 3 min to start, likely an error
0
Process YOLO stopped

Following is the file after initiation command after that i go to browser and refresh the local host:
nvidia@nvidia-desktop:~/cam-base/opendatacam$ npm run start

> Opendatacam@2.0.0 start /home/nvidia/cam-base/opendatacam
> PORT=8080 NODE_ENV=production node server.js

Please specify the path to the raw detections file
-----------------------------------
-     Opendatacam initialized     -
- Config loaded:                  -
{
  "OPENDATACAM_VERSION": "2.0.0",
  "PATH_TO_YOLO_DARKNET": "/home/nvidia/darknet",
  "VIDEO_INPUT": "usb",
  "NEURAL_NETWORK": "yolov2-voc",
  "VIDEO_INPUTS_PARAMS": {
    "file": "opendatacam_videos/demo.mp4",
    "usb": "v4l2src device=/dev/video1 ! video/x-raw, framerate=20/1, width=640, height=360 ! videoconvert ! appsink"
  },
  "VALID_CLASSES": [
    "*"
  ],
  "DISPLAY_CLASSES": [
    {
      "class": "bicycle",
      "icon": "1F6B2.svg"
    },
    {
      "class": "person",
      "icon": "1F6B6.svg"
    },
    {
      "class": "truck",
      "icon": "1F69B.svg"
    },
    {
      "class": "motorbike",
      "icon": "1F6F5.svg"
    },
    {
      "class": "car",
      "icon": "1F697.svg"
    },
    {
      "class": "bus",
      "icon": "1F68C.svg"
    }
  ],
  "PATHFINDER_COLORS": [
    "#1f77b4",
    "#ff7f0e",
    "#2ca02c",
    "#d62728",
    "#9467bd",
    "#8c564b",
    "#e377c2",
    "#7f7f7f",
    "#bcbd22",
    "#17becf"
  ],
  "COUNTER_COLORS": {
    "yellow": "#FFE700",
    "turquoise": "#A3FFF4",
    "green": "#a0f17f",
    "purple": "#d070f0",
    "red": "#AB4435"
  },
  "NEURAL_NETWORK_PARAMS": {
    "yolov3": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov3.cfg",
      "weights": "yolov3.weights"
    },
    "yolov3-tiny": {
      "data": "cfg/coco.data",
      "cfg": "cfg/yolov3-tiny.cfg",
      "weights": "yolov3-tiny.weights"
    },
    "yolov2-voc": {
      "data": "cfg/voc.data",
      "cfg": "cfg/yolov2-voc.cfg",
      "weights": "yolo-voc.weights"
    }
  },
  "TRACKER_ACCURACY_DISPLAY": {
    "nbFrameBuffer": 300,
    "settings": {
      "radius": 3.1,
      "blur": 6.2,
      "step": 0.1,
      "gradient": {
        "1": "red",
        "0.4": "orange"
      },
      "canvasResolutionFactor": 0.1
    }
  },
  "MONGODB_URL": "mongodb://127.0.0.1:27017"
}
-----------------------------------
Process YOLO initialized
> Ready on http://localhost:8080
> Ready on http://192.168.1.92:8080
Success init db
(node:10105) [DEP0001] DeprecationWarning: OutgoingMessage.flush is deprecated. Use flushHeaders instead.

tdurand, 2019-10-04:
Ok.. thanks, can't learn much from this logs.. Seems YOLO crashes somehow..  But it seems we at least confirmed again that the leak isn't coming from Opendatacam.

Also can you try to run darknet with `./darknet detector demo cfg/voc.data cfg/yolov2-voc.cfg yolo-voc.weights -c 1`  ?

I'll post a new Docker image upgrade later today that will use this interface instead of the Gstreamer one.. Would be really helpful if you can test it when I do ;-)

muukkivi, 2019-10-04:
tried
./darknet detector demo cfg/voc.data cfg/yolov2-voc.cfg yolo-voc.weights -c 1
run for 3 hrs, no leaks

tdurand, 2019-10-04:
@muukkivi ok.. Well, what you can try is to use opendatacam from the 2.1 branch: https://github.com/opendatacam/opendatacam/tree/2.1 .. You'll see in the config.json you now can either specify "-c 1" or the gstreamer pipeline as video input . And see if it solve your problem.

I wasn't able to build a docker image yet for jetson for 2.1 , as on TX2 I have a leak with both options...  Maybe because of my OpenCV build... Not sure...

Also I created a dedicated issue for this leak and posted on nvidia and darknet repo: https://github.com/opendatacam/opendatacam/issues/131

tdurand, 2019-10-14:
After trying to investigate this more deeply, I didn't manage to find a solution with our current architecture  based on Darknet / OpenCV / Gstreamer . 

That said, two things were done in v2.1.0

- added support to feed the camera input without using gstreamer: https://github.com/opendatacam/opendatacam/blob/master/config.json#L9 , which in some case seems to improve things

- compiled OpenCV with ffmpeg support on the desktop build of Opendatacam, which fix a memory leak when running on a file

The way forward to improve Opendatacam reliability will definitely be #45

shams3049, 2019-12-09:
I remotely deployed the newest opendatacam docker on a TX2, right now, I dont have any other data, but the RAM usage went to 98% and the device went offline. any idea what would cause the overload, any know memory leaks for docker images on tx2? (just wanted to prep before I physically visit the device)

tdurand, 2019-12-09:
There is a memory leak but I never got to find the source.. Not sure if darknet or opencv is causing it in some setups. Can't help much more than this I'm afraid

shams3049, 2019-12-09:
I am looking into it. 

Do you know how If I can make sure that same error is not logged multiple times. or at-least have some control over the longing frequency.

tdurand, 2019-12-09:
From the opendatacam node app ? Not much besides commenting out the console.log statements

heizie78, 2020-01-10:
I'm also experiencing the memory leak issue. All installs/starts find, but after 15min it becomes very laggy and after about 1hr the system is almost unusable. I'm on a Jetson Xavier with Jetpack 4.3, opendatacam 2.1.0 and usbcam.

tdurand, 2020-03-04:
I'm glad to bring some update on this issue

Two things are going on:

- we had a pretty big contribution from @anasBahou at @AtosInnovationBordeaux of a working integration of Opendatacam with deepstream SDK v4 : https://github.com/opendatacam/opendatacam-deepstream  .  If you follow the readme you should be able to run it, it is very well documented

- we are working on fixing the memory leak issue directly in darknet thanks to a some new update the author made: https://github.com/opendatacam/opendatacam/issues/131 , we are pretty confident this will work and aim to have something testable to share by end of this week 🎉

More generally, we are dedicating some time to make another update of OpenDataCam in the next 2 months and having OpenDataCam more reliable is our top 1 priority for this next release.

tdurand, 2020-03-06:
We just published a new build for jetson nano that might fix the memory leak problem, looking for help to test, see: https://github.com/opendatacam/opendatacam/issues/131#issuecomment-595651705

tdurand, 2020-03-28:
closing this one, as v3 will be much more reliable

closed 2020-03-28 by tdurand