opendatacam/opendatacam #96: UI improvements

tdurand, 2019-07-02:
Known improvements to make to the Web app.

- [ ]  Data view screen: Optimize perfs (currently re-render every tick) + only render current recording if page = 0
- [ ]  Android chrome: bug when returning from input entry, canvas doesn't re-render correctly (going from a tab to another solves it)... Also seems stream out of sync but maybe because of wifi connexion... Maybe add fulllscreen mode
- [ ] Safari mobile on iPhone SE: but with innerHeight and height: 100% not consistent.. try to make something more robust
- [ ] Add routing to be able to use back button

tdurand, 2019-07-04:
FYI I just added a download logs button @b-g 😎 , very good suggestion

llobiancoale, 2019-07-26:
Hi i would like to see an option to resize the video/stream (if it isnt already there) 
And an (auto)restart Funktion if it has connection Problems with an ipcam

tdurand, 2021-05-22:
adding to v4 , will consider this issue as a "fix it or close it" one 😄

vsaw, 2022-09-20:
Won't fix for now. Have not received many requests/complaints about this.

closed 2022-09-20 by vsaw