opendatacam/opendatacam #97: Support use of custom YOLO weights for a standard Docker installation

tdurand, 2019-07-03:
For v2.0.0 release we support this only for a non-docker install of Opendatacam, in order to support it for a standard docker install of opendatacam we should:

- Add a way to mount custom .cfg .weights .data inside the docker container darknet folder when starting the docker container: https://github.com/opendatacam/opendatacam/blob/master/docker/run-jetson/run-docker.sh ( as we do for the opendatacam_videos/ folder )

- Document how to update the config.json file accordingly

It is not hard but it adds too much overhead right now for v2 release for benefits.. People using custom YOLO weights should be able to install opendatacam without docker

MeTaXaS4, 2019-07-10:
why not to have a folder like `configs` and have there the `config.json` and all the `yolo` files and if the files exist then copy them to docker container?

tdurand, 2019-07-10:
Yes, we will do that, it is indeed pretty simple, but do not have time to to it properly for v2 release. 

I want to simplify a bit this mess of install that pulls lots of stuff to something a bit more understandable

tdurand, 2020-06-24:
OK this is finally documented: https://github.com/opendatacam/opendatacam/blob/development/documentation/CONFIG.md#use-custom-neural-network-weights

closed 2020-06-24 by tdurand