opendatacam/opendatacam #100: Test opendatacam v2.0.0-rc.1

tdurand, 2019-07-05:
v2.0.0-rc.1 released 🎉 on all platforms: nano, tx2, xavier and nvidiadocker.

- Documentation reworked a lot, I feel I never wrote as many docs 😎 
- New dedicated [Jetson Nano page](https://opendatacam.github.io/opendatacam/documentation/jetson/JETSON_NANO.html) as those users will be the one that need more guidance (and also because it has lots of specificities), I reflashed my device and built the doc as I was doing thing to get a sense of what issues you can face.
- Last up to dates features (latest docker build was from 1 month & half ago)

Docs are also accessible also via website thanks to github page: https://opendatacam.github.io/opendatacam/

I also organised the roadmap with a 2.1 milestone we can start to work on next: https://github.com/opendatacam/opendatacam/milestone/3 :  Better run from file support, benchmark against MOT challenge, and Improvements to setup when Jetpack 4.2.1 comes out.

Would be good to:

- Have someone flashing nano from scrach and fixing the documentation mistake I can have left
- Proof read all the documentation , feel free to rewrite things / add stuff etc etc..
- Test Nvidiadocker setup on a Ubuntu machine / server with CUDA GPU
- (Low prio), test TX2 and Xavier ...

florianporada, 2019-07-05:
Thanks @tdurand  :) ... I blocked some time for next week to get into the flashing and installation stuff!

MeTaXaS4, 2019-07-05:
I tried what you are suggesting but in the first second there are thousands of logs.

another issue don't use `wlan0` and `eth0` for the ip, i had to change it to `wlp3s0` to make it work...

MeTaXaS4, 2019-07-05:
tried to open {ip}:8080/console and managed to see better log, but there isn't something useful why the video stream don't opens

tdurand, 2019-07-05:
> another issue don't use `wlan0` and `eth0` for the ip, i had to change it to `wlp3s0` to make it work...

This is not very important, it is just to display IP adress when using jetsons.. not very relevant for you.

> tried to open {ip}:8080/console and managed to see better log, but there isn't something useful why the video stream don't opens

I was gonna suggest this, what can make me help you would be the first part of the log before the crasyness.. If you have nothing meaningfull try to restart docker container, first open localhost:8080/console, and then open localhost:8080 on a second tab that makes Opendatacam start, the first tab should capture the first part of the log.

What video input are you running on ? For nvidiadocker it is untested for USB webcam as it was though to run on IP cam from the Cloud or on a file... The docker container might not have access to the usb and might need to run priviligied... In this case try to tweak the run_opendatacam.sh script created by the install process...

BTW does it work on a file with default config setup ?

MeTaXaS4, 2019-07-05:
> I was gonna suggest this, what can make me help you would be the first part of the log before the crasyness.. If you have nothing meaningfull try to restart docker container, first open [...]
video input is a webcam and it doesn't work, how can I tweak it?

> BTW does it work on a file with default config setup ?
it works with the file

tdurand, 2019-07-05:
> it works with the file

Ok great 😁

> video input is a webcam and it doesn't work, how can I tweak it?

I've created this issue https://github.com/opendatacam/opendatacam/issues/102 , I'll add documentation and improve this next week... I'm not sure right now how to do it... Just follow this issue and I'll ping you once fixed. Many thanks for reporting this, we just never tested this use case.

MeTaXaS4, 2019-07-10:
How can we handle issues like that?

```
Nothing detected for frame nΒΊ9594
==== HTTP Stream closed by darknet, reset UI, might be running from file and ended it or have troubles with webcam and need restart =====
Process YOLO error
Error: Cannot stop process that is not running.
    at /var/local/opendatacam/node_modules/forever-monitor/lib/forever-monitor/monitor.js:357:26
    at process._tickCallback (internal/process/next_tick.js:61:11)
```

tdurand, 2019-07-10:
For now you need to manually restart everything... we have an issue for long term reliability: https://github.com/opendatacam/opendatacam/issues/95

florianporada, 2019-07-11:
@tdurand I put some time into testing the new version of the ODC.

Notes:
TX2:
- the installation is still super simple and I don't see a problem with the documentation. Maybe one thing could be that somebody without knowledge about the project could also do an installation process. I will ask around if somebody is willing to do that.
- At least on the TX2 that I'm using is recognizing two cameras with `ls /dev/video*`. Maybe listing them during the installation process could be helpful for the user (maybe a parameter for the install script as well?)

Nano:
- The installation is also straight forward. I added notes to the issues in the flash-guide. 
- One critical thing I encountered is, that downloading the image from docker is very unstable **if** you are using a WiFi Dongle. The initial setup should be made via ethernet IMO.
- In the office we are limited to WiFi only so I will take this issue home and do another test there.
- I will update the case section for the nano in the documentation with some new content later today.

tdurand, 2019-07-11:
@florianporada Many thanks, feel free to update the documentation 😉

felixJets, 2019-07-12:
Hi,

following the instructions for the installation without docker, I had no problems to start opendatacam on my Jetson TX2. ( and I'm not really experienced in this topic, so I guess that is a good sign). 
Just a few comments: 

- I was not able to set up a hotspot following the WIFI_HOTSPOT_SETUP.md instructions ,
  adding the line:  options bcmdhd op_mode=2  to  /etc/modprobe.d/bcmdhd.conf solved the problem
- I was also able to set up a hotspot via command line with these instructions: 
   https://seravo.fi/2014/create-wireless-access-point-hostapd
   https://devtalk.nvidia.com/default/topic/910608/jetson-tx1/setting-up-wifi-access-point-on- 
   tx1/post/4976505/#4976505
   but you can mess up your network settings (like I did) if you are careless and unexperienced, so at least 
   make a backup of your original settings before trying this, and it is not that intuitive, how to change 
    back to normal wifi-mode, so I would recommend the WIFI_HOTSPOT_SETUP.md instructions
- in case people are interested to use other yolo-weights ( e.g. the yolov2.weights) they can do the 
   installation without docker and download the weight files they want to the darknet folder
- you can also change the yolo input size via: cd darknet/cfg , edit the .cfg-file you want,
  change the width and height (only multiples of 32 work, right?)
  changing the width and height to 320 in yolov3.cfg, i got at least 7-8 FPS with yolov3, this should still 
  have a better mAP than 608*608 yolov2
- if you are on the way with your Jetson and you want to make changes( for example in the config.json 
  file), you can access the Jetson in a terminal via: ssh YourUsername@IP-Address  ( which should be 
  10.42.0.1 , if you followed the above mentioned wifi hotspot setup instructions)

I hope this can be a little help for other people in the future. (I am sorry if I mentioned something, which was already said, then I just missed it)

At the moment, I am looking for a convenient way to power my Jetson. I was thinking of a powerbank with controllable voltage and integrated protection,something like :          
 https://www.amazon.com/dp/B076GYGR6M/ref=cm_sw_r_cp_apa_i_wC3QCb122GNTJ  
 which was recommended here:
https://devtalk.nvidia.com/default/topic/1043299/portable-power-supply/          
Has anyone experience with something like that?

And thank you very much for making this project available to everyone! ( it's the topic of my bachelor thesis, so I could write a little bit about my results in the future here #84 )

tdurand, 2019-07-15:
Hi @felixJets , many thanks

> was not able to set up a hotspot following the WIFI_HOTSPOT_SETUP.md instructions ,
>   adding the line:  options bcmdhd op_mode=2  to  /etc/modprobe.d/bcmdhd.conf solved the problem

Could you do a PR adding a Troubleshooting section in the WIFI_HOTSPOT_SETUP documentation ? 

> in case people are interested to use other yolo-weights ( e.g. the yolov2.weights) they can do the
> installation without docker and download the weight files they want to the darknet folder

Yes this is documented here: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#use-custom-neural-network-weights

> you can also change the yolo input size via: cd darknet/cfg , edit the .cfg-file you want,
change the width and height (only multiples of 32 work, right?)
changing the width and height to 320 in yolov3.cfg, i got at least 7-8 FPS with yolov3, this should still
have a better mAP than 608*608 yolov2

We didn't play a lot with this, it's always a tradeoff between FPS / accuracy, we are working on benchmarking better: https://github.com/opendatacam/opendatacam/issues/87 , you are welcome to share your results

> if you are on the way with your Jetson and you want to make changes( for example in the config.json
file), you can access the Jetson in a terminal via: ssh YourUsername@IP-Address ( which should be
10.42.0.1 , if you followed the above mentioned wifi hotspot setup instructions)

Yes, we should document ssh access, also I've recently seen that with jetson nano you can plug it as usb and ssh into it..

> And thank you very much for making this project available to everyone! ( it's the topic of my bachelor thesis, so I could write a little bit about my results in the future here #84 )

Please do !
> At the moment, I am looking for a convenient way to power my Jetson. I was thinking of a powerbank with controllable voltage and integrated protection,something like :
https://www.amazon.com/dp/B076GYGR6M/ref=cm_sw_r_cp_apa_i_wC3QCb122GNTJ
which was recommended here:
https://devtalk.nvidia.com/default/topic/1043299/portable-power-supply/
Has anyone experience with something like that?

We did power Jetson TX2 with external supply for v1, maybe @florianporada can give you some tips

florianporada, 2019-07-15:
Hey @felixJets,

thanks for testing the process and for all the thoughts you've written down! 👍 

Regarding the power supply topic: As long as the external power supply (or the powerbank) matches the criteria of the board, I see nothing wrong with using the powerbank. 
You can also cross-check the specs with this thread: https://devtalk.nvidia.com/default/topic/1006369/jetson-tx2/power-requirement-for-jetson-tx2-development-board/

There you can read some good infos regarding the power consumption.
hope that helps! 🙂

felixJets, 2019-07-15:
@tdurand @florianporada, thanks for the answer.

I did a PR ( I hope it is okay, I'm not very experienced in editing stuff on Github)

I think I will give the above mentioned powerbank a try ( I will give some feedback on it after testing)

muukkivi, 2019-07-23:
> > it works with the file
> 
> Ok great 
> 
> > video input is a webcam and it doesn't work, how can I tweak it?
> 
> I've created this issue #102 , I'll add documentation and improve this next week... I'm not sure right now how to do it... Just follow this issue and I'll ping you once fixed. Many thanks for reporting this, we just never tested this use case.
> 
> We are testing 95% of the time with jetsons

try without framerate=30/1, width=640, height=360, just remove these and check again with cam.

tdurand, 2019-08-29:
Many thanks to everyone for the help to test opendatacam v2 , we just released a new website featuring this new version, https://opendata.cam and stay tune for [upcoming updates](https://github.com/opendatacam/opendatacam/issues?utf8=%E2%9C%93&q=is%3Aopen+is%3Aissue+milestone%3Av2.1)

closed 2019-08-29 by tdurand