opendatacam/opendatacam #101: NVIDIA Jetson Nano Case Build

florianporada, 2019-07-05:
### Instructions

**Here are some steps to set up the Jetson NANO in the [Wildlife Cam Casing from Naturebytes](http://naturebytes.org/our-tech/).**

[image]

The casing is originally designed for the raspberry pi 3. The good thing is, that the form-factor of the nano board is not that different so with some simple modifications of the base plate, the Jetson board will fit in this rather cheap casing.

[image]

In the above picture you can see the hand drilled holes (marked in red) and the existing on (marked in blue)

[image]

The easiest way to fit the board on the baseplate is to fix it with one thread screw on the existing hole and rotate the board to the point where it fits perfectly on the baseplate. Make sure to leave some space for the power adapter since it takes a bit of space. ([Link to the adapter](https://www.amazon.de/gp/product/B004US2XPS/ref=ppx_yo_dt_b_asin_title_o02_s00?ie=UTF8&psc=1))

Once it fits mark the spots for the hole and drill it with an electric drill (preferably with an 3mm bit since the screws used are M3s)

Reference with marked hand-drilled holes (red):

[image]
[image]
[image]

When installing the RaspiCam make sure to cover the screws with some electrical tape to prevent short circuits since the board will be mounted on the other side of the plate.

[image]

Front:
[image]

Back: 
[image]

(cc @b-g , @tdurand )

rraapphhaaeell, 2019-07-05:
Awesome docs @florianporada - we should add this to the wiki. I'll also add other ways to build an ODC - feel free to share yours.

tdurand, 2019-07-05:
Great ! Yes feel free to add a markdown file in the /documentation/jetson section: https://github.com/opendatacam/opendatacam/tree/master/documentation/jetson and link it somewhere ;-) .

tdurand, 2019-07-05:
Maybe we can add a "enclosure" folder and create one per Jetson ?

rraapphhaaeell, 2019-07-05:
> Maybe we can add a "enclosure" folder and create one per Jetson ?

yes, also a shopping list per setup would be nice, and a good (in depth) link for the website section on the different models.

tdurand, 2019-07-05:
Yep, I already added shopping list Information for the jetson nano: https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/JETSON_NANO.md#shopping-list . 

Next week I can create one for TX2 and one for Xavier even if it will be more limited as most of what you need comes with the developer kit.

Also this ressource is really nice: https://elinux.org/Jetson_Nano, https://elinux.org/Jetson_TX2, https://elinux.org/Jetson_AGX_Xavier ... It lists compatible cameras, cases etc etc ...

It might make sense to contribute this enclosure to this wiki also.

b-g, 2019-07-05:
@florianporada Thanks! What about scanning the base plate (think of Kopierer) so that you can print it out and simply mark the custom holes? Then you don't have to fiddle around with a ruler and it will be much quicker ...

florianporada, 2019-07-09:
@b-g . This is a good idea! The pictures with the ruler where meant more for an overview.
My initial thought was, that the build included the 'rotation-technique' 😃 .. but I will include the scan of the baseplate.

florianporada, 2019-07-11:
https://opendatacam.github.io/opendatacam/documentation/jetson/HOUSING.html

Further thoughts @b-g , @rraapphhaaeell ? 

otherwise feel free to close this issue

b-g, 2019-07-11:
@florianporada Just checked it. Mega nice documentation to get a nice housing for the Nano! Great! 👏

b-g, 2019-08-13:
Done! Closed with https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/HOUSING.md

closed 2019-08-13 by b-g