opendatacam/opendatacam #102: Allow usbcam access for nvidiadocker docker container

tdurand, 2019-07-05:
I think we need to mount the dev/video on the docker container, or run it priviligied as we do for the jetson docker container wrapper.

MeTaXaS4, 2019-07-08:
I have added that `--device=/dev/video0:/dev/video0` in both `run-docker` and `run-nvidiadocker` but the stream still fails

tdurand, 2019-07-09:
Many thanks, I'll get the time to have a look today / tomorrow, I'll let you know.

tdurand, 2019-07-10:
> I have added that `--device=/dev/video0:/dev/video0` in both `run-docker` and `run-nvidiadocker` but the stream still fails

For me this works, just tested... Do you still have an error ? (as you sent a PR maybe it works for you know)

tdurand, 2019-07-10:
I merged your PR but I didn't enable this --device flag by default because if no usbcam are connect the docker container won't start... So I just documented it: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#run-from-an-usbcam

For v2.1 I've added to the todo list to improve the setup, we aim to just read the config.json file and add the correct flags accordingly: https://github.com/opendatacam/opendatacam/issues/89

Feel free to reopen if you still have issues

closed 2019-07-10 by tdurand

MeTaXaS4, 2019-07-10:
> I merged your PR but I didn't enable this --device flag by default because if no usbcam are connect the docker container won't start... So I just documented it: [...]

add to the documentation that the user needs to start only the script after that and select platform do not download again all the files

tdurand, 2019-07-10:
Yes this is documented here: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#for-a-standard-install-of-opendatacam

tdurand, 2019-07-10:
At some point we will need to migrate to a website for the documentation to be able to add this information for every config modification, but I do not want to duplicate too much on the markdown files.