opendatacam/opendatacam #106: Without docker version freezes on local host after a few hours

muukkivi, 2019-07-10:
Basically ubuntu freezes for at least a 30 to 45min long period at some point while running the opendatacam on jetson tx2 nvidia 4.2 version giving after the freezing following error:

==== HTTP Stream closed by darknet, reset UI, might be running from file and ended it or have troubles with webcam and need restart =====
Process YOLO error
Error: Cannot stop process that is not running.
    at /home/minuga/opendatacam/node_modules/forever-monitor/lib/forever-monitor/monitor.js:357:26
    at process._tickCallback (internal/process/next_tick.js:61:11)

Any ideas

tdurand, 2019-07-11:
Hi, 

Yes, we are aware of issues with long time runs and want to dedicate time in future release on this, see https://github.com/opendatacam/opendatacam/issues/95  for updates on this, also any pointer from your investigations is welcome.

In this case, seems that the Darknet process crashed...

Best

closed 2019-07-11 by tdurand

shams3049, 2019-07-11:
I keep the frames from the feed to 10 FPS, in the config file. In my prospective, there is some clogging over time. I will share a more deeper report in the next few weeks, till then Low FPS in the way to go, speaking from experience,  as the USB feed fails much sooner then the csi camera feed on Jetson TX2