opendatacam/opendatacam #107: Error with message not valid JSON

shams3049, 2019-07-11:
This is new one, any ideas?!

[image]

tdurand, 2019-07-11:
This happens when nothing is detected in the frame I think... I need to improve this.. But it shouldn't impact the app, just pollute the logs

shams3049, 2019-07-11:
I have a feeling it's coming from corrupt JSON file, that happens when the app crashes. 

What do you think about, making the docker image to iteratbale, so if the JSON fails we can manually take out the data and reset the internal memory.

shams3049, 2019-07-11:
Thank you for the context, it makes it more clear for me. I was wrong

I am seeing this error one two devices now, and not able to start the app,

tdurand, 2019-07-11:
Hum... Since upgrading to rc.1 ? I didn't change anything in that part of the code since the beta so I doubt it's coming from a recent change.

tdurand, 2019-07-11:
I've updated our darknet fork 1 month ago: https://github.com/opendatacam/darknet ... this might be the only change.. but yes this could cause issue that you didn't have previously

tdurand, 2019-07-11:
BTW if you install opendatacam without dogger, and you put the Logger variable to true : https://github.com/opendatacam/opendatacam/blob/master/server/utils/Logger.js#L5 , you will be able to better debug this.