opendatacam/opendatacam #111: trying to install opendatacam-2.0.0-rc.1 on jetson nano - darknet issue

repoAI, 2019-07-30:
i trying to install opendatacam on my jetson nano, without the docker (the docker is working fine)

i followed the instructions but when i try to make the darknet, i getting :

chmod +x *.sh
nvcc -gencode arch=compute_53,code=[sm_53,compute_53] -Iinclude/ -I3rdparty/stb/include -DOPENCV `pkg-config --cflags opencv` -DGPU -I/usr/local/cuda/include/ -DCUDNN --compiler-options "-Wall -Wfatal-errors -Wno-unused-result -Wno-unknown-pragmas -fPIC -Ofast -DOPENCV -DGPU -DCUDNN -I/usr/local/cudnn/include" -c ./src/convolutional_kernels.cu -o obj/convolutional_kernels.o
/bin/sh: 1: nvcc: not found
Makefile:154: recipe for target 'obj/convolutional_kernels.o' failed
make: *** [obj/convolutional_kernels.o] Error 127

i newbie , so please be gentle :)

tdurand, 2019-08-05:
Hi, 

Seems CUDA isn't propertly installed on your jetson...

Try re-flashing your jetson nano with latest Jetpack : https://developer.nvidia.com/embedded/learn/get-started-jetson-nano-devkit

tdurand, 2019-08-06:
I'll close this, feel free to re-open if you still have an issue

closed 2019-08-06 by tdurand