opendatacam/opendatacam #112: Evaluate using Deepstream SDK as our main neural network framework

tdurand, 2019-08-06:
https://news.developer.nvidia.com/deepstream-sdk-4-now-available/ 

_This would replace https://github.com/AlexeyAB/darknet which seems to be causing [some reliability issues](https://github.com/opendatacam/opendatacam/issues/95) + enable us to add more features to opendatacam and rely on an official nvidia compatible + tested solution_  

**Features:**

- at least 50% performance inference boost ( 13 FPS > 23 FPS for Jetson nano on tiny yolo v3) , nvidia does some magic to optimize everything to their hardware.
- memory and reliability tested 
- out of the box Docker compatibility with all Nvidia hardware (desktop + jetsons) 
- [plug and play several neural network](https://docs.nvidia.com/metropolis/deepstream/4.0/dev-guide/index.html#page/DeepStream_Development_Guide%2Fdeepstream_app_architecture.html) , for instance seems you can link the output of YOLO to a second neural network that would determine the car color / car type / license plate ...
- Give a way to write our own pipelines, for example they have an example to blur face and license plate: https://github.com/NVIDIA-AI-IOT/redaction_with_deepstream
- Use custom YOLO weights
- Provide video stream on an RTSP protocol

**Missing to interface with Opendatacam**

- Send detections data output to some stream readable by node.js  (JSON stream or websocket ...), seems it only write to a file
- Stream video to MJPEG instead of RTSP... We could also hook opendatacam to RTSP and make the node.js server expose MJPEG (do the proxy in the node.js app instead of in deepstream)

tdurand, 2019-08-06:
WIP from @anasBahou 🙌  : https://github.com/opendatacam/opendatacam/issues/95#issuecomment-518579455 

- Wrote a plugin for Deepstream v4 to send the detection as a stream (based on the darknet implem)
- Investigating the MJPEG stream part

b-g, 2019-08-06:
@tdurand Many thanks for putting the list together! Yes Deepstream SDK seems the way forward, the performance boost would be welcome and the ability to pipe together a pipeline of "processing modules" like YOLO would be mega as well! 🤞

b-g, 2019-08-06:
@anasBahou Many thanks for the "proof of concept"! Yay!

anasBahou, 2019-08-08:
@tdurand I finally succeded in creating both the JSON Stream and MJPEG Stream( using ffmpeg).
I integrated deepstream in Opendatacam project and It works.
I'm still tracking the overall performances. The results on yolov3-tiny configuration : 

- ~20 FPS .
- No memory problem (I still need to run an overnight test). 

BUT, there is only one [little problem](https://devtalk.nvidia.com/default/topic/1058661/deepstream-sdk/nvinfer-is-not-populating-confidence-field-in-nvdsobjectmeta-ds-4-0-/post/5369589/#5369589) : it's related to deepstream framework. Apparently, deepstream doesn't populate the confidence value of the detected objects. So, I'm getting a zero value on all the detections.

tdurand, 2019-08-08:
@anasBahou Wow ! Mega ! The confidence value isn't used at all in Opendatacam besides displaying it on the screen. The tracker doesn't use it... So it won't impact anything.

Will you be able to share back your work on this ? Totally understand if you can't .. I'm gonna be able to have a look at this end of this month / beg september and it would be really helpful

b-g, 2019-08-08:
@anasBahou Mega!! 🎩! Sounds super!

> The confidence value isn't used at all in Opendatacam besides displaying it on the screen. The tracker doesn't use it... So it won't impact anything.

Yes for now  ... but obviously it would be important to have the confidence value for potential further processing of the collected data. We don't know all the use case the users want to achieve with ODC.

Maybe we could simply file an issue to the deepstream framework team to make it clear that this feature would be great?

anasBahou, 2019-08-08:
> @anasBahou Wow ! Mega ! The confidence value isn't used at all in Opendatacam besides displaying it on the screen. The tracker doesn't use it... So it won't impact anything.
>
> Will you be able to [...]

I'd like to share all of my scripts with you but I can't. However, I'd definitely share my method, the key parts of my works and help with some potential issues with integrating this new framework.

anasBahou, 2019-08-08:
> @anasBahou Mega!! tophat! Sounds super!
> 
> > The confidence value isn't used at all in Opendatacam besides displaying it on the screen. The tracker doesn't use it... So it won't impact anything.
> 
> Yes for now ... but obviously it would be important to have the confidence value for potential further processing of the collected data. We don't know all the use case the users want to achieve with ODC.
> 
> Maybe we could simply file an issue to the deepstream framework team to make it clear that this feature would be great?

Already done. See [this](https://devtalk.nvidia.com/default/topic/1058661/deepstream-sdk/nvinfer-is-not-populating-confidence-field-in-nvdsobjectmeta-ds-4-0-/post/5369589/#5369589).

tdurand, 2019-08-08:
@b-g @anasBahou Regarding confidence score.. yes I'm sure it will be fixed on the deepstream side.. thanks for opening the issue... One downside of deepstream seems to be it is closed source or ? 

@anasBahou Ok ! Looking forward to get some insights about this... and please let us know what you keep learning using it (potential issues, limitations, reliability test ...) . We are enthousiatics about moving to Deepstream v4 but the more info we can get, the better ..

anasBahou, 2019-08-08:
> @b-g @anasBahou Regarding confidence score.. yes I'm sure it will be fixed on the deepstream side.. thanks for opening the issue... One downside of deepstream seems to be it is closed source or ?
> 
> @anasBahou Ok ! Looking forward to get some insights about this... and please keep us posted on what you learn using it (potential issues, limitations, reliability test ...) . We are enthusiastic about moving to Deepstream v4 so the more feedback we get, the better ;-)

What do you mean by closed source ?? No github repository??

tdurand, 2019-08-08:
No no, I though the core of deepstream v4 was closed source, but it is not the case from what I see in the forums.. Just didn't see where to download the code yet as I didn't work on it

anasBahou, 2019-08-08:
@tdurand After installing DS4 you can acces the source codes and remake the deepstream main app. There is also many tests provided to customize the framework.

anasBahou, 2019-08-14:
I've finally fixed the confidence problem. Everything seems to work fine. 
Now, I'm running some tests to see if there is something to improve.

b-g, 2019-08-14:
@anasBahou Mega!

b-g, 2019-08-14:
Interesting would be to check with the upcoming Benchmark / Test mechanism whether deepstream has the same performance ...

Work in progress:
https://github.com/opendatacam/node-moving-things-tracker/blob/mot/documentation/BENCHMARK.md

anasBahou, 2019-08-14:
> Interesting would be to check with the upcoming Benchmark / Test mechanism whether deepstream has the same performance ...
> 
> Work in progress:
> https://github.com/opendatacam/node-moving-things-tracker/blob/mot/documentation/BENCHMARK.md

I'll definitely try that later. For know, I think I'm gonna focus on creating a better way to Stream MJPEG  (meaning without FFMPEG).

anasBahou, 2019-08-23:
Hi @tdurand @b-g. Unfortunately, I couldn't create MJPEG Stream without FFMPEG due to some complications in deepstream application. 
Now, I'm trying to handle errors and add auto-restart of Opendatacam.
I have a problem with starting Opendatacam. It always waits for the first detection to start.
I'd appreciate your help to fix it. I created a separate issue to investigate it : https://github.com/opendatacam/opendatacam/issues/115 
Thanks.

akuckartz, 2019-09-11:
I am not happy with dependencies on NVIDIA hardware...

b-g, 2019-09-11:
@akuckartz thanks for pointing that out! :)

But as we already rely on the very core totally on the NVIDIA CUDA api and the GPU supporting it ... I don't see any pragmatic possibility for not having dependencies to NVIDIA in ODC. 

However if we go down the route of the Deepstream SDK (and foster with this the dependencies to NVIDIA even further), then we have to make sure that ODC runs on every hardware with CUDA and not just the Jetson boards ... also any desktops etc. 

Would that be a pragmatic compromise?

anasBahou, 2019-09-11:
> @akuckartz thanks for pointing that out! :)
>
> But as we already rely on the very core totally on the NVIDIA CUDA api and the GPU supporting it ... I don't see any pragmatic possibility for not having [...]

I think we could manage to use both Deepstream and Darknet. The user can just specify the Framework in the "config.json" file.

tdurand, 2019-09-11:
> I think we could manage to use both Deepstream and Darknet. The user can just specify the Framework in the "config.json" file.

FYI Darknet has already a hard dependency on CUDA (to run on GPU) .. hence on nvidia hardware..  A switch to deepstream SDK wouldn't change anything on that end. Deepstream will have the same CUDA dependency.

By the way almost every deep learning framework only works with CUDA: https://towardsdatascience.com/on-the-state-of-deep-learning-outside-of-cudas-walled-garden-d88c8bbb4342  , that said, when there will be more alternative we will be happy to consider supporting them.

tdurand, 2019-10-02:
Hi @anasBahou , I'm trying to evaluate exactly the amount of work to make this happen.. I spent some time in the documentation and trying to figure out what needs to be done. Could you give me some feedback on this ? 

From what I understood, if we were to use deepstream it gives us out of the box a command line tool that:

- Takes as Input :  A file / stream specified as a Gstreamer pipeline 
- Gives as output : Detections data wrote into a file , and also a video stream as a rtsp stream

So we need to build:

**For Inputs :** nothing ? as we can launch the deepstream process instead of the darknet one in the node app

**For outputs:** (In in order to hook opendatacam to it)

- _For Metadata (detections data):_ A way to send the detections as a stream to opendatacam as it seems to only write to a file out of the box... As Deepstream is written in C we could somehow reuse the code of darknet to send to that same JSONStream: https://github.com/AlexeyAB/darknet/blob/master/src/http_stream.cpp and do not change the communication pipeline with opendatacam... Or build something better like a proper websocket connection or something else.. but that would need to be done in C.. 

We would need to build this as a Deepstream plugin right ? How did you do it ? 

- _For Video stream_ : Seems you struggled to build something that produce a MJPEG stream out of deepstream without introducing also ffmpeg to the mix.. But you did and it worked, do you have more details on this ? 

I've also suggested that as deepstream seems to provide out of box a RTSP stream, maybe we can directly consume this in the node.js app  ...  _(Seen this project that does something slightly more advanced: https://github.com/kyriesent/node-rtsp-stream , it exposes a stream over websocket that the client should decode with another lib)_

**- For Docker builds / easy install**

We would need to use this docker image as a base we extend from   https://ngc.nvidia.com/catalog/containers/nvidia:deepstream , and just add in our docker image container our application, node.js, mongodb stuff but do not take care about anything else (right now we need to add OpenCV build + darknet)

There will still be two distinct images for jetsons / desktop .. but they would be very similiar, and since the new jetpack version, the install script of opendatacam should be much simpler.. no need to have different build depending on which Jetsons ...

shams3049, 2019-10-23:
As of today, I have to look into DeepStream to improve stability on my systems on the ground. 

Can you walk me through changes that are required to current architecture to consume RTSP stream from Deepstream. So I can get started accordingly.

b-g, 2019-10-23:
Hi @shams3049,

Great!! But maybe wait until @anasBahou has uploaded his implementation to https://github.com/opendatacam/opendatacam-deepstream. We have managed to figure out the license issue with DeepStream and have the blessing of NVIDIA to go ahead. Anas should shortly upload his code to the new repo. Stay tuned.

@anasBahou any news from Atos?

tdurand, 2019-10-23:
Thanks @b-g, great news !  

Also here are some pointers in case you want to investigate that part .. it can still be useful as from what I know @anasBahou has done it with  a conversion of RTSP to MJPEG with ffmpeg

- You would need to consume the RTSP stream with something like this project: https://github.com/kyriesent/node-rtsp-stream in the node.js server part here: https://github.com/opendatacam/opendatacam/blob/master/server.js#L143

- On the client we would need to change this component: https://github.com/opendatacam/opendatacam/blob/master/components/shared/WebcamStream.js , to consume the websocket endpoint of node-rtsp-stream, on a <canvas> I think

anasBahou, 2019-10-25:
> Hi @shams3049,
>
> Great!! But maybe wait until @anasBahou has uploaded his implementation to https://github.com/opendatacam/opendatacam-deepstream. We have managed to figure out the license issue with [...]

Nothing, I'm still waiting for they reply.

anasBahou, 2019-10-25:
> 
> 
> As of today, I have to look into DeepStream to improve stability on my systems on the ground.
> 
> Can you walk me through changes that are required to current architecture to consume RTSP stream from Deepstream. So I can get started accordingly.

Sorry for not replying sooner, I was very busy.
To create MJPEG Stream, I use FFMPEG to convert RTSP stream from DeepStream. In order to do so, you'll have to change "YOLO.js" code from "opendatacam/server/processes/" so that you execute 3 commands: DeepStream, FFSERVER and FFMPEG. FFSERVER creates a server that does the conversion. It needs some configuration ---> "/etc/ffserver.conf" (output resolution, IP adress, etc). I recommend a maximum resolution of 640x480 to avoid problems of latency. 
As @tdurand suggested, you can use "node-rtsp-stream" project as an alternative if it didn't work. 
As soon as I get the green light from AtoS I'll share my work with you and give further details on it.

Thanks.

prat96, 2019-11-06:
Hi! @b-g @anasBahou , any timeline on when you'll be uploading your deepstream implementation? I'd love to help out and contribute, I'm currently moving my TensorRT implementation of Darknet-yolo to Deepstream as well.

b-g, 2019-11-06:
Hi @prat96, many thanks for the offer! Great! We still are waiting to hear back for the final green lights in terms of some license issue. Sorry! Should be any minute ... but is out of our hands in terms of timing. Fingers crossed!

anasBahou, 2019-12-12:
Hi,

Sorry for keeping you waiting all this time. It's finally moving forward. We're updating the code to the last version of DeepStream so that it can be useful. Please bear with us.

Thanks

shams3049, 2019-12-13:
@anasBahou waiting anxiously, as increasing the runtime has become really crucial in my workflow. 

So far I have no luck with making a stable pipeline with deepstream :(

aswinkumar2019, 2019-12-27:
Hello @anasBahou ,
Can you please specify the approximate timeframe(in weeks or months) required to update the repo?
Eagerly waiting for the update.
Regards,
Aswin Kumar

tdurand, 2020-03-02:
@shams3049 How far did you get ? I'm finally gonna be able to have a bit of time to investigate this. Maybe a quick chat ?

tdurand, 2020-03-04:
Some update for people following this issue, we had a big contribution from @anasBahou at @AtosInnovationBordeaux of a working integration of Opendatacam with deepstream 🎉: https://github.com/opendatacam/opendatacam-deepstream  . If you follow the README you should be able to run it. 

As we don't have many ressources to dedicate to the next version of OpenDataCam , we may not be able to bring this to "production" ready for next release of ODC, but we encourage everyone to check it out. 

For those interested in this only cause of the memory leak problem, please also see: https://github.com/opendatacam/opendatacam/issues/95#issuecomment-594585899

tdurand, 2020-04-01:
I've seen this repo also which might contain interesting bits about deepstream + yolo : https://github.com/marcoslucianops/DeepStream-Yolo cc @anasBahou

Denizzje, 2020-05-15:
Yesterday by coincidence I have tested Deepstream 5 with the help of that repo you posted above. I was able to have 4x 25fps with Yolov3-tiny on with it.

That sounds promising atleast, hopefully they can iron out the kinks which prevent you from fully considering it.

tdurand, 2020-05-15:
@Denizzje thanks for the test, yes seems https://github.com/marcoslucianops/DeepStream-Yolo does a good job to explain how to use Deepstream v5 .

I still have that benchmark pending to see if pain point have been solved from Deepstream v4 ... 

Do you know if:

- is it possible to stream the video feed out of deepstream in some way (MJPEG , RTSP ??) to hook to opendatacam
- is it possible to stream the detections data (bbox) in some way (JSON-Stream or websocket or whatever) ?

Denizzje, 2020-05-15:
1. Yes, you can make it send out RTSP. I tested out a Microsoft Azure + Deepstream sample which contains this section: https://github.com/Azure-Samples/NVIDIA-Deepstream-Azure-IoT-Edge-on-a-NVIDIA-Jetson-Nano#view-the-processed-videos

I succesfully watched the Nano doing its Deepstream things on my main computer over RTSP in VLC. This is Deepstream 4.0.2 by the way.

MJPEG I do not know, I dislike MJPEG because it is so inefficient.

2. In that same sample they send messages to Azure with their model (ResNet I believe)... it has to grab them from somewhere. It aint Yolo, but they have Yolo samples (or not Microsoft but I have seen them... somewhere?). Unfortunally I am a noob so I progress very slow through those things 😆 .

tdurand, 2020-05-15:
Thanks, no problem !! Yes with Deepstream v4 we have a very good idea of the drawbacks / integration pain points.. but not sure what improved with Deepstream v5 😉.

Denizzje, 2020-05-15:
Actually Marcos seems to have updated his Repo considerably 10 days ago, including expanding the readme it seems. He writes this: https://github.com/marcoslucianops/DeepStream-Yolo#custom-functions-in-your-model 

Have you seen it?

tdurand, 2020-05-15:
yes, but not much more than the README.. I think they didn't provide improvement for an out of box way to hook opendatacam to deepstream... You need to write your glue code in C like @anasBahou did here: https://github.com/opendatacam/opendatacam-deepstream. but would need to properly benchmark.. something I don't have the time to do right now.. Also in the coming weeks / months I'm sure other repository will popup with example of application using Deepstream, which might help.

thepycoder, 2020-06-16:
I'm currently working on implementing https://github.com/opendatacam/opendatacam-deepstream in the v5 SDK of deepstream. Currently I have most things working fine, but I'm still running into some issues. Maybe @anasBahou you can give me some pointers?

Currently the performance of the deepstream app itself is very good, I'm getting ~8fps on full blown yolov3 on a jetson nano. ~24fps if you resize the image input to 320x320 instead of the normal 608x608. However, problems arise when working with the streams.

The rtsp stream that comes out of deepstream is pretty bad quality and has a lot of packet loss on my system. I just found a way to force TCP instead of UDP, so I'll see if that helps.
Next to that the ffserver and ffmpeg steps to get the mjpeg aren't working from inside of opendatacam, it does work when I run both manually, albeit with the bad quality mentioned above.
Then the json stream also has problems, mainly in skipping frames, for which no detection is found.
All of this leads to opendatacam only showing the bounding boxes (but hey, the confidence problem is fixed in v5!) and an fps of 8 while deepstream says it's running at 24.

I will keep working on this for the time being, but could use some help on these issues if anyone has time!

tdurand, 2020-06-17:
Great you are working on this, hope you will be able to share back something at some point..

From an external point of view I think investing time in adding a way to send MJPEG stream from deepstream would be great.. Having to install 2 extra dependencies (ffmpeg + ffserver) just to convert back to MJPEG seems adding lots of complexity.. That said @anasBahou gave it a shot and had issues.

Also now deepstream has a Python API no ? maybe easier in Python.

Good luck !

anasBahou, 2020-06-17:
Hi @thepycoder 
Great to hear that your working on the implementation of DeepStream v5.0.
I didn't got the time to try this new version. However here is a few pointers that may help you.

+ **RTSP problem**:
 -  Try replacing RTSP stream with HLS(HTTP Live Streaming) option to export results using TCP connection over HTTP in gstream pipeline using [this link](https://gstreamer.freedesktop.org/documentation/hls/hlssink.html?gi-language=c) 
 - Use the same method in sending detections via JSON stream to send the frames in MJPEG format based on darknet implementation("http_stream.c" file)

For the other issues, I've got to test it myself and debug it to understand it more. Unfortunately, I'm working on another project BUT I might get some time for it later.

I hope this was helpful. 

Good luck !

thepycoder, 2020-06-19:
Hi @anasBahou 
Thanks for the pointers!

I do like your second suggestion on using the darknet code, but I can't help wondering if the MJPEG format is the best one here? Does it not make more sense to adapt the js codebase to accept the x264 RTSP stream instead?
Deepstream can make use of the gpu encoder which is present in the jetson if I'm not mistaken, making x264 a very efficient format to use. 
However, I have no idea how easy/hard this would be to change in the JS codebase.

Also @tdurand, deepstream does have a python api now, but it is slower and not as mature yet. Yolo needs custom layer implementations for example and afaik the Python sdk doesn't have these yet.

However, I do think that deepstream v5 is looking like the solid choice for this repo!

tdurand, 2020-06-19:
> Does it not make more sense to adapt the js codebase to accept the x264 RTSP stream instead?

I found a couple of pointer for this but never tried.. I'm sure you can somehow handle RTSP stream in node.js but for the front-end side from what I know the best is either HLS or MJPEG for cross browser / mobile compat

anasBahou, 2020-07-02:
Hi @thepycoder,

I've got some stability issues with DeepStream 4.x . So, I'm migrating to v5.0 and trying the new OpenDataCam release to get a more robust solution and compare the results. 

I was wondering if we could work together on the deepstream v5 migration. 

Regards,
Anas

thepycoder, 2020-07-02:
I'd love to work together!

I've gotten something working fine with Deepstream 5 already, so I think the main issue will be the RTSP/MJPEG part.
How do I start? Do we create a new branch on the opendatacam-deepstream repo?

tdurand, 2020-07-02:
I added you write access to the repo : https://github.com/opendatacam/opendatacam-deepstream , @anasBahou has them also

tdurand, 2020-07-02:
Feel free to work there ;-)

anasBahou, 2020-07-02:
> 
> 
> I'd love to work together!
> 
> I've gotten something working fine with Deepstream 5 already, so I think the main issue will be the RTSP/MJPEG part.
> How do I start? Do we create a new branch on the opendatacam-deepstream repo?

Great. We can create a new branch on the opendatacam-deepstream repo and start from there. 

We can schedule a Zoom/Skype call to discuss this if you you'd like to. 

Thanks,
Anas

thepycoder, 2020-07-02:
Sure!
You can add me on discord if you want, so we can chat there, schedule a call and keep the issue clean for progress updates :)
I'm ThePyCoder#0381

Thanks for the access @tdurand !

mildsunrise, 2020-09-18:
hi :) I just arrived here, I'd like to do some 5 cents of what has been brought up in this thread regarding the video stream:

### How to stream video to the browser

 - **MJPEG**: Some people have said MJPEG is not a real video codec. This is correct, it's terribly inefficient. However, it's:
   - easy to deploy (no extra ports needed, the MJPEG stream goes through a regular HTTP request)
   - easy to use (just an `<img />` in browser code, and very easy to handle in server too)
   - real-time! it introduces almost no delay
   - widely supported (you don't even have to worry about support issues AFAIK)

 - **HLS**: My advice is to **avoid** it. HLS is not designed for real-time streaming, so you'll usually get at least several seconds of delay, and I understand this is unacceptable for this use case. However I'm not very experienced in HLS, so I could be wrong.

 - **proprietary plugins** (like Flash): this used to be a good option back in the days, but it's not available now.

 - **WebRTC**: Other than crappy things such as mjpeg, WebRTC is the only option for real-time streaming to a browser. It has proper synchronization and uses modern video codecs (so you'll get less data traffic, more FPS and quality). [H.264 constrained baseline is required to be supported on all browsers](https://developer.mozilla.org/en-US/docs/Web/Media/Formats/WebRTC_codecs), which means *you don't even need to do any conversion*, just take the data from the RTSP stream and broadcast it to your WebRTC clients. [Here's a demo in Go you can try](https://github.com/deepch/RTSPtoWebRTC). Since no conversion (transcoding) is needed, this is uses much less CPU than an MJPEG approach. The H.264 stream is efficiently encoded with the GPU and goes to the browser without intermediate steps!

   I think WebRTC is the way to go. However, it's much more complex than MJPEG or even HLS. You'll need to write some javascript for the browser, with i.e. reconnection logic. You'll (maybe) need to open extra UDP ports. The server code to do the forwarding is... not trivial at all. Some browsers have bugs. Mozilla for Android [no longer has the H.264 codec](https://support.mozilla.org/en-US/kb/firefox-android-openh264) so you'll not be able to view the stream there.

TL;DR: My advice is to use MJPEG or WebRTC. The latter is ideal, but has some drawbacks that should be considered first.

### Converting to MJPEG with ffmpeg

We've been able to get a good MJPEG stream, here's what we did:

 - Set a bitrate. If you've tried the proposed solution with ffserver, you'll likely get a crappy image. You need to set `VideoBitRate` to a generous value (i.e. `4000000`) because it's set very low by default.

 - Increase receive buffer size. By default it's set very low (65535), and since the RTSP stream uses UDP for data transfer, you'll likely get missed packets. You need to add i.e. `-buffer_size 2000000` before `-i` when running `ffmpeg`.

 - Reduce FPS. I recommend reducing frame size and FPS, otherwise you'll need a very high bitrate (i.e. traffic) to get decent quality. I don't know what darknet did, but 10 FPS is good IMHO.

 - Avoid ffserver. You don't really need ffserver at all; it creates intermediate files and does a bunch of unnecessary things. Also ffserver was removed some years ago. Instead, it's easier to [modify `mjpegproxy.js` to execute ffmpeg (and read stdout) instead of making an HTTP request](https://github.com/opendatacam/opendatacam/commit/5438ccb8a63007708e6d44c659fa2e6ad90d1166). Everything is done using just 1 ffmpeg process.

If you want to try my solution in the last link, keep in mind it's not 100% polished and depends on #278 and #279.

mildsunrise, 2020-09-18:
Btw, to avoid ffserver there's a tricky thing: you need to know the boundary used by ffmpeg.
It's the string `ffmpeg`, it's [hardcoded](https://github.com/FFmpeg/FFmpeg/blob/6401a5d4b836a08090bc82e5d3101cc1873ae2a7/libavformat/mpjpeg.c#L26) into libavformat.

tdurand, 2020-09-18:
hi @mildsunrise and welcome !

Many thanks for sharing this... I knew something was possible with WebRTC but the complexity always bugged me.. I guess good old MJPEG will do for now..

It seems your are working on an integration of deepstream v5 with OpenDataCam in your fork: https://github.com/opendatacam/opendatacam/compare/development...mildsunrise:deepstream , which seems almost complete ?

If you succeed to do something that works let us now, I can give you access to the https://github.com/opendatacam/opendatacam-deepstream or create a new repo and you can push your work there to share back with people.. or point to you own repo...

tdurand, 2020-09-18:
Just to clarify this:

> I guess good old MJPEG will do for now..

This doesn't mean that a WebRTC approach is out of the question ;-) .. Seems that it would be actually the best approach if we can handle the complexity of integrating it

mildsunrise, 2020-09-19:
Unfortunately, the `deepstream` branch has code I wrote for work. Since Opendatacam has a non-copyleft license, I'm afraid the code can't be used in an open-source project :( I only presented it as an example, at least for now.

tdurand, 2020-09-20:
@mildsunrise ok ! well, any insight you can give from a deepstream implementation is very welcome in the future even if it's not code

joel-schmidt, 2020-10-26:
JSON detection results and RTSP stream of video are not synchronized in opendatacam.

Hello together! :)

Thanks to this issue we managed to connect deepstream with opendatacam. 
We used the patches of @mildsunrise  to create the RTSP stream (thanks for that!). We just send the pure video without displaying the detection results.
However, we send the detection results as a JSON following this repo: 
https://github.com/opendatacam/opendatacam-deepstream/tree/master/deepstream_patch
We added the send_json function to the write_kitti_output function of deepstream_app.c. 
We use yolov4 (integrated following this repo: https://github.com/NVIDIA-AI-IOT/yolov4_deepstream).

The stream and detection results are sent and displayed in opendatacam. The problem is that the video stream and the detection results are not synchronized. The bounding box is, e.g., in front of the detected car (see image).
We've tried to change the resolution, the bitrate and the FPS, but nothing seems to has an effect on the delay.

Does someone experience the same issue or know what causes this problem?

[image]

vsaw, 2020-10-26:
I'm not using deep stream, but I have the same issue. I've also noticed that my video stream sometimes has a slightly different playback speed, which I believe is based on varaiation WiFi signal strength/data rates. This makes me wonder what effects the underlying TCP connection has.

I've also looked into synchronizing the video and detection streams in the client. If we could get the frame ID from the video, we could display the right detections instead of the most recent one we got. However at least with darknet, the MJPG stream does not contain the frame ID and you would have to patch darknet to e.g. add the frame ID to the exif meta data.

[10 more replies]