opendatacam/opendatacam #114: Please add License to this project

thimabru1010, 2019-08-15:


tdurand, 2019-08-28:
I would also vote for MIT , which in my opinion free us of worrying about legal aspects related to this project.. 

As you said @thapliyalshivam it implies that forks / modifications of this project won't necessarily by contributed back, but I guess I would prefer to encourage wide use of this project (and creating a community) , and discussion around it that can be as valuable as actual code contributions.

tdurand, 2019-08-28:
@thapliyalshivam Regarding CONTRIBUTING.md .. Feel free to open a new issue with some ideas / question you have, would help define what to put into it... Let's structure the project contribution guidelines as contribution are coming..

shams3049, 2019-08-28:
**MIT** licence works in our case as well. 

The legal team from our intitute gave us a green light to share our work and data publiclly at the end of the project for undisclosed use case with no ties.  as opendatacam a core component, I tihnk MIT works perfect to expand upon without any liabilities. 

As the use cases for the project are quite diverse, I also think that we need to have a more detailed vision for the project core, as well as how are use cases communicated for the community to take advantage.

thapliyalshivam, 2019-08-28:
I also think we can go ahead with MIT if we want to encourage a lot of people to try the software and get creative. The only issue I can think of is **if someone makes further modifications to this project, they need not open-source those changes** under MIT. If we provide a GPL license, devs using this project need to open-source the modifications they make. It boils down to whether or not we want to encourage the private use of this or not? 

Also, on a different note. I think there should also be a CONTRIBUTING.md for contributors to get started. What do you guys think?

thapliyalshivam, 2019-08-28:
@tdurand 
I also agree on going ahead with MIT so as to encourage wide discussions and usage without getting into legal aspects,
 
Regarding CONTRIBUTING.md, I will open an issue today.

anasBahou, 2019-08-28:
I don't know a lot about licences  but **MIT** works for my case. Whatever you choose I would be happy to contribute as much as possible. 
@b-g I think "contributor" video conference Skype/Zoom/Hangout would be a great way to discuss big issues. Count me in.

dackdel, 2019-08-30:
https://en.wikipedia.org/wiki/WTFPL just going to leave this here as an option. dont mind me.

b-g, 2019-09-04:
Hi @dackdel thanks for the pointer to WTFPL! I think I'm not very fond of WTFPL, I like very much the bit do what you want with it part, but not that you don't have to credit the original project ais I understand it. For me (someone with a background in Art & Design) it is important to get the credits right in the sense of "authorship" ... hope that makes sense for you too :)

hoijui, 2019-09-10:
If you choose a permissive license like MIT, it means others can take your work, improve it a lot, and keep the improvements for themselves/proprietary, potentially selling it for a lot of money.
I personally would not like that for my projects and the work I put in. I want my work to remain as free as possible, and available for as many people as possible.

shams3049, 2019-09-20:
@b-g will it be possible for you to chare the slides from Yesterday. I wanted to add them to my montly presentation, and let every one on my project be informed of progress on OpenDataCam

b-g, 2019-10-17:
Done

closed 2019-10-17 by b-g