opendatacam/opendatacam #119: moving camera

gargle256, 2019-09-04:
Hi,

what is the performance of OpenDataCam with a video stream embedded on a car/bike or a plane ?
Do you stabilize image before detection ?
Thanks for answer.

Best regards,
Philippe

shams3049, 2019-09-04:
Thats actually a really cool test.

I dont think there is any video stabalization steps in the preprocessing of Darknet. Only downsizing, that I am aware off.

gargle256, 2019-09-04:
Hi,
if you want, there is some videos like this one : https://www.youtube.com/watch?v=BlSaYUgKOcA 
to test and improve the software.

Best regards,
Philippe

tdurand, 2019-09-05:
Hello, 

I confirm that there is no stabilization process .. And I think this would be outside the scope of this project as it is a pretty complex thing on his own. That said we could mention somewhere in the documentation some pointers to stabilization techniques.. feel free to open a PRs if you have things to share on this subject.

Also please note that opendatacam is heavily oriented for fixed camera scenarios.. As its main usecase right now is for counting things, if you move the camera the counting lines will also move and results will be useless. Another thing to consider is that the tracker (https://github.com/opendatacam/node-moving-things-tracker)  has also been optimized for fixed viewpoint.. If you move the camera it will likely loose track of the unique ID of objects in the frame... That said I think if the camera viewpoint move slowly, it should still work pretty well ... Would be interested in feedback on this if you get a chance to test.

Also we are definitely interested in learning new usecases that could benefit from opendatacam.. What is your usecase 
 

Best, 

Thibault

closed 2020-01-28 by tdurand