opendatacam/opendatacam #120: CONTIBUTING.md

thapliyalshivam, 2019-09-04:
Hey all, since the project seems to be growing well, I wanted to open this as an idea to add a CONTIBUTING.md to help people guide how to build and contribute to opendatacam. This will also make easier for us to bring some standard in PR raised and making a procedure. A very rough skeleton of the file I thought of is given below. Please add your ideas and thoughts on this. 
1 - Welcome
2 - getting started
3 - How can I help?
4 - Hardware
5 - testing
6 - raising PR

b-g, 2019-09-04:
Hi @thapliyalshivam cc @tdurand,

Many thanks for this! Yes a structure like this could work ... but it obviously depends on the contents of each sections :) 

You might want to browser a few projects which have IMO a nice CONTRIBUTING text e.g.:
https://github.com/ml5js/ml5-library/blob/development/CONTRIBUTING.md
https://p5js.org/community/

I think it also would be nice to incorporate something like the all-contributors standard:
https://github.com/all-contributors/all-contributors

I wrote a couple of weeks ago https://github.com/hfg-gmuend/openmoji/blob/master/CONTRIBUTING.md ... which worked so far quite well. I rather tried to keep it short and sweet :)

@tdurand please feel free to jump in / veto etc.

tdurand, 2019-09-05:
- @b-g Very much like the idea of all-contributors . I think Opendatacam could use lots of contribution coming from non-code contribution, which is already the case.. real world usage feedback, ideas of improvements, testing etc etc... and mentionning people contributing this kind of things would be great.. Seems we can easily add a bot that keeps it up to date: https://allcontributors.org/docs/en/tooling 

- One of the things needed is to rework the [DEVELOPMENT_NOTES.md](https://github.com/opendatacam/opendatacam/blob/master/documentation/DEVELOPMENT_NOTES.md) including more guidelines on how to get started with the codebase... The architecture schema gives a good overview, but then a deeper writeup  / or a screencast explaining a bit deeper how the codebase works could be nice... Especialy explaining where each pieces interface.

- At that point I don't think we need a deeply detailed contribution workflow as the complexity of opendatacam is pretty high, people diving into the code will know how to send  PRs.. 

- Mentionning next milestone links somewhere, but also that we are open to other improvements

tdurand, 2021-05-22:
will try to work on this as part of the v4 release

would like to integrate to https://allcontributors.org/ so we can better showcase the people helping out

b-g, 2021-05-24:
Yes it would be lovely to have a contributor gallery á la https://github.com/processing/p5.js/#contributors ... just we should bee able to credit also non-github people e.g. most people of the old "People involved" are not on Github.