opendatacam/opendatacam #122: Increase/decrease counter depending on direction

pacmanii, 2019-09-10:
Is it possible to increase or decrease counter depending on the direction the object is crossing the line? For example how many cars are in the parking garage or people in the theater?

shams3049, 2019-09-10:
To my knowledge it not possible to decrease the counter, as the traking ID is not that consistant.

The way I approched it, is by having two simultanius counters (short distance appart). That makes it possible to identify the direction from the time log (also gives you speed, if you place them cleaberly, and know the excat distance between the tracker lines).

tdurand, 2019-09-10:
I think it doesn't matter if the ID is consistent when the car is parked because what @pacmanii seems to be interested here is the absolute value of how many cars crossed the line in one direction or the other. In my opinion this use case should work pretty well with opendatacam.

@shams3049 solution is clever, but I think we could do simpler as the tracker: https://github.com/opendatacam/node-moving-things-tracker does estimate the [bearing of the moving object](https://opendatacam.github.io/opendatacam/apidoc/#api-Tracker-Data) for each position.  With this bearing info and the counting lines coordinates, you can compute at which angle the object is crossing the line and then operate the counting logic the way you want. 

Right now it's a bit hard to access the bearing data in the API from the counter history, as you need for every object of the counter history: https://opendatacam.github.io/opendatacam/apidoc/#api-Recording-Counter_data to fetch the tracker history and find the bearing of this object at that time: https://opendatacam.github.io/opendatacam/apidoc/#api-Recording-Tracker_data . 

One thing that we should do is to expose this bearing info in the counter API: https://opendatacam.github.io/opendatacam/apidoc/#api-Recording-Counter_data , so it's easier to write your own logic on top of opendatacam.

But I also think this feature could also use to be in the integrated in opendatacam , be able to parameter a counting line. By default it would be the actual behaviour of counting in both direction, but we could also add unidirectional counting lines, have a line that count in one direction and decrease in the other ... There is a bit of UX thinking to do here in order to keep things simple..,

tdurand, 2020-03-25:
I'm working on adding this feature to OpenDataCam, it is almost done, here is a [quick screencast](https://www.youtube.com/watch?v=dO-8mRMIjvc) of how it works:

[image]

Re-reading this issue, I think we could add another "mode" along with bidirectional and unidirectional , which would be this increase / decrease request, that we can only do with the API right now .. but I'm not sure of the icon to have for this, any idea @b-g ? 

Also let me know of any improvements you can think of.. will provide a build to test it later this week 😉

tdurand, 2020-03-25:
Thinking a bit more on this increase / decrease, it would mean two icons to specify which side is increasing and which side is decreasing, something like: 

[image]

But I'm sure @b-g can come with something a bit cleaner

shams3049, 2020-03-25:
@tdurand I like this feature the most.

shams3049, 2020-03-25:
@tdurand All I want to test is if we have lines on top of each other(perpendicular), can we detect traffic flow in diffrent direction: forexample: at a traffic light crossing.

tdurand, 2020-03-25:
from what I understand from your use case it should be possible in two ways with this feature:

- Draw two lines with unidirectional counting (for each direction)
- Draw a bidirectional line, and then filter one side or the other thanks to the new attributes in the raw data of the counter

Will release something testable by the end of the week

tdurand, 2020-03-26:
After thinking it over, we decided not to add the specific increase / decrease mode, an only provide the three "modes" explained on the video:  bidirectional & unidirectional both side.. We don't want to overload the core of opendatacam with lots of features and prefer to rely on the API for people to build their own use cases. Also this increase / decrease mode could be solved by drawing two unidirectional lines and compare both counters.

tdurand, 2020-03-27:
Quick bonus feature done this morning, being able to save / import counting lines configuration from json file directly in the UI (previously only possible via API). cc @b-g  😉

[image]

b-g, 2020-03-27:
@tdurand MEGA! 🙏 :)

tdurand, 2020-03-30:
closing this one, let's gather feedback in #184

closed 2020-03-30 by tdurand