opendatacam/opendatacam #123: Question

gmonk, 2019-09-15:
I work in the Traffic Industry at local agency in Arizona.  My question is how much does performance degrade in high temps ? Arizona is know to reach 110+ at certain times of the summer and I was wondering if you have any deployments in high temp areas such as AZ and did those temps impact detection and operation.  I would also like to know which nvidia product is recommended TX2 or Nano for Vehicle/Pedestrian detection and counting.

Thanks

tdurand, 2019-09-15:
Hi !

- I have no answer regarding high temperature... Only tested at "normal" temps. I guess if you reach the point that the hardware need to throttle cause of cooling issues, it might crash... 

- TX2 will always perform better than a Nano as it can run a better config of the neural network. See: https://github.com/opendatacam/opendatacam/blob/master/documentation/CONFIG.md#change-neural-network-weights

b-g, 2019-09-15:
Hi @gmonk, Thanks for the interest! But I think this is not the perfect forum for really detailed hardware question around the Jetson boards. I would recommend to reaching out to NVIDIA directly. Great would if you could post the answer here again. Would love to hear what NVIDIA is saying in terms of high temperatures! :)

shams3049, 2019-09-16:
We had few hot days in Germany this summer, and all our TX2 were shut down within 15 min after we left them in the sun (temprature that day was 42 CC, 107 F).  

We now have a cooling setup with our tx2, but unfortunatly no more hot days in Germany anymore to know if our solution works. 

TX2 can be easially mounted in mini ATX case, which allow for many cooling options.

tdurand, 2019-09-16:
thanks @shams3049 ! @gmonk I'll close this issue but feel free to reopen if you have further questions

closed 2019-09-16 by tdurand