opendatacam/opendatacam #125: Please specify the path to the raw detections file

osfa, 2019-09-17:
trying to follow [this](https://opendatacam.github.io/opendatacam/documentation/USE_WITHOUT_DOCKER.html
) on jetson nano.

getting error on npm start: "Please specify the path to the raw detections file"
found some old threads about ffmpeg? installed ffmpeg3. no luck.

something with a wrong path in config?
where do the "raw detections file" need to be? what is it? darknet seems to run fine.

`"NEURAL_NETWORK": "yolov3",
"PATH_TO_YOLO_DARKNET": "/home/myuser/darknet",`

weights are in `/home/myuser/darknet/yolov3.weights`

thx.

tdurand, 2019-09-17:
Hi, 

Could you paste the full output error ?

Did you run `npm run build` before `npm run start`

I can confirm that ffmpeg is not the issue, since ODC v2.0 we do no rely on it

Thanks

tdurand, 2019-09-17:
PS: _"Please specify the path to the raw detections file"_ isn't a meaningful error, you can ignore it. (It is there because of the node-moving-things package that just print this... need to be fixed)

osfa, 2019-09-17:
not sure. if I run on yolov3 it just freezes. if I use yolov3-tiny it just spews out 'stream stopped! 
' and then "Video-stream stopped!" until I ctrl+c out. been trying to get a better error message for you for awhile now but no luck..

osfa, 2019-09-17:
aha. ok the demo video is not in the repo like for the docker install...

osfa, 2019-09-17:
trying with raspberrycam_no_docker now.

getting:

` JSON-strError with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON`

mixed in with some

`FPS:13.2
 MJPEG-stream sent. 
Objects:

 JSON-stream sent. 

FPS:12.7
 MJPEG-stream sent. 
Objects:`

but after awhile it is just

`Error with message send by YOLO, not valid JSON`

I have verified that raspberrycam works as per documentation.

osfa, 2019-09-17:
```
> Ready on http://localhost:8080
> Ready on http://192.168.1.9:8080
Success init db
Send request to connect to YOLO JSON Stream
Process YOLO started
{ OPENDATACAM_VERSION: [32m'2.0.0'[39m,
  PATH_TO_YOLO_DARKNET: [32m'/home/jbe/darknet'[39m,
  VIDEO_INPUT: [32m'raspberrycam_no_docker'[39m,
  NEURAL_NETWORK: [32m'yolov3-tiny'[39m,
  VIDEO_INPUTS_PARAMS:
   { file: [32m'opendatacam_videos/demo.mp4'[39m,
     usbcam:
      [32m'v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink'[39m,
     experimental_raspberrycam_docker:
      [32m'v4l2src device=/dev/video2 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink'[39m,
     raspberrycam_no_docker:
      [32m'nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink'[39m,
     remote_cam:
      [32m'YOUR IP CAM STREAM (can be .m3u8, MJPEG ...), anything supported by opencv'[39m },
  VALID_CLASSES: [ [32m'*'[39m ],
  DISPLAY_CLASSES:
   [ { class: [32m'bicycle'[39m, icon: [32m'1F6B2.svg'[39m },
     { class: [32m'person'[39m, icon: [32m'1F6B6.svg'[39m },
     { class: [32m'truck'[39m, icon: [32m'1F69B.svg'[39m },
     { class: [32m'motorbike'[39m, icon: [32m'1F6F5.svg'[39m },
     { class: [32m'car'[39m, icon: [32m'1F697.svg'[39m },
     { class: [32m'bus'[39m, icon: [32m'1F68C.svg'[39m } ],
  PATHFINDER_COLORS:
   [ [32m'#1f77b4'[39m,
     [32m'#ff7f0e'[39m,
     [32m'#2ca02c'[39m,
     [32m'#d62728'[39m,
     [32m'#9467bd'[39m,
     [32m'#8c564b'[39m,
     [32m'#e377c2'[39m,
     [32m'#7f7f7f'[39m,
     [32m'#bcbd22'[39m,
     [32m'#17becf'[39m ],
  COUNTER_COLORS:
   { yellow: [32m'#FFE700'[39m,
     turquoise: [32m'#A3FFF4'[39m,
     green: [32m'#a0f17f'[39m,
     purple: [32m'#d070f0'[39m,
     red: [32m'#AB4435'[39m },
  NEURAL_NETWORK_PARAMS:
   { yolov3:
      { data: [32m'cfg/coco.data'[39m,
        cfg: [32m'cfg/yolov3.cfg'[39m,
        weights: [32m'yolov3.weights'[39m },
     [32m'yolov3-tiny'[39m:
      { data: [32m'cfg/coco.data'[39m,
        cfg: [32m'cfg/yolov3-tiny.cfg'[39m,
        weights: [32m'yolov3-tiny.weights'[39m },
     [32m'yolov2-voc'[39m:
      { data: [32m'cfg/voc.data'[39m,
        cfg: [32m'cfg/yolo-voc.cfg'[39m,
        weights: [32m'yolo-voc.weights'[39m } },
  TRACKER_ACCURACY_DISPLAY:
   { nbFrameBuffer: [33m300[39m,
     settings:
      { radius: [33m3.1[39m,
        blur: [33m6.2[39m,
        step: [33m0.1[39m,
        gradient: [36m[Object][39m,
        canvasResolutionFactor: [33m0.1[39m } },
  MONGODB_URL: [32m'mongodb://127.0.0.1:27017'[39m }
layer     filters    size              input                output
   0 (node:7783) [DEP0001] DeprecationWarning: OutgoingMessage.flush is deprecated. Use flushHeaders instead.
Send request to connect to YOLO JSON Stream
[... 654 lines cut ...]
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
Error with message send by YOLO, not valid JSON
```

osfa, 2019-09-17:
I'm having a really hard time gauging whether something is broken or not. The Nano really chokes and sometimes everything slows to a crawl. I'm running with a monitor is that the issue? Is this message not an actual error message with the app? Found something about that in #70 

I never get to the actual application screen on localhost:8080 anyway, just the load screen.

tdurand, 2019-09-17:
Some comments:

- On jetson nano, you should run `yolov3-tiny` , otherwise nano doesn't have the power to run other weights

- From your output, seems everything is working fine, maybe the issue is that it doesn't detect any object.. have you tried to point the cam at your face ? 

Maybe you can try to run on the demo file to make sure this is the issue, in order to do so:

- create an `opendatacam_videos` folder in your darknet repo , and download inside this video: https://github.com/opendatacam/opendatacam/blob/master/static/demo/demo.mp4

- set VIDEO_INPUT to "file" in your config.json

- `npm build; npm start;`

If this works, the issue would be with the cam.

tdurand, 2019-09-17:
Also yes, running with a monitor causes big slowdown... That is why we recommend doing the setup and then connecting to opendatacam ui from an external device in the "docker" guide: https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/JETSON_NANO.md

But you could SSH into your jetson nano from another devices and run the commands from there also

tdurand, 2019-09-23:
Have you been able to solve this ? Closing but feel free to reopen

closed 2019-09-23 by tdurand

collabnix, 2020-01-10:
Yes, this error occur when YOLO doesnt detect any object. I tried pointing my USB cam to very lowest light it failed to detect object. As soon as I put it in bright area, it worked without any error message.