opendatacam/opendatacam #126: Problem when trying to access from external port

ItelBz, 2019-09-19:
Hello,
I have installed OpenDataCam on a Jetson nano, all works grate but I have a problem when I try to access the OpenDataCam web page from the external port.
If I browse on jetsonip: 8080 it works but when I browse from externalip: 8080 it returns the error 500.
In my firewall I forwarded ports 8080, 8070, 8090 and 21017, ports 8070 and 8090 work properly, the only one that didn't work is port 8080.

Is there a kind of block when accessing from an external port? How can I change port 8080 to try another port?

Thanks in advance.

tdurand, 2019-09-19:
Hi, 

This is weird, I would think it is the machine you are trying to access from that doesn't allow 8080. Maybe try to open this url on your browser phone (if connected to the same network) to make sure...

It is not straithforward to change the port ,  you will need to change in this line: https://github.com/opendatacam/opendatacam/blob/master/server.js#L21 

But to able to do it, you need to install opendatacam without docker: https://github.com/opendatacam/opendatacam/blob/master/documentation/USE_WITHOUT_DOCKER.md .

And not , there are no block because the actual "official" setup do tell you to access the jetson nano from an external device: https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/JETSON_NANO.md

ItelBz, 2019-09-20:
Hi,
i've checked that the port 8080 is opened on my device, I've also tried to change port, after install opendatacam without docker, but it gives error 500 when trying to browser the page, but the json stream (port 8070) and the video stream (port 8090) are working.

Any idea?

Thank you!

tdurand, 2019-09-20:
If 8070 and 8090 works, it's definitely a problem of "firewall" on port 8080 .. either on the jetson nano side or on your other device... I would think it is on your other device that is why I asked you if you tried with another laptop / smartphone to access 8080 ?

Also you can try to start a dump server on 8080 on the jetson nano and try to access it from outside

```
npm install -g serve
serve -l 8080
```

Try to access http://IP_OF_YOUR_MACHINE:8080

ItelBz, 2019-09-22:
If i try to access to http://IP:8080 it gives me error, but if I access to http://IP:8080/status (or other api links) it work. If i try with serve -l 8080 or with a simple apache server the port 8080 is working.
I attache the page with the error and the code, console is blank.
[image]
[image]

tdurand, 2019-09-23:
Ok, so this is not a problem of the firewall, sorry. It is the app that gives a 500 error... Very weird if the rest is working.

Have you installed with or without docker ?

ItelBz, 2019-09-23:
I've tried with and without docker, same error. Can it be an Next.js error?

tdurand, 2019-09-23:
Yes it is.. But very weird it crashes only when you access from an external device.. 

Could you have a look at the logs ? 

If you run it with docker it should be:

```
### get the container id
sudo docker container list
### Get logs
sudo docker logs <containerID>
```

If you run without docker, just see what is printed in the console , or if you run with a process manager like pm2 , do `pm2 logs`

tdurand, 2019-09-23:
You might be able also to see logs by opening http://ipofjetson:8080/console

ItelBz, 2019-09-23:
I attach the log file.
[log.txt](https://github.com/opendatacam/opendatacam/files/3642809/log.txt)

`

tdurand, 2019-09-23:
Hum.. I don't see any errors from next.js... Actually I struggle a lot to understand how you can load the interface from localhost but not from an external IP... You confirm that the interface load as expected when launching http://localhost:8080 from the jetson ? And what happen if from this same jetson you try the http://IP:8080 instead of locahost ? 

You can try to start it in dev mode.. (with the version without docker) . In order to do that instead of starting the app with npm run start, start it from "npm run dev" , it will run on simulation mode and not pick your video stream, but it might give more insight of what is falling on next.js side.

ItelBz, 2019-09-24:
Yes, when I launch it from http://localhost:8080 or http://localIP:8080 it works perfectly. Only if i launch from http://PublicIP:8080 it gives me error 500.

In dev mode it returns this error

> An undefined error was thrown sometime during render... See https://err.sh/zeit/next.js/threw-undefined
Error: An undefined error was thrown sometime during render... See https://err.sh/zeit/next.js/threw-undefined
    at DevServer.renderErrorToHTML (/home/itel/darknet/opendatacam/node_modules/next/dist/server/next-dev-server.js:148:19)
    at process._tickCallback (internal/process/next_tick.js:68:7)

tdurand, 2019-09-24:
Hum, the error doesn't explain much unfortunately...

We can try to debug it, what happen if you replace the index.js file content with

```javascript
import React from 'react'
import Layout from '../components/shared/Layout'
import MainPage from '../components/MainPage'

import { setURLData, loadConfig, restoreUiSettings } from '../statemanagement/app/AppStateManagement';
import { restoreCountingAreas } from '../statemanagement/app/CounterStateManagement';

class Index extends React.Component {

  static async getInitialProps (params) {
    const { store, isServer, req, query } = params;
    if (isServer) {
      await store.dispatch(restoreCountingAreas(req))
      await store.dispatch(restoreUiSettings(req))
      await store.dispatch(setURLData(req));
      await store.dispatch(loadConfig(req));      
    }
  }

  render () {
    return (
      <div>TEST</div>
    )
  }
}

export default Index
```

tdurand, 2019-09-24:
I mean this file: `pages/index.js`  : https://github.com/opendatacam/opendatacam/blob/master/pages/index.js

ItelBz, 2019-09-25:
I've tried but it gives me the same error

> An undefined error was thrown sometime during render... See https://err.sh/zeit/next.js/threw-undefined
Error: An undefined error was thrown sometime during render... See https://err.sh/zeit/next.js/threw-undefined
    at DevServer.renderErrorToHTML (/home/itel/darknet/opendatacam/node_modules/next/dist/server/next-dev-server.js:148:19)
    at process._tickCallback (internal/process/next_tick.js:68:7)

[newlog.txt](https://github.com/opendatacam/opendatacam/files/3650980/newlog.txt)

I attach the log file.

tdurand, 2019-09-25:
Ok, now try to change to this: This should work and display TEST

```javascript
import React from 'react'
import Layout from '../components/shared/Layout'
import MainPage from '../components/MainPage'

import { setURLData, loadConfig, restoreUiSettings } from '../statemanagement/app/AppStateManagement';
import { restoreCountingAreas } from '../statemanagement/app/CounterStateManagement';

class Index extends React.Component {

  render () {
    return (
      <div>TEST</div>
    )
  }
}

export default Index
```

If it does work, then try to change to this

```javascript
import React from 'react'
import Layout from '../components/shared/Layout'
import MainPage from '../components/MainPage'

import { setURLData, loadConfig, restoreUiSettings } from '../statemanagement/app/AppStateManagement';
import { restoreCountingAreas } from '../statemanagement/app/CounterStateManagement';

class Index extends React.Component {

  static async getInitialProps (params) {
    const { store, isServer, req, query } = params;
    if (isServer) {
      await store.dispatch(restoreCountingAreas(req))
      await store.dispatch(restoreUiSettings(req))
      //await store.dispatch(setURLData(req));
      await store.dispatch(loadConfig(req));      
    }
  }

  render () {
    return (
      <div>TEST</div>
    )
  }
}

export default Index
```

If this also works, this means that the error is coming from the setURLData() method. Will debug it further

ItelBz, 2019-09-25:
The first code is working, shows "TEST".
The second one returns error 500.

tdurand, 2019-09-25:
Does this works ?

```javascript
import React from 'react'
import Layout from '../components/shared/Layout'
import MainPage from '../components/MainPage'

import { setURLData, loadConfig, restoreUiSettings } from '../statemanagement/app/AppStateManagement';
import { restoreCountingAreas } from '../statemanagement/app/CounterStateManagement';

class Index extends React.Component {

  static async getInitialProps (params) {
    const { store, isServer, req, query } = params;
    if (isServer) {
      await store.dispatch(restoreCountingAreas(req))
      await store.dispatch(restoreUiSettings(req))
      //await store.dispatch(setURLData(req));
      //await store.dispatch(loadConfig(req));      
    }
  }

  render () {
    return (
      <div>TEST</div>
    )
  }
}

export default Index
```

ItelBz, 2019-09-25:
> Does this works ?
> 
> ```js
> import React from 'react'
> import Layout from '../components/shared/Layout'
> import MainPage from '../components/MainPage'
> 
> import { setURLData, loadConfig, restoreUiSettings } from '../statemanagement/app/AppStateManagement';
> import { restoreCountingAreas } from '../statemanagement/app/CounterStateManagement';
> 
> class Index extends React.Component {
> 
>   static async getInitialProps (params) {
>     const { store, isServer, req, query } = params;
>     if (isServer) {
>       await store.dispatch(restoreCountingAreas(req))
>       await store.dispatch(restoreUiSettings(req))
>       //await store.dispatch(setURLData(req));
>       //await store.dispatch(loadConfig(req));      
>     }
>   }
> 
>   render () {
>     return (
>       <div>TEST</div>
>     )
>   }
> }
> 
> export default Index
> ```

No, this returns error 500.

tdurand, 2019-09-25:
Ok, seems there is an error with this code: https://github.com/opendatacam/opendatacam/blob/master/statemanagement/app/AppStateManagement.js#L76  when it run on the server with an external IP.. 

I'll try to reproduce myself but it's definitely a bug, it'll work on fixing this in the next release.

Many thanks for the help

ItelBz, 2019-09-30:
Hi, sorry for being so annoying but are there some news? I tried to do some changes in the code but I have no idea how to proceed. Any help so I can do some test myself?
Thanks!

tdurand, 2019-09-30:
Hi, could you try this PR : https://github.com/opendatacam/opendatacam/pull/128

You can just checkout branch : fix-externalip

I didn't reproduce myself so this might not work

tdurand, 2020-04-28:
closing due to inactivity, some work has been done here in case we have a new report of this: https://github.com/opendatacam/opendatacam/pull/128

closed 2020-04-28 by tdurand