opendatacam/opendatacam #131: Memory leak using Gstreamer pipelines as video input

tdurand, 2019-10-04:
There is memory leaks issues when using video inputs like:

- Webcam as a gstreamer pipeline: `"v4l2src device=/dev/video0 ! video/x-raw, framerate=30/1, width=640, height=360 ! videoconvert ! appsink"`
- Raspberrypi cam : `"nvarguscamerasrc ! video/x-raw(memory:NVMM),width=1280, height=720, framerate=30/1, format=NV12 ! nvvidconv ! video/x-raw, format=BGRx, width=640, height=360 ! videoconvert ! video/x-raw, format=BGR ! appsink"`

It seems related to OpenCV and VideoCapture API . 

Filled two issue on nvidia forum and darknet:

- https://devtalk.nvidia.com/default/topic/1064350/jetson-nano/memory-leak-running-darknet-yolo-with-opencv-on-a-gstreamer-input/
- https://github.com/AlexeyAB/darknet/issues/4031

One thing that would be good would be to repro the bug with OpenCV 3.4.7 (latest) and fill an issue on the OpenCV repo.

muukkivi, 2019-10-07:
-c 1 did not help :(, it still grows darknet memory till it freezes around 55% of usage in my jetson tx2 case.

tdurand, 2019-10-07:
Hi @muukkivi . I had the same experience with TX2 while testing last friday. With either -c 1 or gstreamer pipeline I had a leak... Whereas with the jetson nano I don't with -c 1.. I even tried to update to OpenCV latest 2.4.7 .. 

I have no time to investigate more right now... I think the way forward will be to use Deepstream: https://github.com/opendatacam/opendatacam/issues/112

muukkivi, 2019-10-07:
ok, will try that. I might be able to test something on this issue for the future releases if you need?
I tried i cannot run the separate darknet with -c 1 version, it asks for gstreamer and errors pop up. Anyway if you need testing let me know.

muukkivi, 2020-01-09:
interesting got myself a jetson nano, and exactly the same problem - in 10 to 11 minutes the system freezes. I wonder how many people are there who do not have this problem? I mean they an run the system for hours or days? And if yes, what is the difference in set up?

muukkivi, 2020-01-10:
tried something similar on nano but it did not change things:
https://devtalk.nvidia.com/default/topic/1057460/jetson-tx2/custom-built-opencv-3-4-0-following-nvidia-advice-not-working/post/5363355/#5363355

still if some of us are not getting the memory leak, then it would be great to know what is the difference with their nano or jetson tx2 configuration.

muukkivi, 2020-01-10:
Considering this - if by following exactly the opendatacam installing instructions on how to implement the software on jetson tx2 and jetson nano - the result is in average 4-15min of working time. Then this should be well-stated on the projects main page that currently the solution is not stable due to memory leakage and therefore system cannot be functionally used but only for short testing periods (anyone who wants to help in testing and developing feel free to contribute). I mean I've put in over thousand euros on this and over 100 hours of trying to get it working from June 2019 with no good results. No offence but everyone should be aware of this leakage before starting to buy nvidia cards including the high end xavier for only vehicle counting purpose as is my case for example. I mean the information should be very clear that the program currently probably will not function.

shams3049, 2020-01-10:
I have the solution running on multiple devices, yes, its crashes, but its works longer than in your case.

Regress to earliest commits of the code(if you want to make the code running), as from my experience, in the last 6 months, alot had changed from Jetpack to Cuda to all the surrounding libraries, and the documents do not reflect that. Right now I think the focus for this project on my part in to get deepstream running and linked (alot of the issues related to this upgrade).

tdurand, 2020-01-14:
Also, as a temp fix, make sure you have a swap partition on your jetson nano, it might help with runnin time: https://github.com/opendatacam/opendatacam/blob/master/documentation/jetson/JETSON_NANO.md#3-setup-a-swap-partition

muukkivi, 2020-01-14:
yes, did that already before proceeding to test the opendatacam

tdurand, 2020-03-02:
Seems there was some improvements on the darknet side, the author added an alternative way to run darknet which seems to improve things for some people: https://github.com/AlexeyAB/darknet/issues/4453#issuecomment-569513695 

Also Jetpack now ships with OpenCV v4 and gstreamer support built in

I'm gonna update everything and try if the runtime and memory usage is improved. cc @shams3049

tdurand, 2020-03-02:
After installing on a nano with Jetpack latest (4.3) and making the latest version of darknet with OpenCV v4 and gstreamer (default from this jetpack version) I found out that:

* Running darknet with `./darknet detector demo` - original C implementation of detection on video that is built-in darknet , is **still leaking like previously**, crashed after only 10 min on a jetson nano

* Running darknet with `./uselib` - a new C++ implementation of detection on video that was [made by the author of the darknet fork](https://github.com/AlexeyAB/darknet/issues/4453#issuecomment-569513695), is **not leaking** 🎉🎉 , still running with no memory increase after 2 hours

This confirms that the problems comes from darknet and not from OpenCV / Gstreamer.

The less good news is that this new way of running darknet with `./uselib` is lacking essentials features we need to integrate to opendatacam:

- Beeing able to run without a GUI
- Beeing able to foward the video as a MJPEG stream
- Beeing able to send detections as a json stream
- Beeing able to input a gstreamer pipeline

This "theoretically" shouldn't be that hard to integrate as we would "just" need to port them from the old C API if I were a bit more proficient with C / C++  . I answered the author of the fork for some pointers: https://github.com/AlexeyAB/darknet/issues/4453#issuecomment-593450826  , will see as he is generally very responsive.

This is definitely very promising to be able to get rid of this issue and make OpenDataCam much more reliable. cc @b-g

tdurand, 2020-03-03:
Good news 🎉, I have been able to enhance this new ./uselib API for darknet with missing things, still a work in progress: https://github.com/tdurand/darknet/pull/1/files

I'm still figuring out some details for the JSON stream of detections to be able to hook opendatacam node app to it but it is mostly under control, and adding things didn't seem to create any memory leak at first sight. 

Will finish tomorrow and hopefully have a first build of opendatacam to test without this infamous memory leak.

tdurand, 2020-03-04:
Some update on this, I've finished the first integration (see changes to be made to darknet: https://github.com/tdurand/darknet/pull/1/files) and was able to run properly opendatacam using this new darknet API on both Jetsons and Ubuntu CUDA machine.

So far no sign of memory leaks from the tests I did. (ran more than 10 hours on a jetson nano)

I'm now working on providing a build installable via Docker, and also on updating the documentation for a non-docker install.. I aim to have something ready to be tested by end of this week.

The Docker build is a bit more work to do than expected because with this new darknet API our docker pipeline for jetson's isn't working out of the box (cause of opencv versions I think). Instead of patching even more an already pretty hacky way of dockerize opendatacam on jetson's, I'm looking into using the newly released Jetpack base container for docker, which should simplify the pipeline (and unify between CUDA machine and jetsons) and also normally gives us the possibility to use CSI camera (raspberry pi cam for example) from docker.

cc @b-g

b-g, 2020-03-04:
Mega! 🎉 + 🤞!!

tdurand, 2020-03-05:
Some update,  switching to the new "dockerization" pipeline for jetson turns out to be trickier than expected: https://github.com/opendatacam/opendatacam/issues/178 

That said, I successfully managed to generate a build for jetson nano with our current pipeline after fixing another issue 😎..  So build v3.0.0-alpha.1 is on its way, I will be available by tomorrow to test as I'm uploading it to dockerhub right now

Also I've updated the documentation to run v3.0.0-alpha.1 without docker: https://github.com/opendatacam/opendatacam/blob/v3.0.0-alpha.1/documentation/USE_WITHOUT_DOCKER.md

tdurand, 2020-03-06:
Looking for people to test OpenDataCam v3.0.0-alpha.1 on Jetson nano to see if memory leak problem is fixed

#### Docker setup

Build only available on jetson nano for now.

Should work from jetpack 4.2.x , if not please flash to 4.3

```
# Remove previous install files or run setup from a new folder

Remove config.json, install-opendatacam.sh, run-docker.sh, run-opendatacam.sh

# Download install script
wget -N https://raw.githubusercontent.com/opendatacam/opendatacam/v3.0.0-alpha.1/docker/install-opendatacam.sh

# Give exec permission
chmod 777 install-opendatacam.sh

# NB: You will be asked for sudo password when installing the docker container

# Install command for Jetson Nano
./install-opendatacam.sh --platform nano
```

#### Non-docker setup:

If you want to install without docker (on another platform than jetson nano), follow this documentation:

https://github.com/opendatacam/opendatacam/blob/v3.0.0-alpha.1/documentation/USE_WITHOUT_DOCKER.md

b-g, 2020-03-06:
@tdurand That is great! Many thanks!! Could you also share here how to track the memory usage / potential leaks in an easy way?

tdurand, 2020-03-06:
Yes you can use htop from terminal

```
sudo apt-get install htop

htop
```

It will display the processes and the memory usage, you will have a line with `./uselib blabalbalbala` which should be on top and on the left you have the memory usage..

tdurand, 2020-03-06:
Also this change might have introduced other bugs, interested if you can test a bit all the features... I've seem for example that the FPS indicator wasn't very reliable.

tdurand, 2020-03-23:
Ok, this seems to be fixed, closing in favor of : https://github.com/opendatacam/opendatacam/issues/182

closed 2020-03-23 by tdurand