opendatacam/opendatacam #133: UnhandledPromiseRejectionWarning

felixJets, 2019-10-07:
Hi,

I use the OpenDataCam without docker on a Jetson TX2 and it seems like everything was ok until I started a training process, like shown here: https://github.com/AlexeyAB/darknet#how-to-train-to-detect-your-custom-objects . (The process is already finished)

Now I face this issue:

-----------------------------------
Process YOLO initialized
> Ready on http://localhost:8080
{ Error: connect ECONNREFUSED 127.0.0.1:27017
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1106:14)
  name: 'MongoNetworkError',
  errorLabels: [ 'TransientTransactionError' ],
  [Symbol(mongoErrorContextSymbol)]: {} }

.
.
.
FPS:9.4
 MJPEG-stream sent. 
Objects:

 JSON-stream sent. 

FPS:9.1
 MJPEG-stream sent. 
Objects:

 JSON-stream sent. 

FPS:8.8
MJPEG(node:9303) UnhandledPromiseRejectionWarning: Error: connect ECONNREFUSED 127.0.0.1:27017
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1106:14)
(node:9303) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 698)
(node:9303) UnhandledPromiseRejectionWarning: Error: connect ECONNREFUSED 127.0.0.1:27017
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1106:14)
(node:9303) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 699)
(node:9303) UnhandledPromiseRejectionWarning: Error: connect ECONNREFUSED 127.0.0.1:27017
    at TCPConnectWrap.afterConnect [as oncomplete] (net.js:1106:14)
(node:9303) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 700)
.
.
.
.
Objects:

 JSON-stream sent. 

FPS:9.6
 MJPEG-stream sent. 
Objects:

 JSON-stream sent. 

FPS:9.4
 MJPEG-stream sent. 
Objects:

 JSON-stream sent. 
.
.
.

This seems a bit similar to #39 , but I could not observe the OpenDataCam stop counting. In the Live View or Counter Moder it behaves normal. But the main problem is that all the data in the Data mode disappeared (fortunately I saved the most important information somewhere else before) and the data from new recordings is not saved. The Data mode only displays the data of an ongoing recording. And if I try to access the JSON or CSV-file, it just keeps loading, but does not open.

Any ideas what I can do?

tdurand, 2019-10-07:
I think it can't connect to the Mongodb, verify mongodb is started.

Try to type
```
mongo
```

in the shell.

If it doesn't work try restarting mongodb

```
# Start service
sudo systemctl start mongod

# Enable service on boot
sudo systemctl enable mongod
```

felixJets, 2019-10-07:
@tdurand, it did not work but it brought me to the right solution.

For the training-setup I needed to install cmake, and cmake required libcurl4.
It seems like libcurl3 and libcurl4 together are causing trouble (respectively, I think libcurl4 removed libcurl3 and vice versa), so I ran:

`sudo apt-get install -y openssl libcurl3 mongodb-org`

and after that

`sudo systemctl start mongo `
and
`sudo systemctl enable mongod`
worked

Furthermore all the data is visible again.
Many thanks for the help!

closed 2019-10-07 by felixJets